boschsensortec/BMI160_SensorAPI #4: bmi160_accel_foc_trigger_xyz() and bmi160_set_foc_gyro_enable() error handling is incorrect

joaoc-oliveira, 2017-01-19:
These operation have 2 issues:

1. The timeout is too restrictive. The maximum time they wait is 50ms when the datasheet clearly says that the calibration can take up to 250ms.

2.  When the calibration times out, the return result is not correct due to a bug in the code. It is missing an else. The functions should be:
```
BMI160_RETURN_FUNCTION_TYPE bmi160_accel_foc_trigger_xyz(u8 v_foc_accel_x_u8,
u8 v_foc_accel_y_u8, u8 v_foc_accel_z_u8, s8 *v_accel_off_x_s8,
s8 *v_accel_off_y_s8, s8 *v_accel_off_z_s8)
{

(...)

            com_rslt += bmi160_get_foc_rdy(
            &focstatus);
            if ((com_rslt != SUCCESS) ||
            (focstatus != BMI160_FOC_STAT_HIGH)) {
                while ((com_rslt != SUCCESS) ||
                (focstatus != BMI160_FOC_STAT_HIGH
                && v_timeout_u8 <
                BMI160_MAXIMUM_TIMEOUT)) {
                    p_bmi160->delay_msec(
                    BMI160_DELAY_SETTLING_TIME);
                    com_rslt = bmi160_get_foc_rdy(
                    &focstatus);
                    v_timeout_u8++;
                }
            }
            if ((com_rslt == SUCCESS) &&
            (focstatus == BMI160_GEN_READ_WRITE_DATA_LENGTH)) {
                com_rslt +=
                bmi160_get_accel_offset_compensation_xaxis(
                &v_foc_accel_offset_x_s8);
                *v_accel_off_x_s8 =
                v_foc_accel_offset_x_s8;
                com_rslt +=
                bmi160_get_accel_offset_compensation_yaxis(
                &v_foc_accel_offset_y_s8);
                *v_accel_off_y_s8 =
                v_foc_accel_offset_y_s8;
                com_rslt +=
                bmi160_get_accel_offset_compensation_zaxis(
                &v_foc_accel_offset_z_s8);
                *v_accel_off_z_s8 =
                v_foc_accel_offset_z_s8;
            }
            
            /**
            * This else must be added for correct handling of timeout
            * or it will return success even though no offsets were read
            */
            else {
                com_rslt =  ERROR;
            }
        } else {
        com_rslt =  ERROR;
        }
    }
return com_rslt;
}
```
```
BMI160_RETURN_FUNCTION_TYPE bmi160_set_foc_gyro_enable(
u8 v_foc_gyro_u8, s16 *v_gyro_off_x_s16,
s16 *v_gyro_off_y_s16, s16 *v_gyro_off_z_s16)
{

(...)

            com_rslt += bmi160_get_foc_rdy(&focstatus);
            if ((com_rslt != SUCCESS) ||
            (focstatus != BMI160_FOC_STAT_HIGH)) {
                while ((com_rslt != SUCCESS) ||
                (focstatus != BMI160_FOC_STAT_HIGH
                && v_timeout_u8 <
                BMI160_MAXIMUM_TIMEOUT)) {
                    p_bmi160->delay_msec(
                    BMI160_DELAY_SETTLING_TIME);
                    com_rslt = bmi160_get_foc_rdy(
                    &focstatus);
                    v_timeout_u8++;
                }
            }
            if ((com_rslt == SUCCESS) &&
            (focstatus == BMI160_FOC_STAT_HIGH)) {
                com_rslt +=
                bmi160_get_gyro_offset_compensation_xaxis
                (&offsetx);
                *v_gyro_off_x_s16 = offsetx;

                com_rslt +=
                bmi160_get_gyro_offset_compensation_yaxis
                (&offsety);
                *v_gyro_off_y_s16 = offsety;

                com_rslt +=
                bmi160_get_gyro_offset_compensation_zaxis(
                &offsetz);
                *v_gyro_off_z_s16 = offsetz;
            }

            /**
            * This else must be added for correct handling of timeout
            * or it will return success even though no offsets were read
            */
            else {                                                   
                com_rslt = ERROR;                           
            }                                                           
        } else {
        com_rslt = ERROR;
        }
    }
return com_rslt;
}
```

BoschSensortec, 2017-01-20:
@joaoc-oliveira,

We will correct this at the earliest.

BoschSensortec, 2017-07-20:
@joaoc-oliveira ,

This is fixed in the latest release.

closed 2017-07-20 by BoschSensortec