boschsensortec/BMI160_SensorAPI #9: INT on BMM150 through BMI160

MOREmbedded, 2017-07-26:
Hi Bosh,

Could you help me and explain how to enable int pin on BMM150 through BMI160? Is any way to enable pull-up resistor on INT on BMM150 side?

Regards,
Bartosz

MOREmbedded, 2017-08-07:
How to disable internal PU resistors on secondary I2C of BMI160???

BoschSensortec, 2017-08-08:
@MOREmbedded ,

We will be releasing the BMM150 API shortly which will cover your first question. Also refer to the example code in the README file related to connecting and auxiliary sensor. Regarding your second question, I would request you to use our [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form) for obtaining information about the sensor.

WINikolaev, 2018-08-03:
Hello. When the magnetometer PMM150 was operated, its freezing occurred. After switching the power, he again earned. Could you tell me how to deal with this hang