boschsensortec/BMI160_SensorAPI #10: How to initializate SPI user interface?

XaviLC, 2017-08-21:
HEllo,

I have been following the configuration of the BMI160, but I have a couple of questions concerning to the initialization of the reading and writting function for the API.

I see in your help that you saythis
```
sensor.read = user_spi_read;
sensor.write = user_spi_write;
```
Looking into the `bmi160defs.h` code, I see that it has been defined  a pointer to a concrete format when  using read/write function from the SPI interface, should my MCU implement those methods in that way right?
```
typedef int8_t (*bmi160_com_fptr_t)(uint8_t dev_addr, uint8_t reg_addr,
		uint8_t *data, uint16_t len);
```

What is the `sensor.delay_ms` parameter for? And what value should be set tO?

Thanks!

BoschSensortec, 2017-08-21:
@XaviLC ,

For the moment, you can refer to the section **Templates for function pointers**  in the [BME280 README file](https://github.com/BoschSensortec/BME280_driver).

Basically the delay_ms function pointer is required by the API to wait for a fixed duration while an operation is being completed by the sensor.

XaviLC, 2017-08-21:
@BoschSensortec 

Thanks for replying, 
So If I define my functions like 
```
bmi160_com_fptr_t spi1_read_transfer(unsigned char slaveDeviceId,unsigned char reg_addr, unsigned char *rxData, unsigned char bytesNumberRx) ;
 bmi160_com_fptr_t spi1_write_transfer(unsigned char slaveDeviceId, unsigned char reg_addr, unsigned char *txData, unsigned char bytesNumberTx) ;
```
then the initialization of my spi function should be

sensor.read = *spi1_read_transfer;
sensor.write = *spi1_write_transfer;

BoschSensortec, 2017-08-21:
@XaviLC ,

The simplified way would be 
``` c
int8_t spi1_read_transfer(uint8_t slaveDeviceId, uint8_t reg_addr, uint8_t *txData, uint16_t bytesNumberTx);
int8_t spi1_write_transfer(uint8_t slaveDeviceId, uint8_t reg_addr, uint8_t *txData, uint16_t bytesNumberTx);

sensor.read = spi1_read_transfer;
sensor.write = spi1_write_transfer;
```

Depending on your compiler,
``` c
/* uint8_t could be unsigned char and 
   uint16_t could be unsigned int */
```

The functions that you implement should return 0 for success and non-zero for a failure.

Can you describe which platform you are attempting to use the BMI160 API on?

XaviLC, 2017-08-21:
I am trying to use a Nordic NRF52832 platform which will connect to the BMI160 via its SPI interface

BoschSensortec, 2017-08-21:
@XaviLC ,

Which SDK are you using? Are you using any known SDKs like Arduino, mbed, or are you using Nordic's native SDK?

XaviLC, 2017-08-21:
I am using native [Nordic SDK12.2](https://www.nordicsemi.com/eng/nordic/Products/nRF5-SDK/nRF5-SDK-v12-zip/54291)

BoschSensortec, 2017-08-21:
@XaviLC 

Hope this snippet helps.

``` c
nrf_drv_spi_t *p_spi_instance;

/* Initialize p_spi_instance without an ss_pin. ss_pin = NRF_DRV_SPI_PIN_NOT_USED*/	 

int8_t spi1_read_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length);
int8_t spi1_write_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length);

void link_read_write(void)
{
        sensor.read = spi1_read_transfer;
        sensor.write = spi1_write_transfer;
}

int8_t spi1_read_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	ret_code_t ret;
	
	/* Set chip/slave select low */
	
	ret = nrf_drv_spi_transfer(p_spi_instance, &reg_addr, 1, NULL, 0);
	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(p_spi_instance, NULL, 0, reg_data, length);
	
	/* Set chip/slave select high */
	
	return (int8_t)ret;	
}

int8_t spi1_write_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	ret_code_t ret;
	
	/* Set chip/slave select low */
	
	ret = nrf_drv_spi_transfer(p_spi_instance, &reg_addr, 1, NULL, 0);
	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(p_spi_instance, reg_data, length, NULL, 0);
	
	/* Set chip/slave select high */
	
	return (int8_t)ret;	
}
```

XaviLC, 2017-08-21:
@BoschSensortec 

Thank you very much!!

When in your snippet you comment  `/* Set chip/slave select low */`, do you mean that I have to set my CSN Pin down/high right?

BoschSensortec, 2017-08-21:
@XaviLC ,

Before the transactions, set the pin down/low and after set it to up/high.

You could use the NRF SDK functions 
``` c
nrf_drv_gpiote_out_clear(pin); /* To activate the sensor on the bus, i.e., set the chip/slave select low */

nrf_drv_gpiote_out_set(pin); /* To deactivate the sensor on the bus, i.e., set the chip/slave select high*/
```

In case you are using multiple sensor on the bus, you could also store the **pin** in **dev_id** of the device structure before initialization like sensor.read and sensor.write and use **dev_id** instead of **pin** in the code above.

XaviLC, 2017-08-21:
@BoschSensortec 

Thanks again, I have followed your advise and I have done
`bmi160Sensor->bmi160Settings.id         = SPIS_CSN_PIN;`

However, just one remark so that no one complains of your answer @BoschSensortec , better to use the functions for nrf52 otherwise, you have to initialise the gpiote service..

```
nrf_gpio_pin_clear(pin); /* To activate the sensor on the bus, i.e., set the chip/slave select low */

nrf_gpio_pin_set(pin); /* To deactivate the sensor on the bus, i.e., set the chip/slave select high*/
```

One related question, the code that you have set bove for reading/writting is not blocking, is it?

BoschSensortec, 2017-08-21:
@XaviLC ,

If you are using the id parameter to pass the pin, then you can pass dev_id to the gpio function.
``` c
nrf_gpio_pin_clear(dev_id); 
```

Additionally, whether or not the call is blocking solely depends on the implementation of the transfer functions and it's configuration in the SDK. For instance, if the EasyDMA feature is used, the RTOS can process another task to run instead of idling while the transfer is ongoing. The API itself is numb to the implementations of the linked functions.

XaviLC, 2017-08-21:
@BoschSensortec 

Thanks, I was using it so :)

IN my case the EasyDMA is enabled indeed, but I have seen the following loop in on of their examples, so I wonder whether this will make the function blocking or not.

```
 while (!spi_xfer_done) {
  __WFE();
 }
```

BoschSensortec, 2017-08-21:
@XaviLC ,

That snippet will block. 
It is recommended to block when configuring or setting up the sensor.

You may close the issue if there are no further questions.

XaviLC, 2017-08-22:
@BoschSensortec one last question,  in the case of reading your function was

```
int8_t spi1_read_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	ret_code_t ret;
	
	/* Set chip/slave select low */
	
	ret = nrf_drv_spi_transfer(p_spi_instance, &reg_addr, 1, NULL, 0);
	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(p_spi_instance, NULL, 0, reg_data, length);
	
	/* Set chip/slave select high */
	
	return (int8_t)ret;	
}
```

But should the pointer to `reg_data`  be passed as reference so that the buffer it points to can be addressed?

	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(p_spi_instance, NULL, 0, &reg_data, length);

BoschSensortec, 2017-08-22:
@XaviLC ,

The basic idea is to send a byte over MOSI and discard the byte read from MISO in order to tell the sensor about the reg_addr. 
This implementation can be modified depending on the available API. You could also use the macro NRF_DRV_SPI_XFER_TX for instance to setup the transaction. 
Chapter 3.2.2 of the[ datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000-07.pdf) gives a clearer idea of what the data on the bus should look like.

Unario, 2017-09-17:
@BoschSensortec 
I noticed that this thread is related to one of my previous questions about SPI write. In the above code snippet:
```C
int8_t spi1_write_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	ret_code_t ret;
	
	/* Set chip/slave select low */
	
	ret = nrf_drv_spi_transfer(p_spi_instance, &reg_addr, 1, NULL, 0);
	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(p_spi_instance, reg_data, length, NULL, 0);
	
	/* Set chip/slave select high */
	
	return (int8_t)ret;	
}
```
It seems that you are writing multiple bytes by sending the register address of the first byte followed by all the data bytes. Is it correct?

But in the datasheet of BMI160 Chapter 3.2.2, the above method (1-byte addr + multiple-byte) is only mentioned for read operations. So just wanted to double check. Should multiple write operations be performed in which of the following fashion?
1) 1-byte addr + multiple-byte data
or
2) 1-byte addr + 1-byte data + 1-byte addr + 1-byte +... (If I remember correctly, this is the example in the previous version of the BMI 160 driver, bmi160_support.c)

BoschSensortec, 2017-09-18:
@Unario ,

There are 2 write modes supported by the BMI160 depending on the power mode. The normal mode supports 
> CS-low, 1-byte addr + multiple-byte data, CS-High

and the low power modes supports

> CS-low, 1-byte addr + 1-byte data, CS-High | delay | CS-low, 1-byte addr + 1-byte data, CS-High

This switch is handled by the API (bmi160_set_regs) and not be done at the spi write and read functions for the sake of unified abstraction. 

It is important to note that the delay plays a very crucial role here.

pendragonformsrepo, 2017-10-11:
Initialization (bmi160_init) and configuration seems to succeed with BMI160_OK.  However, the bmi160_perform_self_test fails.  

Stepping through the bmi160_perform_self_test for the accelerator seems to have no errors until it reaches the validate_accel_self_test comparison.  Stepping through the bmi160_perform_self_test for the gyro fails at the last step BMI160_GET_BITS.   

Any ideas what the issue may be here?

BoschSensortec, 2017-10-12:
Hi @pendragonformsrepo ,

We have tested this on multiple platforms and haven't run into this error. We will get back to you as soon as we have an update. In the meantime, can you elaborate a little on the nature of the statement 

> bmi160_perform_self_test for the gyro fails at the last step BMI160_GET_BITS.

pendragonformsrepo, 2017-10-12:
We get through the bmi160_init and it seems to be correct returning the correct 0xd1 chip_id.  

The accel selftest (bmi160_perform_self_test) fails every time.   Since posting this, the self test on the gyro works once but after calling bmi160_get_sensor_data (which is returning gyro data for x/y/z as zeros), we loop back around after 1000ms and try the gyro test again and the second time it fails.  

It is returning BMI160_W_GYRO_SELF_TEST_FAIL (bmi160.c, v3.6.1, line 4834) which is the result of the BMI160_GET_BITS on line 4829.  The accel is failing with BMI160_W_ACCEl_SELF_TEST_FAIL (bmi160.c, v3.6.1, line 4773).

 **Just to confirm our BMI160 shuttle board wiring:**
VDD (shuttle pin 1 @ 3.3V)
GND (shuttle pin 3)
MISO (shuttle pin 4)
MOSI (shuttle pin 5)
SCK (shuttle pin 6)
SS (shuttle pin 7)
J206 detached?
J207 detached?

BoschSensortec, 2017-10-13:
@pendragonformsrepo ,

It is a bug due to loss of sync between the dev struct and the sensor. As a work-around you can call the bmi160_init function after a self-test until the next release where this will be rectified.

pendragonformsrepo, 2017-10-16:
Thanks.  We're still getting the chip_id but not consistently getting data from the gyro and never from the accelerometer.  Also, the gyro periodically succeeds self test but the accelerometer never succeeds (we have commented the self test out but periodically have uncommented it to see if it succeeds). 
 We ordered a new shuttle board just in case it is hardware issue. Is there any way we could be inadvertently disabling one or both these functions?

BoschSensortec, 2017-10-17:
@pendragonformsrepo ,

Could you create a brief communication log so we can attempt to duplicate the issue at hand?

pendragonformsrepo, 2017-10-19:
We still need to do some more testing but believe we figured it out.  Admittedly, an air-headed mistake.  We were look deep functions and states but it was right in front of us.  We had hard coded the **bmi160_dev** _delay_ms_ function to a static period based on an example we saw when we should have been using the _period_ value that was passed.  This understandably caused erratic behavior.  Now it seems to work more consistently, where we are getting both accel and gyro readings.
For the nRF52, this is what we are now using for the sensor delay_ms callback:

void user_delay_ms(uint32_t period)
{
	if (period==NULL){
		period = 1;
	}

    nrf_delay_ms(period);
}

feliwir, 2017-11-19:
Hey,

do you have any suggestions how I can implement the template function pointers for the SPI interface using the Arduino SPI library? I think this snippet would be interesting since many people use Arduino. For delay_ms I can just use the built in delay function on atmega

BoschSensortec, 2017-11-20:
@feliwir ,

The SPI functions would look something like,

```c

const uint8_t BMI160_CS = 5; // An example for digital pin 5 being used as a chip select

// Int the settings of the struct bmi160_dev, set the dev_id to BMI160_CS

int8_t Sensors::user_spi_read(uint8_t dev_id, uint8_t reg_addr,
                            uint8_t *reg_data, uint16_t len)
{
    SPISettings set;
    int pin = 0;
    switch(dev_id)
    {
        case BMI160_CS: // The same CS is now available here
            set = set_bmi;
            pin = BMI160_CS; // Gpio used as a chip select
            break;
    }

    //read all bytes
    SPI.beginTransaction(set);
    digitalWrite (pin, LOW);

    SPI.transfer(reg_addr); // Write the register address, ignore the return
    for(int i=0;i<len;++i)
        reg_data[i] = SPI.transfer(0); //Store the bytes in reg_data

    digitalWrite (pin, HIGH);
    SPI.endTransaction();
}

int8_t Sensors::user_spi_write(uint8_t dev_id, uint8_t reg_addr,
                            uint8_t *reg_data, uint16_t len)
{
    SPISettings set;
    int pin = 0;
    switch(dev_id)
    {
        case BMI160_CS:
        set = set_bmi;
        pin = BMI160_CS; // Gpio used as a chip select
        break;
    }

    //send all bytes
    SPI.beginTransaction(set);
    digitalWrite (pin, LOW);

    SPI.transfer(reg_addr); // Write the register address, ignore the return
    for(int i=0;i<len;++i)
        SPI.transfer(reg_data[i]); // Write bytes from reg_data

    digitalWrite (pin, HIGH);
    SPI.endTransaction();
}
```

feliwir, 2017-11-20:
thanks a lot

feliwir, 2017-11-21:
Is it on purpose that read/write don't return anything?

BoschSensortec, 2017-11-22:
@feliwir ,

A couple of notes from your code,
1) Maximum SPI frequency is 10MHz. Depending on the length of the wire, other sources of impedance and interference, this could be lower.
2) The supported modes for both the sensors is SPI_MODE0 and SPI_MODE3. In addition, it should MSBFIRST.
3) General [convention](https://stackoverflow.com/questions/484462/difference-between-i-and-i-in-a-loop) to increment the index of the loop is to use a postfix `for(int i=0;i<len;i++)`
4) The use of the return parameter is to inform the API if the communication failed. 0 for success, failure otherwise.

feliwir, 2017-11-22:
@BoschSensortec  thanks that helped again.
After making all your proposed changes and testing the BME280 started working, but unfortunaly the BMI160 is still giving me "DEV_NOT_FOUND" on bmi_init . Any idea why one of the sensors might be working but the other isn't? Thanks

BoschSensortec, 2017-11-23:
@feliwir ,

Is your hardware fixed, as in a soldered PCB? Or is it a prototype where you can change the connections? If latter, then you could try swapping the chip selects in the code and also the chip selects connected to the sensors to see if the BMI160 works with the former BME280 connections.

feliwir, 2017-11-23:
@BoschSensortec
it’s soldered on a PCB

BoschSensortec, 2017-11-23:
@feliwir ,

Is it a commercially available board? Can you drop a link here?

feliwir, 2017-11-23:
@BoschSensortec no, it's a custom designed pcb. I'll link the schematic here: 
[scem.pdf](https://github.com/BoschSensortec/BMI160_driver/files/1498302/scem.pdf)

As i said both sensors worked before with some Arduino libraries for BMI and BME, so it seems to be all soldered correctly

feliwir, 2017-11-24:
But as i said it is only not working with this driver in particular

BoschSensortec, 2017-11-24:
@feliwir ,

Do you have a logic trace with a tool like the Saleae logic analyzer?

BoschSensortec, 2017-11-24:
@feliwir ,

Which driver of the BMI160 was it working with?

feliwir, 2017-11-24:
@BoschSensortec This one: https://github.com/hanyazou/BMI160-Arduino
And yes we have a logic analyzer and pins for SPI debug

BoschSensortec, 2017-11-24:
@feliwir ,

I have looked through the schematics and the code that you linked and do not see anything obvious that is wrong.  Can you trace the lines from power on until the end of the bmi160_init?

Also, is your updated code available at the same gist?

feliwir, 2017-11-24:
@BoschSensortec I‘ve updated the public gist now

BoschSensortec, 2017-11-24:
@feliwir ,

We tested with the following snippet. Can you check if the BMI160 works with this Arduino snippet? Note that the bmi160.c, bmi160.h & bmi160_defs.h are in the same project.

The board settings in the Arduino IDE were, `Arduino Pro or Pro mini, Atmega328P(3.3V, 8MHz) `, `Arduino avr boards version 1.6.20 `

```c
#include "SPI.h"
#include "bmi160.h"

/* Helper functions */
int8_t spiTransfer(uint8_t cs, uint8_t reg_addr, uint8_t *reg_data, uint16_t length);
void delay_ms(uint32_t period)
{
  delay(period);
}

/* Constants and variables */
SPISettings set_bmi(1000000, MSBFIRST, SPI_MODE0);
const uint8_t bmi160_cs = 8;
struct bmi160_dev bmi_dev;
String output;

/* Application code execution starts here */
void setup() {
  /* Initialize debug serial communication at 9600 bits per second: */
  Serial.begin(9600);

  /* Initialize sensor interface */
  pinMode(bmi160_cs, OUTPUT);
  SPI.begin();

  /* Set the inferace configurations */
  bmi_dev.id = bmi160_cs;
  bmi_dev.read = spiTransfer; // Duplex
  bmi_dev.write = spiTransfer; // Duplex
  bmi_dev.delay_ms = delay_ms;
  bmi_dev.interface = BMI160_SPI_INTF;

  /* Initialize the BMI160 */
  output = "BMI160 init "  + String(bmi160_init(&bmi_dev)) + " , chip ID: " + String(bmi_dev.chip_id, HEX);
  Serial.println(output);

  /* Select the Output data rate, range of accelerometer sensor */
  bmi_dev.accel_cfg.odr = BMI160_ACCEL_ODR_100HZ;
  bmi_dev.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
  bmi_dev.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

  /* Select the power mode of accelerometer sensor */
  bmi_dev.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

  /* Select the Output data rate, range of Gyroscope sensor */
  bmi_dev.gyro_cfg.odr = BMI160_GYRO_ODR_100HZ;
  bmi_dev.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
  bmi_dev.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

  /* Select the power mode of Gyroscope sensor */
  bmi_dev.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;

  /* Set the sensor configuration */
  output = "BMI160 set sensor config:  " + String(bmi160_set_sens_conf(&bmi_dev));
  Serial.println(output);
}

/* The loop routine runs over and over again forever after the setup */
void loop() {
  bmi160_sensor_data accel;
  bmi160_sensor_data gyro;

  /* To read both Accel and Gyro data */
  output = "R: " + String(bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL), &accel, &gyro, &bmi_dev));
  output += " A: " + String(accel.x) + ", "  + String(accel.y) + ", " + String(accel.z);
  output += " G: " + String(gyro.x) + ", "  + String(gyro.y) + ", " + String(gyro.z);
  Serial.println(output);
  delay(100);
}

int8_t spiTransfer(uint8_t cs, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
  SPI.beginTransaction(set_bmi);
  digitalWrite(cs, LOW);
  SPI.transfer(reg_addr);
  for (uint8_t i = 0; i < length; i++)
    reg_data[i] = SPI.transfer(reg_data[i]);
  digitalWrite(8, HIGH);
  SPI.endTransaction();

  return 0;
}
```

feliwir, 2017-11-24:
@BoschSensortec 
Wow, thanks a lot. The snippet is indeed working. Thanks for your effort I‘ll compare to my code and check the differemces

feliwir, 2017-11-26:
@BoschSensortec It never works when both BMI & BME pin are in Output mode

BoschSensortec, 2017-11-27:
@feliwir ,

After `pinMode(bmi160_cs, OUTPUT);`, you will also need to call `digitalWrite(bmi160_cs, HIGH);` to deselect the sensor. Same for the bme280.

VladyslavUsenko, 2017-12-21:
Hi all,

I'm trying to read the data from BMI160 on the Intel Aero drone (https://github.com/intel-aero/meta-intel-aero/wiki/94-(References)-Sensors-GPIOs-LEDs-CAN-bus#command-to-access-the-imu-sensor-on-the-aero-compute-board) and have few questions about it:

1. Is it possible to use this driver when BMI160 is exposed as linux device? There are code snippets (https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/Documentation/spi/spidev_fdx.c?id=refs/tags/v4.4.32) for read and write functions, but they work with files, not with registers.
2. If not, is there another BMI160 driver for Linux?

Thanks!

BoschSensortec, 2018-01-10:
Hi @VladyslavUsenko,

Check out the following issue reported for another driver, you may find hints on how to implement it for your platform.

https://github.com/BoschSensortec/BME680_driver/issues/14 

Basically the code provided by us is an API abstraction over 3 functions, read, write and delay.

matiasvc, 2018-01-22:
@BoschSensortec That example is for i2c. The BMI160 is connected with SPI on the Aero board. Is there an example of how the use the driver with SPI?

BoschSensortec, 2018-01-22:
Hi @matiasvc,

Can you replace with the read and write functions with the following snippets? It is not clear what the communicated values are (reg_data) because you are printing the memory address of reg_data.

```c
int8_t user_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    printf("READ: dev_id: %d reg_addr: 0x%x reg_data: 0x%x len: %d\n", dev_id, reg_addr, reg_data[0], len);

    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

    if (write(spiDeviceHandle, &reg_addr, 1) != 1) {
		rslt = -1;
    }
    if (read(spiDeviceHandle, reg_data, len) < len) {
		rslt = -1;
    }

    return rslt;
}

int8_t user_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    printf("WRITE: dev_id: %d reg_addr: 0x%x reg_data: 0x%x len: %d\n", dev_id, reg_addr, reg_data[0], len);
    
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

    uint8_t reg[16];
    reg[0]=reg_addr;
	
    for (int idx = 1; idx < len + 1; idx++)
       reg[idx] = reg_data[idx - 1];

    if (write(spiDeviceHandle, reg, len + 1) < len + 1) {
		rslt = -1;
        exit(1);
    }

    return rslt;
}
```

matiasvc, 2018-01-23:
@BoschSensortec when replacing the read and write functions with your code i get the following output:
```
READ: dev_id: 0 reg_addr: 0xff reg_data: 0x0 len: 1
READ: dev_id: 0 reg_addr: 0x80 reg_data: 0x0 len: 1
READ: dev_id: 0 reg_addr: 0x80 reg_data: 0xff len: 1
READ: dev_id: 0 reg_addr: 0x80 reg_data: 0xff len: 1
Return Value -3
```

BoschSensortec, 2018-01-23:
Hi @matiasvc,

Testing an example on the RaspberryPi's ioctl module. Will get back to you if it can be extended on the Intel's Aero platform.

BoschSensortec, 2018-01-23:
Hi @matiasvc,

Based on Gordon Henderson's wiringPi library, 
https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPi.c 
https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPiSPI.c

Check the following snippet as reference. Note that the snippet is using the ioctl & linux spidev drivers.

```c

#include <stdio.h>
#include <stdarg.h>
#include <stdint.h>
#include <stdlib.h>
#include <ctype.h>
#include <poll.h>
#include <unistd.h>
#include <errno.h>
#include <string.h>
#include <time.h>
#include <fcntl.h>
#include <pthread.h>
#include <sys/time.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <sys/wait.h>
#include <sys/ioctl.h>
#include <asm/ioctl.h>
#include <linux/spi/spidev.h>

/*********************** User function prototypes ************************/
int8_t comm_spi_setup(uint8_t channel, uint32_t speed, uint8_t mode);

int8_t comm_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length);

int8_t comm_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length);

void comm_delay_ms(uint32_t period);

/************************ User static variables *****************************/
static uint32_t spiSpeeds[2];
static int spiFds[2];
static const char *spiDev0 = "/dev/spidev0.0";
static const char *spiDev1 = "/dev/spidev0.1";
static const uint8_t spiBPW = 8;
static const uint16_t spiDelay = 0;

/*********************** Local function prototypes ************************/
static int8_t spi_transfer(uint8_t channel, uint8_t *data, uint16_t length);

/*********************** User function definitions ************************/
int8_t comm_spi_setup(uint8_t channel, uint32_t speed, uint8_t mode)
{
	int fd;

	mode &= 3;	// Mode is 0, 1, 2 or 3
	channel &= 1;	// Channel is 0 or 1

	if ((fd = open(channel == 0 ? spiDev0 : spiDev1, O_RDWR)) < 0) {
		printf("Unable to open SPI device: %s\n", strerror(errno));
		return -1;
	}

	spiSpeeds[channel] = speed;
	spiFds[channel] = fd;

	// Set SPI parameters.

	if (ioctl(fd, SPI_IOC_WR_MODE, &mode) < 0) {
		printf("SPI Mode Change failure: %s\n", strerror(errno));
		return -1;
	}

	if (ioctl(fd, SPI_IOC_WR_BITS_PER_WORD, &spiBPW) < 0) {
		printf("SPI BPW Change failure: %s\n", strerror(errno));
		return -1;
	}

	if (ioctl(fd, SPI_IOC_WR_MAX_SPEED_HZ, &speed) < 0) {
		printf("SPI Speed Change failure: %s\n", strerror(errno));
		return -1;
	}

	return fd;
}

int8_t comm_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	uint16_t i;
	uint16_t len = length + 1;
	uint8_t data[len];
	data[0] = reg_addr;

	spi_transfer(dev_id, data, len);

	for (i = 0; i < len; i++)
		reg_data[i] = data[i + 1];

	/*printf("Read:: D:%d, RA:%d, C:%d\n", dev_id, reg_addr, length);

	 for( i=0; i<len+1;i++)
	 printf("%d\n", data[i]);*/

	return 0;
}

int8_t comm_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{

	uint16_t i;
	/*printf("Write:: D:%d, RA:%d, RD:%d, C:%d\n", dev_id, reg_addr, reg_data[0], length);
	 for( i=1; i<length;i++)
	 printf("%d\n", reg_data[i]);*/

	uint8_t data[length + 1];
	data[0] = reg_addr;

	for (i = 0; i < length; i++)
		data[i + 1] = reg_data[i];

	spi_transfer(dev_id, data, length + 1);
	return 0;
}

void comm_delay_ms(uint32_t period)
{
	struct timespec sleeper, dummy;

	sleeper.tv_sec = (time_t) (period / 1000);
	sleeper.tv_nsec = (long) (period % 1000) * 1000000;

	nanosleep(&sleeper, &dummy);
}

/*********************** Local function definitions ***********************/
static int8_t spi_transfer(uint8_t channel, uint8_t *data, uint16_t length)
{
	struct spi_ioc_transfer spi;

	channel &= 1;

	// Mentioned in spidev.h but not used in the original kernel documentation
	//	test program )-:

	memset(&spi, 0, sizeof(spi));

	spi.tx_buf = (__u64) data;
	spi.rx_buf = (__u64) data;
	spi.len = length;
	spi.delay_usecs = spiDelay;
	spi.speed_hz = spiSpeeds[channel];
	spi.bits_per_word = spiBPW;

	return ioctl(spiFds[channel], SPI_IOC_MESSAGE(1), &spi);
}

```

matiasvc, 2018-01-26:
@BoschSensortec Thanks, I was able to get it working using this example.
For anyone else with the same problem. The following is a bit messy, but working proof of concept code:

```
#include <stdio.h>
#include <time.h>
#include <fcntl.h>
#include <unistd.h>
#include <errno.h>
#include <sys/ioctl.h>
#include <linux/spi/spidev.h>

#include "bmi160.h"

#define SPIDEV "/dev/spidev%d.%d"

static const uint16_t spiDelay = 0;

static int8_t spi_transfer(uint8_t channel, uint8_t *data, uint16_t length);

typedef struct{
	char node[20];
	uint8_t mode;
	uint8_t bits;
	uint32_t speed;
	int fd;
} spidev;

spidev dev;

int spi_open(int bus, int cs)
{
	sprintf(dev.node, SPIDEV, bus, cs);

	dev.fd = open(dev.node, O_RDWR);
	if (dev.fd < 0)
	{
		printf("Unable to open SPI device: %s\n", dev.node);
		return -1;
	}

	if (ioctl(dev.fd, SPI_IOC_RD_MODE, &dev.mode) == -1)
	{
		printf("ERROR: faild to get mode %s\n", dev.node);
		return -1;
	}

	if (ioctl(dev.fd, SPI_IOC_RD_BITS_PER_WORD, &dev.bits) == -1)
	{
		printf("ERROR: faild to get bits per word %s\n", dev.node);
		return -1;
	}

	/* max speed hz */
	if (ioctl(dev.fd, SPI_IOC_RD_MAX_SPEED_HZ, &dev.speed) == -1)
	{
		printf("ERROR: faild to get max speed hz %s\n", dev.node);
		return -1;
	}

	printf("spi mode: %d\n", dev.mode);
	printf("bits per word: %d\n", dev.bits);
	printf("max speed: %d Hz\n", dev.speed);

	return 0;
}

void spi_close()
{
	close(dev.fd);
}

void user_delay_ms(uint32_t period)
{
	struct timespec sleeper, dummy;

	sleeper.tv_sec = (time_t) (period / 1000);
	sleeper.tv_nsec = (long) (period % 1000) * 1000000;

	nanosleep(&sleeper, &dummy);
}

int8_t comm_spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	uint16_t i;
	uint16_t len = length + 1;
	uint8_t data[len];
	data[0] = reg_addr;

	spi_transfer(dev_id, data, len);

	for (i = 0; i < len; i++)
		reg_data[i] = data[i + 1];
	/*
	printf("Read:: D:%d, RA:%d, C:%d\n", dev_id, reg_addr, length);

	 for( i=0; i<len+1;i++)
	 printf("%d\n", data[i]);
	*/
	return 0;
}

int8_t comm_spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{

	uint16_t i;
	/*
	printf("Write:: D:%d, RA:%d, RD:%d, C:%d\n", dev_id, reg_addr, reg_data[0], length);
	 for( i=1; i<length;i++)
	 printf("%d\n", reg_data[i]);
	 */

	uint8_t data[length + 1];
	data[0] = reg_addr;

	for (i = 0; i < length; i++)
		data[i + 1] = reg_data[i];

	spi_transfer(dev_id, data, length + 1);
	return 0;
}

static int8_t spi_transfer(uint8_t channel, uint8_t *data, uint16_t length)
{
	struct spi_ioc_transfer spi;

	memset(&spi, 0, sizeof(spi));

	spi.tx_buf = (__u64) data;
	spi.rx_buf = (__u64) data;
	spi.len = length;
	spi.delay_usecs = spiDelay;
	spi.speed_hz = dev.speed;
	spi.bits_per_word = dev.bits;

	return ioctl(dev.fd, SPI_IOC_MESSAGE(1), &spi);
}

int main(int argc, char const *argv[])
{	
	
	spi_open(3, 0);

	struct bmi160_dev sensor;

	/* You may assign a chip select identifier to be handled later */
	sensor.id = 0;
	sensor.interface = BMI160_SPI_INTF;
	sensor.read = comm_spi_read;
	sensor.write = comm_spi_write;
	sensor.delay_ms = user_delay_ms;

	int8_t rslt = BMI160_OK;
	rslt = bmi160_init(&sensor);

	printf("Result %i\n", rslt);

	rslt = BMI160_OK;

	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE; 

	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&sensor);

	for (int i = 0; i < 10000; ++i)
	{
		rslt = BMI160_OK;
		struct bmi160_sensor_data accel;
		struct bmi160_sensor_data gyro;

		/* To read only Accel data */
		rslt = bmi160_get_sensor_data(BMI160_ACCEL_SEL, &accel, NULL, &sensor);

		/* To read only Gyro data */
		rslt = bmi160_get_sensor_data(BMI160_GYRO_SEL, NULL, &gyro, &sensor);

		/* To read both Accel and Gyro data */
		bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL), &accel, &gyro, &sensor);

		/* To read Accel data along with time */
		rslt = bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_TIME_SEL) , &accel, NULL, &sensor);

		/* To read Gyro data along with time */
		rslt = bmi160_get_sensor_data((BMI160_GYRO_SEL | BMI160_TIME_SEL), NULL, &gyro, &sensor);

		/* To read both Accel and Gyro data along with time*/
		bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL | BMI160_TIME_SEL), &accel, &gyro, &sensor);

		printf("Result %i\n", rslt);
		printf("%i %i %i %i\n", accel.x, accel.y, accel.z, accel.sensortime);
		printf("%i %i %i %i\n", gyro.x, gyro.y, gyro.z, gyro.sensortime);

		user_delay_ms(100);
	}

	

	spi_close();
	return 0;
}
```

Yamini7895, 2018-01-30:
Hey,
I am working with STM32 NUCLEO-F401RE board. do you have any suggestions how I can implement the template function pointers for the SPI interface using the bmi160 -driver [https://github.com/BoschSensortec/BMI160_driver] with STM32 board?

BoschSensortec, 2018-01-30:
Hi @Yamini7895,

Can you link us to the HAL (Hardware abstraction layer) or the IDE that you are using for the STM32 board? Is it the [STM32 Cube](http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-mcu-packages/stm32cubef4.html)?

Yamini7895, 2018-01-30:
hi @BoschSensortec 
We are using System Workbench IDE for the STM32 Board.
yes,it is STM32 Cube.

kont40, 2018-02-23:
Hello,
I am working on a custom board that involves BMI160.
Initialization passes well, and I am getting data from gyroscope, but always getting zeros from accelerometer. I have checked the discussions above, but so far didn't resolve the problem.
Any advice will be appreciated.

BoschSensortec, 2018-02-26:
Hi @kont40,

Do you have a read/write log of the communication?

trixr4kdz, 2018-03-03:
Hi @matiasvc, 

I tried to run your code but the print outs that I'm getting don't seem to be correct or even changing. Unless I restart running the file, I don't get the any different readings for the accel and gyro. Would you know what I am doing wrong? What I did is essentially copy your code into a new file and called it `aero_bmi160.c` and then I compiled it with`gcc aero_bmi160.c bmi160.c -o aero_bmi160`. I then ran it with `./aero_bmi160` but I get a continuous printout of something like the following over and over:
```
Result 0
8192 -17953 -109 127
-17953 -109 127 127
```
and my Aero is definitely not moving when I run `./aero_bmi160`

BoschSensortec, 2018-03-05:
@trixr4kdz,

You will have to check for the valid spidev configuration on the Intel Aero. It is possible that the spidev is disabled.

trixr4kdz, 2018-03-06:
Hi @BoschSensortec it turns out that I just had to run it with `sudo`. Thanks for all the help!

[7 more replies]