boschsensortec/BMI160_SensorAPI #12: i2c user read/write functions

Jean-Bert, 2017-08-30:
Hi,
Im currently working on a project using the bmi160 with a Texas Instruments cc2650 board. I am having trouble interfacing between the two using i2c. Unfortunately, I don't have access to a logic analyser, so its been a rather painful exercise. Can some please elaborate on the code required for the "user_read/write_funciton"s required for i2c interfacing?
This is what I have so far...

int8_t my_i2c_read (uint8_t dev_addr, uint8_t reg_addr, uint8_t *data, uint16_t len)
{
        int j = 0;
        uint8_t txBuffer[100];
        i2cTransaction.slaveAddress = dev_addr;                   
        txBuffer[0] = reg_addr;
        for(j = 0; j < len-1; j++)
        {
            txBuffer[j+1] = data[j];
        }
        i2cTransaction.writeBuf = txBuffer;
        i2cTransaction.writeCount = len;
        i2cTransaction.readBuf = rxBuffer;
        i2cTransaction.readCount = 1;
        return rxBuffer[0];
}

BoschSensortec, 2017-08-30:
Hi @Jean-Bert ,

Can you provide a reference on where we can find the code or documentation for the CC2650 I2C driver that you are using here, specifically the i2cTransaction structure?

Thank you.

Jean-Bert, 2017-08-30:
Hi,
Thanks for getting back to me. I have attached the i2c code that is provided by Texas Instruments for the cc2650.
Kind regards

[I2C_Driver.txt](https://github.com/BoschSensortec/BMI160_driver/files/1265376/I2C_Driver.txt)

BoschSensortec, 2017-08-31:
Hi @Jean-Bert ,

Your snippet for reading data assuming the I2C write is performed before the read,  would look like
```c
int8_t my_i2c_read (uint8_t dev_addr, uint8_t reg_addr, uint8_t *data, uint16_t len)
{
    i2cTransaction.slaveAddress = dev_addr;
    i2cTransaction.writeBuf = &reg_addr;
    i2cTransaction.writeCount = 1;
    i2cTransaction.readBuf = data;
    i2cTransaction.readCount = len;

    if (!I2C_transfer(handle, &i2cTransaction)) {
         System_printf("Unsuccessful I2C transfer");
        return -1; /* Return a non-zero to indicate communication error to the API */
    }
    return 0; /* Return 0 to indicate success */
}
```

closed 2017-10-18 by BoschSensortec

pejad, 2018-01-19:
Hi there,
I'm also interfacing with the BMI160 using a BGM11S SiLabs micro and am having trouble writing  the I2C read/write function . Could you please assist me in providing an example of the **user_i2c_read** and **user_i2c_write** functions?  

I'm trying to configure an any_motion interrupt.

Thanks

I2C drivers attached here
[em_i2c.txt](https://github.com/BoschSensortec/BMI160_driver/files/1645864/em_i2c.txt)
[i2cspm.txt](https://github.com/BoschSensortec/BMI160_driver/files/1645865/i2cspm.txt)

Here's an example of I2C comms with a pressure sensor.

void Pressure_Sensor_Read(I2C_TypeDef *i2c) {
	I2C_TransferSeq_TypeDef    seq;
	I2C_TransferReturn_TypeDef ret;

	uint8_t                    i2c_read_data[3];
	uint8_t                    i2c_write_data[1];

	seq.addr  = PSR_ADDR;
	seq.flags = I2C_FLAG_WRITE;

	UDELAY_Delay(500);
	i2c_write_data[0] = PSR_CONVERT;
	seq.buf[0].data   = i2c_write_data;
	seq.buf[0].len    = 1;

	ret = I2CSPM_Transfer(i2c, &seq);
	if(ret != i2cTransferDone) {
		return;
	}

	Delay(30);

	seq.addr  = PSR_ADDR;
	seq.flags = I2C_FLAG_WRITE_READ;

	i2c_write_data[0] = PSR_ADC_READ;
	seq.buf[0].data   = i2c_write_data;
	seq.buf[0].len    = 1;

	seq.buf[1].data = i2c_read_data;
	seq.buf[1].len  = 3;

	ret = I2CSPM_Transfer(i2c, &seq);
	if(ret != i2cTransferDone) {
		return;
	}

	D = 0;
	D =i2c_read_data[0];
	D = (D << 8) | i2c_read_data[1];
	D = (D << 8) | i2c_read_data[2];
}

BoschSensortec, 2018-01-19:
Hi @pejad, 

Hope the following snippet helps!

```c
#include <stdint.h>

#include "em_chip.h"
#include "em_gpio.h"
#include "em_i2c.h"
#include "em_rtc.h"
#include "em_cmu.h"
#include "em_emu.h"

#define I2C_TXBUFFER_SIZE	32
#define I2C_RXBUFFER_SIZE	32

uint8_t I2C_write_buffer[I2C_TXBUFFER_SIZE] = { 0 };
uint8_t I2C_read_buffer[I2C_RXBUFFER_SIZE] = { 0 };

void I2C0_init(uint8_t channel)
{
	// Using default settings
	I2C_Init_TypeDef i2cInit_s = I2C_INIT_DEFAULT;
#if I2C_HIGH_FREQUENCY ==  1
	i2cInit_s.refFreq=CMU_ClockFreqGet(cmuClock_HFPER);
	i2cInit_s.freq=I2C_FREQ_FASTPLUS_MAX;
	i2cInit_s.clhr=_I2C_CTRL_CLHR_FAST;
#endif
	PTD_PinModeSet(I2C0_SDA, gpioModeWiredAnd, 1);
	PTD_PinModeSet(I2C0_SCL, gpioModeWiredAnd, 1);
	CMU_ClockEnable(cmuClock_I2C0, true);
	// Using default settings for I2C 0
	// Enable pins at location 1
	I2C0->ROUTE = I2C_ROUTE_SDAPEN | I2C_ROUTE_SCLPEN | (1 << _I2C_ROUTE_LOCATION_SHIFT);
	// Initializing the I2C0
	I2C_Init(I2C0, &i2cInit_s);
	//Enable NVIC I2C Interrupts
	NVIC_ClearPendingIRQ(I2C0_IRQn);
}

int8_t I2C0_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	I2C_TransferReturn_TypeDef ret;	
	// Transfer structure
	I2C_TransferSeq_TypeDef i2cTransfer;
	// Copy data into a single buffer
	I2C_write_buffer[0] = reg_addr;
	int i;
	for (i = 0; i < length; i++)
	{
		I2C_write_buffer[i + 1] = reg_data[i];
	}
	// Initializing I2C transfer
	i2cTransfer.addr = (dev_id << 1); // Shift to 8 bit address
	i2cTransfer.flags = I2C_FLAG_WRITE;
	i2cTransfer.buf[0].data = I2C_write_buffer;
	i2cTransfer.buf[0].len = length + 1;
	i2cTransfer.buf[1].data = I2C_read_buffer;
	i2cTransfer.buf[1].len = I2C_RXBUFFER_SIZE;

	ret = I2C_TransferInit(I2C0, &i2cTransfer);
	// Wait while data is transferred
	do {
		ret = I2C_Transfer(I2C0);
	} while (ret == i2cTransferInProgress);

	return (int8_t)ret;
}

int8_t I2C0_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	I2C_TransferReturn_TypeDef ret;	
	// Transfer structure
	I2C_TransferSeq_TypeDef i2cTransfer;
	// Initializing I2C transfer
	i2cTransfer.addr = (dev_id << 1);  // Shift to 8 bit address
	i2cTransfer.flags = I2C_FLAG_WRITE_READ;
	i2cTransfer.buf[0].data = &reg_addr;
	i2cTransfer.buf[0].len = 1;
	i2cTransfer.buf[1].data = reg_data;
	i2cTransfer.buf[1].len = length;

	ret = I2C_TransferInit(I2C0, &i2cTransfer);
	// Wait while data is transferred
	do {
		ret = I2C_Transfer(I2C0);
	} while (ret == i2cTransferInProgress);

	return (int8_t)ret;
}

```

kranthibv, 2018-02-19:
Hi there,
I'm also interfacing with the BMI160(boostxl-sensors hardware) using a TEXAS CC3220SF and am having trouble writing the I2C read/write and delay functions . Could you please assist me in providing an example of the user_i2c_read and user_i2c_write functions?

in CC3220SF, i cannot access the csb pin as it is internally connected to vdd. and SPI communication is not possible for boostxl-sensor booster pack with cc3220sf hardware.

i am using the https://github.com/BoschSensortec/BMI160_driver
thank you

BoschSensortec, 2018-02-19:
Hi @kranthibv,

Can you provide a quick link for the I2C driver for the CC3220SF?

kranthibv, 2018-02-19:
i have to use the i2c files from the SDK http://www.ti.com/tool/download/SIMPLELINK-CC3220-SDK in energia

[I2CCC32XX_c.txt](https://github.com/BoschSensortec/BMI160_driver/files/1736289/I2CCC32XX_c.txt)
[I2CCC32XX_h.txt](https://github.com/BoschSensortec/BMI160_driver/files/1736290/I2CCC32XX_h.txt)

kranthibv, 2018-02-21:
hi @BoschSensortec 
i was able to get all the sensor outputs from boostxl sensor booster pack using cc3220sf by following the instructions from http://www.ti.com/tool/download/SIMPLELINK-SDK-SENSOR-ACTUATOR-PLUGIN. but this is using Code Composer Studio and using RTOS kernals. 
i want to use the same headers and drivers from sensor plugin in energia by following the original bosch sensortech drivers. will this information help you in guiding me.

thank you
[guiboostxlsensors.zip](https://github.com/BoschSensortec/BMI160_driver/files/1743010/guiboostxlsensors.zip)

BoschSensortec, 2018-02-21:
Hi @kranthibv,

Assuming you would like to use the Arduino like interfaces present in the Energia IDE, check out the ArduinoExample.cpp on our [Gist](https://gist.github.com/BoschSensortec) page.

kranthibv, 2018-02-22:
Hi @BoschSensortec 
This is what i need exactly. i will test this and will definitely confirm the status of working with demos.
thank you

EliKrumholz, 2018-09-07:
@BoschSensortec The ArduinoExample.cpp gist is very helpful, it got me up and running. However, it took me a bit of digging to find this issue and then get directed to the gist, maybe a brief snippet in the readme on what behavior is expected of the user defined functions and a link to the ArduinoExample.cpp would be helpful to other users who are starting out?