boschsensortec/BMI160_SensorAPI #14: When will BMM150 driver be available?

Unario, 2017-09-11:
It's mentioned in the BMI160 document how to configure BMM150 using the secondary interface. However, the corresponding BMM150 driver is currently unavailable. Could you please let me know when you will publish the BMM150 driver? It would be great if we can see it as soon as possible, because it is important to our project. Thanks!

BoschSensortec, 2017-09-12:
@Unario ,

You can now access the [BMM150 API](https://github.com/BoschSensortec/BMM150-Sensor-API).

Unario, 2017-09-12:
Hello,

Is there a function to extract the mag data from the FIFO frame of BMI160 as for the accel (bmi160_extract_accel) and gyro data (bmi160_extract_gyro)?

Thanks,

BoschSensortec, 2017-09-13:
Hi @Unario ,

This operation is done in 2 steps. A bmi160_extract_aux (will be added in an upcoming release) and [bmm150_aux_mag_data ](https://github.com/BoschSensortec/BMM150-Sensor-API/blob/c7774c906ce98c6a5558db3a752d5cdd0abe260f/bmm150.h#L321) which converts the raw values to usable data.

leetoun, 2017-09-14:
hi@Unario,
i have the same problem. i want to read the  ID of BMM150 using the secondary interface of BMI160. I wonder if you have solve the problem? could you give me some advice?

BoschSensortec, 2017-10-18:
@Unario ,

Please find the latest release of the API with the auxiliary data over FIFO and how to parse it.

closed 2018-02-13 by BoschSensortec