boschsensortec/BMI160_SensorAPI #15: Can't read BMM150 id???

leetoun, 2017-09-12:
hi bosch:
how can i configure the BMI160  to read the id of BMM150? shall i need to read/write the register 0x7F? please give me some hints, or demo better.

BoschSensortec, 2017-09-12:
Hi @leetoun,

Check the chapter [Initialization of auxiliary sensor BMM150](https://github.com/BoschSensortec/BMI160_driver#initialization-of-auxiliary-sensor-bmm150) in the README of the BMI160 API. 

In addition you can use the [BMM150 API](https://github.com/BoschSensortec/BMM150-Sensor-API) to control the BMM150 sensor.

Regarding a demo, is there any specific hardware that you are currently using?

leetoun, 2017-09-12:
hi,
so you mean i can follow the instruction in the README of BMI160 API, and use the BMM150 API to read the id of BMM150?
I'm going to have a try.

thanks for your replying!

leetoun@163.com
 
From: Bosch Sensortec
Date: 2017-09-12 17:26
To: BoschSensortec/BMI160_driver
CC: leetoun; Mention
Subject: Re: [BoschSensortec/BMI160_driver] Can't read BMM150 id??? (#15)
Hi @leetoun,
Check the chapter Initialization of auxiliary sensor BMM150 in the README of the BMI160 API.
In addition you can use the BMM150 API to control the BMM150 sensor.
Regarding a demo, is there any specific hardware that you are currently using?

BoschSensortec, 2017-09-12:
@leetoun ,

Yes. In case you cannot get it to work, you can post your queries here.

leetoun, 2017-09-13:
Q1：
I have followed your instruction, and i can read the id of BMI160,
**but** after this statement **"**rslt = bmi160_aux_init(&sensor);**"**, 
**in** this function "**bmm150_init(&bmm150)**"
**after**  /* Power up the sensor from suspend to sleep mode */
 i still can't read the id of BMM150, is always 0x00???

Q2：
i wonder if you can show me the key points when i want to get the data or id of BMM150 through 
BMI160???

Q3：
It is assumed that secondary interface of bmi160 has external pull-up resistor in order to access the auxiliary sensor bmm150.
what does this mean? it's from https://github.com/BoschSensortec/BMI160_driver

BoschSensortec, 2017-09-14:
@leetoun ,

Looking through your code,
A couple of comments,
1) Ensure that the delay function really waits for the required amount of milliseconds or more. 
2) Check the 'rslt' variable values after the call of each function. It should be 0. 

Also the schematics surrounding the sensors would be helpful.

leetoun, 2017-09-14:
hi @BoschSensortec 
[6轴+地磁.zip](https://github.com/BoschSensortec/BMI160_driver/files/1302337/6.zip)
This is my schematic between BMI160 and BMM150, cloud you give me some advice about that?
Do i need the pullup resistors between BMI160 and BMM150 if I use auxiliary I2C interface between the BMI160 and the BMM150? 
If it is needed to add, there must be something wrong with your data sheet on Page94( **3.3.1 Magnetometer connected to sencondary interface**) 
[https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000-07.pdf](url)

Really appreciate your help!

leetoun, 2017-09-15:
Need help! @BoschSensortec 
[6轴+地磁.zip](https://github.com/BoschSensortec/BMI160_driver/files/1305506/6.zip)

This is my schematic between BMI160 and BMM150, cloud you give me some advice about that?
Do i need the pullup resistors between BMI160 and BMM150 if I use auxiliary I2C interface between the BMI160 and the BMM150?

BoschSensortec, 2017-09-15:
@leetoun ,

Could you share your schematic as a pdf?

leetoun, 2017-09-15:
hi @BoschSensortec 

here it is.
[6轴+地磁.pdf](https://github.com/BoschSensortec/BMI160_driver/files/1305546/6.pdf)

leetoun, 2017-09-15:
hi @BoschSensortec 
[6轴+地磁.pdf](https://github.com/BoschSensortec/BMI160_driver/files/1305569/6.pdf)

Do i need the pullup resistors between BMI160 and BMM150 if I use auxiliary I2C interface between the BMI160 and the BMM150?

BoschSensortec, 2017-09-15:
Hi @leetoun ,

Yes you will need pull-up resistors. As per your schematic, the R01 and R02 are not connected as pull-up resistors. Refer to this [article](http://www.edn.com/design/analog/4371297/Design-calculations-for-robust-I2C-communications) for some information on pull-up resistors on I2C.

leetoun, 2017-09-15:
hi @BoschSensortec 
Thank you very much !

BoschSensortec, 2017-09-15:
@leetoun ,

You're welcome. Do link your project or board here once completed.

closed 2017-09-15 by leetoun