boschsensortec/BMI160_SensorAPI #16: BMI160 secondary interface

leetoun, 2017-09-14:
hi, @BoschSensortec
Q0：
"It is assumed that secondary interface of bmi160 has external pull-up resistor in order to access the auxiliary sensor bmm150."  it's from https://github.com/BoschSensortec/BMI160_driver
but the data sheet says there's no other resistor needed???
do i need pullup resistors when i use BMI160's secondary interface ?

Q1：
I have followed your instruction, and i can read the id of BMI160,
but after this statement "rslt = bmi160_aux_init(&sensor);",
in this function "bmm150_init(&bmm150)"
after /* Power up the sensor from suspend to sleep mode */
i still can't read the id of BMM150, is always 0x00???
and i check the register 0x1B bit2, there's  a manul magnetometer interface operation via MAG_IF[].
But i can't read the right ID form BMM150, SO i  think there's something wrong with the IIC interface,
between BMI160 and BMM150???

BoschSensortec, 2017-09-14:
Hi @leetoun ,

The auxiliary I2C interface between the BMI160 and the BMM150 needs pull-up resistors. Do you already have these pull-ups connected?

leetoun, 2017-09-14:
@BoschSensortec 
Thank you very much!
I'll have a try about that.
May you have a good day!

BoschSensortec, 2017-09-15:
@Unario ,

Burst write or continuous write within a single CSB low transaction is not possible while you are in suspend mode. Find the related note in Chapter 2.2 Power modes of the [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000-07.pdf). 

In order to enable burst writes, put the sensor in normal mode, configure the sensor and then put it back to suspend until needed at a later point in time.

BoschSensortec, 2017-09-15:
@Unario ,

In addition, the API explicitly handles [this case](https://github.com/BoschSensortec/BMI160_driver/blob/e69b9472c2a49300fd86325f374ce0c467a64792/bmi160.c#L1407) related to low power and suspend modes. The typical point of failure here is the delay API.

closed 2017-09-15 by leetoun

neiledgard, 2018-01-16:
I also get 0x00 for CHIP_ID on the BMI160 shuttle board. Do I have to use a pull-up resistor when using the shuttle board?

BoschSensortec, 2018-01-16:
Hi @neiledgard,

Yes, you have to use pull-up resistors while using the shuttle board.

neiledgard, 2018-01-16:
Hi @BoschSensortec, 

Thank you for the info. 

Can you please tell me the pin numbers on the shuttle board I have to connect the pull-up?

Currently I have VDD (J1-1), VDDIO (J1-2), GND (J1-3), INT1 (J1-20), SCL (J1-18), and SDA (J1-17) connected to my nRF52 dev board, plus IO2 (J1-15) to ground (for some reason I had to do this to communicate with the chip).

Also what does COD mean on the Shuttleboard Flyer?

Thank you, Neil

BoschSensortec, 2018-01-16:
Hi @neiledgard,

Assuming you are using the BMI160,

For I2C,
- 1 Vdd
- 2 Vddio
- 3 Gnd
- 4 SPI MISO / I2C addr select - GND for the low address, VddIO for the high address.
- 5 SPI MOSI / I2C SDA
- 6 SPI SCK / I2C SCL
- 7 SPI CS - Tie this pin to VddIO before power on. Once it is toggled, the sensor will be in SPI until the next reset.
- 20 INT1
- 21 INT2 

Depending on which shuttle, IO2 can be connected to any other pin. 

The COD is a 9 bit encoding a unique identifier. This identifier can be used to determine which shuttle is connected.

neiledgard, 2018-01-16:
I reconnected the wires. But still only zeros (0x00) on the secondary interface. Could you advise me on which pin should be connected with pull-up?

BoschSensortec, 2018-01-16:
Hi @neiledgard,

Can you create an independent issue and describe your use-case and what you are trying to do with the sensor? It would be easier to track it.

Thank you.

lucasrangit, 2019-01-07:
> Do I need pull-up resistors when I use BMI160's secondary interface? The data sheet says there's no other resistor needed.

@leetoun , between Feb 2015 and October 2018, the datasheet changed the following in section 3.3.1.

From: _No external pull-up resistors need to be connected, since an internal pull-up register can be configured in the BMI160. No additional I 2 C master or slave devices must be attached to the magnetometer interfaces._

To: _External pull-up resistors need to be connected and no additional I 2 C master or slave devices must be attached to the magnetometer interfaces._

I think this was changed because how to configure the internal pull-ups was never documented. There appears to be undocumented registers to enable the internal pull-ups on the secondary interface.

For example, I found code that uses register 0x85 bit 4 and 5, and somewhere else registers 0x37, 0x9A, 0xC0, 0x90, and 0x80. And there's mention of a hidden register accessed by changing the page table. Very confusing.

@BoschSensortec can you clarify why the internal pull-up resistor documentation was removed? And if it's still supported, what the correct way to enable them?

Thanks!

leetoun, 2019-01-08:
@lucasrangit  thank you for you tips，and  I think you are right 。
 hope bosch will solve the problem！@BoschSensortec 。