boschsensortec/BMI160_SensorAPI #17: Timeout in fast offset calibration

nikosamus, 2017-11-06:
Hi, I'm working with the example provided in the README to set a FOC. After getting the error BMI160_FOC_FAILURE on each try of calling it, I figured out that the timeout of 250ms isn't enought so as a simple patch I changed the code to get at least 325ms which I found enought. 

```
/* original code */
while ((foc_status != BMI160_ENABLE) && (timeout < 11)) {
/* modification */
while ((foc_status != BMI160_ENABLE) && (timeout < 14)) {

```

Additional info:
I'm working with Silicon Labs chips EFM32HGxxx with the RTC Driver configured.

My delay function is:

```
void user_delay_ms( uint32_t ms)
{
	RTCDRV_Delay(ms);
}
```

BoschSensortec, 2017-11-06:
Hi @nikosamus ,

How precise is your RTCDRV_Delay? Does it have a millisecond resolution?

nikosamus, 2017-11-06:
I don't know its precision, but yes resolution its 1ms.

Looking on RTC driver source code I found:
```
// Assume 32kHz RTC/RTCC clock, cmuClkDiv_8 prescaler, 4 ticks per millisecond
#define RTC_DIVIDER                   (cmuClkDiv_8)
```
And in my case I'm using the Low Frequency Crystal Oscillator as a system clock at 32kHz so I suppose its the case.

Thanks!

nikosamus, 2017-11-06:
I found that my patch is not necessary adding a delay of 150ms at least between setting the normal mode (as the README suggest) and the FOC setting.

Thanks!

closed 2017-11-06 by nikosamus