boschsensortec/BMI160_SensorAPI #18: FIFO sensor time reading

qinwang717, 2017-11-13:
Hi, I am having some problem reading the sensor time data through FIFO. I am wondering whether whether this is a bug in the driver or there's anything that I missed out. 

I am trying to read sensor data for both gyro, accel, and magnetometer (configured as the secondary interface) along with the sensor time data. After configured two sensors and the fifo as the readme file instructed, I can read the sensor data correctly, but for the sensortime data, it seems always be 0 in the most of the case, eventhough a overreading is performed. (overread by having a shorter delay time than the time needs to wait for fifo to be filled.)

My FIFO is set up as the same as in the readme file, and but reading from three sensors: 
`bmi160_set_fifo_config(BMI160_FIFO_GYRO | BMI160_FIFO_ACCEL | BMI160_FIFO_AUX Z BMI160_FIFO_HEADER | BMI160_FIFO_TIME, BMI160_ENABLE, &m_bmi160);`

An overread (for triggering the sensortime data) is performed by : 
`m_bmi160_delay_ms(300)`, where the delay time should be over 400 ms because of the output data rate.

I can see that the available FIFO frames are less than before because of the shorter waiting time, so I assume I am "overreading" the FIFO since the description of 'overread' in the datasheet is : 

> if more data bytes are read from the FIFO than valid data bytes are available, '0x80' is returned. Since a header '0x80' indicates an invalid frame, the SW can recognize the end of valid data. After the invalid header the data is undefined.....if header mode and the sensortime frame are enabled, the last data frame is followed by a sensortime frame. 

However, when I output the raw FIFO frame, I can't see any 0x80 frame, unless I change the wait time to extreme short (or zero), but in this case, I am not able to get any useful sensor data.

Also I tried to modify the driver and read more frames than the available frames, for example, if the number of available frames are 10, I read 10 + 4 (sensortime bytes) + 4 (more overread). But this is still not working. 

So I am wondering: 

1. Is this a correct 'overread' operation to get the sensortime data? 
2. If not, how do I get the sensortime data properly through the FIFO ? (to get a steady reading along with other data)

I am looking forward to your reply. 

Many thanks, 
Qin

qinwang717, 2017-11-17:
@BoschSensortec 
Any suggestion ?

BoschSensortec, 2017-11-17:
@qinwang717 ,

We are looking into this and will get back to you shortly.

qinwang717, 2017-11-24:
Is there any information that I can provide you to speed up? 

Thanks.

BoschSensortec, 2017-11-24:
Hi @qinwang717 ,

We understand the issue. It is currently being tested under various scenarios.

BoschSensortec, 2017-11-28:
@qinwang717 ,

The release v3.7.4 should resolve the issue with reading the sensor time from the FIFO.

BoschSensortec, 2017-11-30:
@qinwang717 ,

The example assumes you have configured the Gyroscope at an ODR of 100Hz.

The configuration for FIFO settings should be as below. The wrong macros are used in the example, although the value is identical.
```c
/* Configure the sensor's FIFO settings */
	rslt = bmi160_set_fifo_config(BMI160_FIFO_G_ENABLE | BMI160_FIFO_HEAD_ENABLE | BMII160_FIFO_TIME_ENABLE, BMI160_ENABLE, dev);
```
In the current version of the code, it does the overread if `BMI160_FIFO_TIME_ENABLE` (or as per the typo in the example, `BMI160_FIFO_TIME`) is requested, provided a sufficient FIFO size is available. 

As per your screenshot, it seems that a buffer of sufficient size is available (300 bytes). The odd thing is that the 109 bytes that are read. 25 bytes are the overread which is currently hardcoded but can be modified via your makefile. That leaves 84 bytes of the FIFO length which corresponds to 12 frames of gyro data. But the extract function finds 15 valid frames.

Could you print the FIFO contents and the parsed gyro frames as per the example? It will give an insight as to why the sensor time frame was not detected or parsed.

qinwang717, 2017-11-30:
Hi, thanks for replying.

I set up the gyro odr as 100Hz as below:
`			m_bmi160.gyro_cfg.odr = BMI160_GYRO_ODR_100HZ; `
`m_bmi160.gyro_cfg.range = BMI160_GYRO_RANGE_1000_DPS;`
			`m_bmi160.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;`
			`m_bmi160.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;`

And the fifo_gyro_header_time_data() method is the same as in the example code. 

And here's my FIFO contents and gyro frames. 

 USER REQUESTED FIFO LENGTH : 300

 AVAILABLE FIFO LENGTH : 300

 FIFO DATA INDEX[0] = 136
 FIFO DATA INDEX[1] = 15
 FIFO DATA INDEX[2] = 0
 FIFO DATA INDEX[3] = 40
 FIFO DATA INDEX[4] = 0
 FIFO DATA INDEX[5] = 213
 FIFO DATA INDEX[6] = 255
 FIFO DATA INDEX[7] = 136
 FIFO DATA INDEX[8] = 15
 FIFO DATA INDEX[9] = 0
 FIFO DATA INDEX[10] = 41
 FIFO DATA INDEX[11] = 0
 FIFO DATA INDEX[12] = 213
 FIFO DATA INDEX[13] = 255
 FIFO DATA INDEX[14] = 136
 FIFO DATA INDEX[15] = 17
 FIFO DATA INDEX[16] = 0
 FIFO DATA INDEX[17] = 43
 FIFO DATA INDEX[18] = 0
 FIFO DATA INDEX[19] = 213
 FIFO DATA INDEX[20] = 255
 FIFO DATA INDEX[21] = 136
 FIFO DATA INDEX[22] = 18
 FIFO DATA INDEX[23] = 0
 FIFO DATA INDEX[24] = 41
 FIFO DATA INDEX[25] = 0
 FIFO DATA INDEX[26] = 213
 FIFO DATA INDEX[27] = 255
 FIFO DATA INDEX[28] = 136
 FIFO DATA INDEX[29] = 15
 FIFO DATA INDEX[30] = 0
 FIFO DATA INDEX[31] = 44
 FIFO DATA INDEX[32] = 0
 FIFO DATA INDEX[33] = 212
 FIFO DATA INDEX[34] = 255
 FIFO DATA INDEX[35] = 136
 FIFO DATA INDEX[36] = 13
 FIFO DATA INDEX[37] = 0
 FIFO DATA INDEX[38] = 42
 FIFO DATA INDEX[39] = 0
 FIFO DATA INDEX[40] = 212
 FIFO DATA INDEX[41] = 255
 FIFO DATA INDEX[42] = 136
 FIFO DATA INDEX[43] = 14
 FIFO DATA INDEX[44] = 0
 FIFO DATA INDEX[45] = 43
 FIFO DATA INDEX[46] = 0
 FIFO DATA INDEX[47] = 213
 FIFO DATA INDEX[48] = 255
 FIFO DATA INDEX[49] = 136
 FIFO DATA INDEX[50] = 15
 FIFO DATA INDEX[51] = 0
 FIFO DATA INDEX[52] = 43
 FIFO DATA INDEX[53] = 0
[... 313 lines cut ...]
GYRO X-DATA : 13 	 Y-DATA : 44 	 Z-DATA : -46
FIFO GYRO FRAME[32]
GYRO X-DATA : 14 	 Y-DATA : 44 	 Z-DATA : -45
FIFO GYRO FRAME[33]
GYRO X-DATA : 16 	 Y-DATA : 42 	 Z-DATA : -41
FIFO GYRO FRAME[34]
GYRO X-DATA : 18 	 Y-DATA : 44 	 Z-DATA : -43
FIFO GYRO FRAME[35]
GYRO X-DATA : 16 	 Y-DATA : 42 	 Z-DATA : -41
FIFO GYRO FRAME[36]
GYRO X-DATA : 15 	 Y-DATA : 43 	 Z-DATA : -43
FIFO GYRO FRAME[37]
GYRO X-DATA : 16 	 Y-DATA : 44 	 Z-DATA : -43
FIFO GYRO FRAME[38]
GYRO X-DATA : 16 	 Y-DATA : 41 	 Z-DATA : -45
FIFO GYRO FRAME[39]
GYRO X-DATA : 18 	 Y-DATA : 43 	 Z-DATA : -46
FIFO GYRO FRAME[40]
GYRO X-DATA : 15 	 Y-DATA : 42 	 Z-DATA : -44
FIFO GYRO FRAME[41]
GYRO X-DATA : 15 	 Y-DATA : 44 	 Z-DATA : -46
 SENSOR TIME DATA : 0 
SKIPPED FRAME COUNT : 0

Many thanks!

BoschSensortec, 2017-11-30:
@qinwang717,

You do not see the sensor time frame as the length of the fifo was insufficient or your delay was too long. The example needs to be updated!

Anyway, in the meantime use a shorter delay, or use of a fifo length of at least 350 bytes.

qinwang717, 2017-11-30:
Hi, 

I did try to increase the FIFO size, but if I change the FIFO size to 350, and remain everything else the same, all I can only observe is that the **available FIFO length** is 326 now when **the user requested FIFO length** is 350.  Sensor time data is still 0, and still no FIFO frame has value 0x44: 

USER REQUESTED FIFO LENGTH : 350

 AVAILABLE FIFO LENGTH : 326

 FIFO DATA INDEX[0] = 136
 FIFO DATA INDEX[1] = 19
 FIFO DATA INDEX[2] = 0
 FIFO DATA INDEX[3] = 43
 FIFO DATA INDEX[4] = 0
 FIFO DATA INDEX[5] = 210
 FIFO DATA INDEX[6] = 255
 FIFO DATA INDEX[7] = 136
 FIFO DATA INDEX[8] = 18
 FIFO DATA INDEX[9] = 0
 FIFO DATA INDEX[10] = 44
 FIFO DATA INDEX[11] = 0
 FIFO DATA INDEX[12] = 210
 FIFO DATA INDEX[13] = 255
 FIFO DATA INDEX[14] = 136
 FIFO DATA INDEX[15] = 17
 FIFO DATA INDEX[16] = 0
 FIFO DATA INDEX[17] = 44
 FIFO DATA INDEX[18] = 0
 FIFO DATA INDEX[19] = 213
 FIFO DATA INDEX[20] = 255
 FIFO DATA INDEX[21] = 136
 FIFO DATA INDEX[22] = 18
 FIFO DATA INDEX[23] = 0
 FIFO DATA INDEX[24] = 42
 FIFO DATA INDEX[25] = 0
 FIFO DATA INDEX[26] = 212
 FIFO DATA INDEX[27] = 255
 FIFO DATA INDEX[28] = 136
 FIFO DATA INDEX[29] = 17
 FIFO DATA INDEX[30] = 0
 FIFO DATA INDEX[31] = 44
 FIFO DATA INDEX[32] = 0
 FIFO DATA INDEX[33] = 209
 FIFO DATA INDEX[34] = 255
 FIFO DATA INDEX[35] = 136
 FIFO DATA INDEX[36] = 16
 FIFO DATA INDEX[37] = 0
 FIFO DATA INDEX[38] = 42
 FIFO DATA INDEX[39] = 0
 FIFO DATA INDEX[40] = 213
 FIFO DATA INDEX[41] = 255
 FIFO DATA INDEX[42] = 136
 FIFO DATA INDEX[43] = 17
 FIFO DATA INDEX[44] = 0
 FIFO DATA INDEX[45] = 45
 FIFO DATA INDEX[46] = 0
 FIFO DATA INDEX[47] = 212
 FIFO DATA INDEX[48] = 255
 FIFO DATA INDEX[49] = 136
 FIFO DATA INDEX[50] = 18
 FIFO DATA INDEX[51] = 0
 FIFO DATA INDEX[52] = 43
 FIFO DATA INDEX[53] = 0
 FIFO DATA INDEX[54] = 214
 FIFO DATA INDEX[55] = 255
 FIFO DATA INDEX[56] = 136
 FIFO DATA INDEX[57] = 16
 FIFO DATA INDEX[58] = 0
 FIFO DATA INDEX[59] = 44
 FIFO DATA INDEX[60] = 0
 FIFO DATA INDEX[61] = 210
[... 654 lines cut ...]
GYRO X-DATA : 15 	 Y-DATA : 40 	 Z-DATA : -45
FIFO GYRO FRAME[24]
GYRO X-DATA : 18 	 Y-DATA : 42 	 Z-DATA : -44
FIFO GYRO FRAME[25]
GYRO X-DATA : 19 	 Y-DATA : 43 	 Z-DATA : -43
FIFO GYRO FRAME[26]
GYRO X-DATA : 18 	 Y-DATA : 43 	 Z-DATA : -44
FIFO GYRO FRAME[27]
GYRO X-DATA : 18 	 Y-DATA : 41 	 Z-DATA : -47
FIFO GYRO FRAME[28]
GYRO X-DATA : 14 	 Y-DATA : 43 	 Z-DATA : -43
FIFO GYRO FRAME[29]
GYRO X-DATA : 18 	 Y-DATA : 46 	 Z-DATA : -45
FIFO GYRO FRAME[30]
GYRO X-DATA : 13 	 Y-DATA : 45 	 Z-DATA : -44
FIFO GYRO FRAME[31]
GYRO X-DATA : 16 	 Y-DATA : 41 	 Z-DATA : -44
FIFO GYRO FRAME[32]
GYRO X-DATA : 18 	 Y-DATA : 41 	 Z-DATA : -43
FIFO GYRO FRAME[33]
GYRO X-DATA : 17 	 Y-DATA : 45 	 Z-DATA : -44
 SENSOR TIME DATA : 0 
SKIPPED FRAME COUNT : 0

I am wondering if there's any recommended value combinations for the delay time and the FIFO size in a given odr?

BoschSensortec, 2017-11-30:
Hi @qinwang717 ,

In order to receive the sensor time, you  need to empty the FIFO of all its current frames and the sensor time get appended as an extra frame, i.e., an overread is required to get the sensor time. In the FIFO log that you have sent, the FIFO data still contains valid gyro data frames even until the last byte of the additional 25 bytes overread which basically doesn't provide room to fit in a sensor time frame. You can observe that every 7th frame has a header of 136d and this continues till the last partial gyro data frame. Basically between the read of the FIFO length and the reading of the FIFO, there is a large delay. The API calls the delay function only for a duration of 1ms between reading the FIFO's length and the data of the FIFO. It would seem that the callback that you have linked is delaying for a real-time duration of much longer. I would recommend checking the callback to the delay_ms function. The delay provided should be as accurate as possible to the expected delay period.

qinwang717, 2017-11-30:
Hi, 

Thanks about the previous suggestion, it was very helpful. I found I set the I2C bus speed to a wrong value. I changed that value and now the callback delay has been solved. 

Now I can get the sensor time frames (68d) appended at the end of valid frames, see the attached log:

[SensorTimeFramesUnpacked.txt](https://github.com/BoschSensortec/BMI160_driver/files/1518767/SensorTimeFramesUnpacked.txt)

However, it seems like it's not unpacking the sensortime frame anywhere. Since in the extract_gyro_header_mode() method， switch sentence: 

`if (*gyro_length == gyro_index) `
`{`
      `    /*Number of frames to read completed*/`
      `    break;`
`}`

, which means it only unpack the gyro frames, and break out from the switch after gyro frames are finished. And Nowhere else is unpacking the sensor time frame. 

Is this a issue or am I understanding it wrong? 

Many thanks.

BoschSensortec, 2017-12-01:
Hi @qinwang717,

Yes, this is a potential issue. The API should parse the whole FIFO to search for other frames if they exist. For now you need to request more gyro frames than they are in the FIFO so as to parse the sensor time. 

We will in the meantime work on fixing the issue.

T721, 2018-04-03:
Hi @BoschSensortec ,

I experience the same problem as @qinwang717 
I cannot read out the sensortime and when i use the provided example i can't even read out the data, is it seems to be not calculated properly.

I tried out several configurations and i came to that point where i actually can read the gyro data but the sensor time data is allways zero.
It doesn't matter how many bytes i read, the only thing happening is the sensordata fails to parse.

i provide you a snippet of my code. Maybe you can help me solve this issue.
Im using no delay at all.
What is the advantage of using more than one frame to read out the data? is this a processing thing to provide minimal errors or can i just read one frame and it is the same result?

Here is my code snippet and a formatted log of the output. 
Im using the suggested ODR for the gyro of 100Hz

```
// Declare memory to store the raw FIFO buffer information
uint8_t fifo_buff[600];

//Modify the FIFO buffer instance and link to the device instance
struct bmi160_fifo_frame fifo_frame;
fifo_frame.data = fifo_buff;
fifo_frame.length = 40;
dev->fifo = &fifo_frame;
```

the fifo set cmd:
`rslt = bmi160_set_fifo_config(BMI160_FIFO_G_ENABLE | BMI160_FIFO_HEAD_ENABLE | BMI160_FIFO_TIME_ENABLE, BMI160_ENABLE, dev);`

the Log output:
What does header byte 136 mean? what means this 128 and is it a problem for the processing?
when i want to read 6 bytes of payload i need to read 7bytes as the 7th is the header byte - correct?
I tried to overread the FIFO (40 bytes) but it is not working. There are allways 6 data bytes + header returned.

 AVAILABLE FIFO LENGTH : 21

 FIFO DATA INDEX[0] = 136
 FIFO DATA INDEX[1] = 246
 FIFO DATA INDEX[2] = 255
 FIFO DATA INDEX[3] = 10
 FIFO DATA INDEX[4] = 0
 FIFO DATA INDEX[5] = 5
 FIFO DATA INDEX[6] = 0
 
 FIFO DATA INDEX[7] = 136
 FIFO DATA INDEX[8] = 238
 FIFO DATA INDEX[9] = 255
 FIFO DATA INDEX[10] = 16
 FIFO DATA INDEX[11] = 0
 FIFO DATA INDEX[12] = 10
 FIFO DATA INDEX[13] = 0
 
 FIFO DATA INDEX[14] = 136
 FIFO DATA INDEX[15] = 245
 FIFO DATA INDEX[16] = 255
 FIFO DATA INDEX[17] = 250
 FIFO DATA INDEX[18] = 255
 FIFO DATA INDEX[19] = 6
 FIFO DATA INDEX[20] = 0
 REQUESTED GYRO DATA FRAMES : 1
 
 AVAILABLE GYRO DATA FRAMES : 1
 
FIFO GYRO FRAME[0]
GYRO X-DATA : -10        Y-DATA : 10     Z-DATA : 5
 SENSOR TIME DATA : 0 
SKIPPED FRAME COUNT : 0 

 AVAILABLE FIFO LENGTH : 25

 FIFO DATA INDEX[0] = 136
 FIFO DATA INDEX[1] = 248
 FIFO DATA INDEX[2] = 255
 FIFO DATA INDEX[3] = 253
 FIFO DATA INDEX[4] = 255
 FIFO DATA INDEX[5] = 3
 FIFO DATA INDEX[6] = 0
 FIFO DATA INDEX[7] = 68
 FIFO DATA INDEX[8] = 61
 FIFO DATA INDEX[9] = 20
 FIFO DATA INDEX[10] = 0
 FIFO DATA INDEX[11] = 128
 FIFO DATA INDEX[12] = 0
 FIFO DATA INDEX[13] = 128
 FIFO DATA INDEX[14] = 0
 FIFO DATA INDEX[15] = 128
 FIFO DATA INDEX[16] = 0
 FIFO DATA INDEX[17] = 128
 FIFO DATA INDEX[18] = 0
 FIFO DATA INDEX[19] = 128
 FIFO DATA INDEX[20] = 0
 FIFO DATA INDEX[21] = 128
 FIFO DATA INDEX[22] = 0
 FIFO DATA INDEX[23] = 128
 FIFO DATA INDEX[24] = 0
 REQUESTED GYRO DATA FRAMES : 1
 
 AVAILABLE GYRO DATA FRAMES : 1
 
FIFO GYRO FRAME[0]
GYRO X-DATA : -8         Y-DATA : -3     Z-DATA : 3
 SENSOR TIME DATA : 0 
SKIPPED FRAME COUNT : 0

BoschSensortec, 2018-04-04:
Hi @T721,

The delay is crucial for the settings to be appropriately set. The delay function linked should wait for a a minimum of the requested period.

T721, 2018-04-05:
I'm sorry @BoschSensortec 
i cannot make myself a clue of this delay and FIFO stuff.
What means a minimum of the requested period?
when i wait for, let's say 100ms with ODR 100HZ i would expect to have 10 measurements in the FIFO register. I only need one so my delay should be a minimum of 10ms, correct?
I'm curious abot that because i tried so many different setups with more or less wait times but i were not able to read out that timestamp - even not when using get_regs to jsut get the raw data stuff.

T721, 2018-04-05:
My BMI160 Init;
`void BMI160::initSensor() {
	sensor.id = 0;
	sensor.interface = BMI160_SPI_INTF;
	sensor.read = user_spi_read;
	sensor.write = user_spi_write;
	sensor.delay_ms = user_delay_ms_config;

	int8_t rslt = BMI160_OK;
	rslt = bmi160_init(&sensor);

	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_100HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_100HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;

	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&sensor);
	
	std::cout << rslt << std::endl;
}`

and the init from your example, just edited in the array sizes
`   // Declare memory to store the raw FIFO buffer information
    uint8_t fifo_buff[600];

    // Modify the FIFO buffer instance and link to the device instance
    struct bmi160_fifo_frame fifo_frame;
    fifo_frame.data = fifo_buff;
    fifo_frame.length = 7;
    dev->fifo = &fifo_frame;
    uint16_t index = 0;

    // Declare instances of the sensor data structure to store the parsed FIFO data
    struct bmi160_sensor_data gyro_data[1]; // 300 bytes / ~7bytes per frame ~ 42 data frames
    uint8_t gyro_frames_req = 1;//42;
    uint8_t gyro_index;

    // Configure the sensor's FIFO settings
    rslt = bmi160_set_fifo_config(BMI160_FIFO_G_ENABLE | BMI160_FIFO_HEAD_ENABLE | BMI160_FIFO_TIME_ENABLE, BMI160_ENABLE, dev);`

So with this config i would think that my wait time is a minimum of 10ms to retrieve a value.
i want one frame of sensordata or what i understand as a frame one pair of values gx,gy,gz + timestamp when it was recorded.

this is why i set fifo_frame.length to 7, 3 * unint16 for data + one header byte. is this correct?
i tried with 11 bytes of frame length to get the 4 bytes of time data but this did not work.
Am i misunderstanding the FIFO?

BoschSensortec, 2018-04-06:
Hi @T721,

Couple of notes,
- Before each read iteration from the FIFO,  reload the values of `gyro_frames_req` and `fifo_frame.length`.
- If you would like to read sensor time, always provide a buffer of `fifo_frame.length` + `BMI160_FIFO_BYTES_OVERREAD`.
- Since you are using the header mode, the frame contains a 1 byte header and 6 bytes of payload. In order to read the sensor time, one has to read more bytes from the FIFO than the length of the FIFO advertised. This is done automatically in the API provided you have requested for the sensor time and an adequate buffer is provided.