boschsensortec/BMI160_SensorAPI #19: BMI160 driver 3.7.2 interrupt pin issue

ytham, 2017-11-13:
We seem to have an issue with changing the levels of the interrupt pins w/ the new 3.7.2 driver. When we place the BMI160 into normal mode w/ the FIFO enabled, we are attempting to read the INT2 pin in order to time synchronize the output from the IMU FIFO. Previously, in 3.6.1, we were able to read this pin and it would be high/low depending on the state of the FIFO. Now, the pin is always reading low. 

When we reverted the driver to 3.6.1, we were able to see the INT2 pin go high/low again.

In your testing, were you guys able to read any different values from the interrupt pins while using the FIFO?

BoschSensortec, 2017-11-14:
Hi @ytham ,

Were you using the FIFO watermark interrupt?

jieliu8sc, 2017-11-14:
I am not using the FIFO watermark interrupt.

BoschSensortec, 2017-11-15:
@ytham & @jieliu8sc ,

Which interrupt is mapped to the Int2 pin?

jieliu8sc, 2017-11-15:
No interrupt is mapped to int2 pin. We are pulling int2 pin high/low, and expect frame header to contain its value at frame generation time.

ytham, 2017-11-16:
@BoschSensortec

BoschSensortec, 2017-11-17:
@ytham ,

We are trying to duplicate the scenario. Can you send your configuration?

ytham, 2017-11-17:
@jieliu8sc Can you send Bosch how you've configured the BMI160?

jieliu8sc, 2017-11-18:
These are initial settings:  
`
  mSensor.accel_cfg.odr = BMI160_ACCEL_ODR_100HZ;
  mSensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
  mSensor.accel_cfg.bw = BMI160_ACCEL_BW_OSR2_AVG2;
  mSensor.accel_cfg.power = BMI160_ACCEL_LOWPOWER_MODE;

  // Set gyro parameters
  mSensor.gyro_cfg.odr = BMI160_GYRO_ODR_100HZ;
  mSensor.gyro_cfg.range = BMI160_GYRO_RANGE_1000_DPS;
  mSensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;
  mSensor.gyro_cfg.power = BMI160_GYRO_SUSPEND_MODE;

  rslt = bmi160_set_sens_conf(&mSensor);
  APP_ERROR_CHECK_BOOL(rslt == BMI160_OK);

  struct bmi160_int_settg int2Config = {0};
  /* Select the Interrupt channel/pin */
  int2Config.int_channel = BMI160_INT_CHANNEL_2;  // Interrupt channel/pin 2
  /* Select the interrupt channel/pin settings */
  int2Config.int_pin_settg.input_en = BMI160_ENABLE;           // Enabling interrupt pin to act as input
  int2Config.int_pin_settg.latch_dur = BMI160_LATCH_DUR_NONE;  // non-latched output
  rslt = bmi160_set_int_config(&int2Config, &mSensor);
`
Before reading FIFO we do
`
  mSensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
  int8_t rslt = bmi160_set_power_mode(&mSensor);

  mSensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;
  int8_t rslt = bmi160_set_power_mode(&mSensor);

  int8_t rslt = bmi160_set_fifo_config(BMI160_FIFO_GYRO | BMI160_FIFO_ACCEL | BMI160_FIFO_HEADER | BMI160_FIFO_TIME | BMI160_FIFO_TAG_INT2, BMI160_ENABLE, &mSensor);
`

BoschSensortec, 2017-11-20:
Thanks @ytham  & @jieliu8sc ,

This should speed up the testing.

BoschSensortec, 2017-11-21:
@ytham  & @jieliu8sc ,

Note the release 3.7.3. In addition you have to add `int_config.int_type = BMI160_FIFO_TAG_INT_PIN`