boschsensortec/BMI160_SensorAPI #22: Sensortime register and sensor data synchronization

qinwang717, 2017-12-04:
Hi, I am trying to understand the synchronization between the sensor data and the sensor time register. Since I want to get the precise timestamp of the moment when data register is updated.

As it mentioned in the datasheet, All sensor events are synchronous to the sensor time counter. And "The sensor time is synchronized with the update of the data register." 

However, In the Data register, the sensor data are updated according to output data rate of each sensor. It means if, for example, accel is set to have a odr of 200Hz, then the data register of accel will update every 5ms. 

In the sensor time register, however, it keeps increasing with a 39us increment. (Page 50)

When reading the data register and the sensortime together, the current value of both of the register will be shadowed and read. 

That means, the sensor time is not indicating the time at the moment when data is captured, but the time when the read operation is performed. 

In the case of odr = 200Hz, it will be a time difference of 5ms.  

Same situation when reading data from the FIFO. 

Thus, when reading the sensor time from data register or the FIFO, I am reading the time when the read operation is occurred, not the time when new sensor data updated. Is my understanding correct?

BoschSensortec, 2017-12-04:
Hi @qinwang717,

Yes you are right. 

For the FIFO, the timestamp is shadowed when the host read finishes but not when the read of the data occurs. 

When you finish reading the last data frame from the FIFO, a sensor time frame follows indicating the current time of the device. 

Since the sensortime is synced with the update of the data register as you mentioned, one can also find out when the last frame is updated by ignoring the last few LSBs of the sensortime. See table 11.

For example, if the ODR is 200Hz, the bit 7 in sensor time is toggling when data is updating. So, when you get the sensor time from the FIFO, bits 0-6 in the sensor time actually represents the age of the last updated frame. It is updated at the time from bit 23 to bit 7. So in other words, to get when the data is updated at 200Hz using a FIFO timestamp, just ignore the bit 6 down to bit 0 in the sensor time frame.

In case you need very precise timestamps, you will need a host that is timed with a crystal clock and probably timer input capture to calculate the clock skew of the BMI160 from time to time. Note that the BMI160 has a rated deviation of timing accuracy of +-1% over the operable temperature range but is nearly a constant skew for a fixed temperature.

BoschSensortec, 2017-12-04:
@qinwang717 ,

Note that the period is actually 39.0625us which is derived from an internal clock of 25.6kHz

qinwang717, 2017-12-04:
Thanks a lot. 

The information is very helpful.

qinwang717, 2017-12-05:
Hi, just to make sure. 

Does this apply for both FIFO sensor time reading and when reading sensor time with data registers? 

When reading data from the data register among with sensor time, as it mentioned in the data sheet, the sensor time register is shadowed at the beginning of the operation, while FIFO shadowed at the end of the operation. 

That means the sensortime read with data register will be slightly more accurate to the moment when last frame updated, compare to the sensor time frame from FIFO.   

But for both of them, the method for getting the moment when last frame updated is the same - ignore 0:6 bits (when odr = 200Hz). 

Is my understanding correct? 

Many thanks.

BoschSensortec, 2017-12-05:
Hi @qinwang717,

Any read to the data registers is shadowed, so if there is new data available while you are reading, it does not corrupt the data of your current read. Hence the timestamp is associated with the start of the read from the data registers. 

On the other hand, during a FIFO read, since there is no possibility of corrupting the last data frame in the FIFO, the sensor time is related to the time instance that the overread started (similar to the shadowing of the data register read).

As you mentioned, at an ODR of 200Hz, the bits 0:6 tell you how old the sample is that you are reading since it was measured. 

In a simplified sense, 
- With a data register read you get a timestamp of when you start reading the data 'frame'
- With a FIFO read, you get a timestamp of when you start the overread. 

Solely from a precision point of view, the FIFO sensor time is more accurate as it removes the latency associated with the read (of that particular frame) and you get a time which is closer to the time of processing, i.e, the end of read. 

Then again, it depends on what you are using the data for. Typically for low latency image/video processing applications, the dedicated OIS interface over SPI is used.

qinwang717, 2017-12-05:
Thanks a lot. 
It's all clear now.

BoschSensortec, 2017-12-05:
@qinwang717 ,

Additional notes:
 1) Suggested reading : Sensor time for the [BMA456](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMA456-DS000-01.pdf)
 2) If you are using the FIFO, read the data after the FIFO watermark interrupt is fired and reach the last frame before the next sensor time is reached, else use the data registers.

BoschSensortec, 2018-10-15:
Hi @swip10,

The sensor time counter is derived from a 25.6kHz clock. Depending on the selected sampling frequency, the measurement occurs at the on a fixed slot, f.i., assuming you have set the sampling frequency to 100Hz, then a measurement is performed every 10ms (or 256 counts). Any additional counts/ticks at the time of reading correspond to the age of the sample since measured.

Regarding drift, it depends on whether the gyroscope is enabled or not. Without the gyroscope the clock has the accuracy of a typical RC oscillator and with the gyroscope it has the accuracy of a MEMS oscillator.

swip10, 2018-10-15:
Hello, 

Thank you for the answer. However, my question was all about the accuracy of the clock. 
Let's say you have a 100Hz sampling every 256 counts. I do understand it should be 10ms, but what is the error with and without the gyroscope?  What is the accuracy of a RC and MEMS oscillator?