boschsensortec/BMI160_SensorAPI #23: How to read PMU_STATUS register?

XaviLC, 2017-12-07:
Hello,

I  am using your driver version  _@version 3.6.0_ for the BMI160. 

I am facing an issue when configuring the device on Normal Mode (both for GYRO and ACC), I configure my device with these settings

```
  /* Select the Output data rate, range of accelerometer sensor */
  bmi160Sensor->bmi160Settings.accel_cfg.odr = BMI160_ACCEL_ODR_100HZ;
  bmi160Sensor->bmi160Settings.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
  bmi160Sensor->bmi160Settings.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

  /* Select the power mode of accelerometer sensor */
  bmi160Sensor->bmi160Settings.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

  /* Select the Output data rate, range of Gyroscope sensor */
  bmi160Sensor->bmi160Settings.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
  bmi160Sensor->bmi160Settings.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
  bmi160Sensor->bmi160Settings.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

  /* Select the power mode of Gyroscope sensor */
  bmi160Sensor->bmi160Settings.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;
 

  /* Set the sensor configuration */
  rslt = bmi160_set_sens_conf(&(bmi160Sensor->bmi160Settings));
```
However when reading data, the device seems to be slept and the retrieved data is zero. Therefore I would like to read the register _PMU_STATUS(0x03)_ but this register is not mapped on the driver bmi160_defs.h. How can I check them the power configuration of the device in order to know if the device is set to work in Normal Mode?

THanks in advance

BoschSensortec, 2017-12-11:
Hi @XaviLC,

What is your setup like? Are the return results (rslt values) from the APIs handled?

BoschSensortec, 2018-02-13:
@XaviLC,

Added the bmi160_get_powrer_mode API

closed 2018-02-13 by BoschSensortec