boschsensortec/BMI160_SensorAPI #24: step counter   low power consumption？  H欧文、

ouki-wang, 2017-12-20:
手册上说setp counter的低功耗是5微安，而我测出来的高达1100微安，什么原因呢？

BoschSensortec, 2017-12-20:
Hi @ouki-wang ,

As per the [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000-07.pdf) page 7, Table 1, it is mentioned that to run only the step counter, the current consumption is 20uA.

Even if you have the Gyroscope and Accelerometer enabled at the highest performance mode, you should not see an average current of more than 990uA. 

It seems that you are currently operating the sensor at the highest performance mode and not observing the average current.

Note: If you are using I2C, the pull-up resistors also consume some current. Assuming you have a pull-up resistor of 10kOhm connected to 3.3V, you would see spikes in current consumption during communication up to 330uA per resistor, that is 660uA in total!

ouki-wang, 2017-12-22:
As per the datasheet page 7, Table 1.When I set the step detector to the mode:gyro in suspend,accel in low power mode @50Hz, but not work. Could you please help me figure it out?

BoschSensortec, 2018-01-11:
Hi @ouki-wang,

Can you please share your code snippets for initializing and configuring the sensor?

Thank you.

ouki-wang, 2018-01-15:
Thanks for you reply.

This is the code snippet for initializing:

bmi160->dev=0x69
bmi160->interface=BMI160_I2C_INTF

rslt = bmi160_get_regs(BMI160_CHIP_ID_ADDR, &chip_id, 1, bmi160);
if ((rslt == BMI160_OK)&&(bmi160->chip_id==BMI160_CHIP_ID)){
  bmi160->any_sig_sel = BMI160_BOTH_ANY_SIG_MOTION_DISABLED;
  rslt = bmi160_soft_reset(bmi160);
  if (rslt==BMI160_OK){
    rslt = bmi160_set_sens_conf(bmi160);
  }                   |
}                     |
                      |
                     \|/
This is the code snippet for configuring:

bmi160_set_sens_conf::
bmi160->accelCfg.odr = BMI160_ACCEL_ODR_50HZ;
bmi160->accelCfg.range = BMI160_ACCEL_RANGE_2G;
bmi160->accelCfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

bmi160->accelCfg.power = BMI160_ACCEL_LOWPOWER_MODE;

bmi160->gyroCfg.odr = BMI160_GYRO_ODR_50HZ;
bmi160->gyroCfg.range = BMI160_GYRO_RANGE_2000_DPS;
bmi160->gyroCfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

bmi160->gyroCfg.power = BMI160_GYRO_SUSPEND_MODE;

set_accel_conf(bmi160)
set_gyro_conf(bmi160)
bmi160_set_power_mode(bmi160)

BoschSensortec, 2018-01-16:
Hi @ouki-wang,

Can you try the following snippet?

```c

	struct bmi160_dev imu;
	int8_t rslt = 0;
	/* Detect 30 steps */
	uint16_t cnt = 30;
	uint8_t first_fail = 1;
	/* Variable to store steps counted */
	uint16_t step_count = 0;
	uint8_t step_detector_count = 0;
	uint8_t step_enable = 1;
	/* Structure to configure the interrupts */
	struct bmi160_int_settg int_config;
	/* Structure to read the PMU status */
	struct bmi160_pmu_status pmu_status;

	union bmi160_int_status int_status;
	int_status.data[0] = 0;

	/* Initialize the sensor */
	rslt = bmi160_init(&imu);

	/* Select the Output data rate, range of accelerometer sensor */
	imu.accel_cfg.odr = BMI160_ACCEL_ODR_50HZ;
	imu.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	imu.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;
	/* Select the power mode of accelerometer sensor */
	imu.accel_cfg.power = BMI160_ACCEL_LOWPOWER_MODE;

	imu.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;
	imu.gyro_cfg.odr = BMI160_GYRO_ODR_50HZ;
	imu.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	imu.gyro_cfg.power = BMI160_GYRO_SUSPEND_MODE;

	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&imu);

	rslt = bmi160_get_power_mode(&pmu_status, &imu);

	if ((pmu_status.accel_pmu_status == BMI160_ACCEL_PMU_LOW_POWER) &&
		(pmu_status.gyro_pmu_status == BMI160_GYRO_PMU_SUSPEND)) {
			printf("\n Accel in LP and gyro in suspend mode\n");
	}

	/* Select the Interrupt channel/pin */
	int_config.int_channel = BMI160_INT_CHANNEL_2;	// Interrupt channel/pin 2

	/* Select the Interrupt type */
	int_config.int_type = BMI160_STEP_DETECT_INT;

	/* Select the interrupt channel/pin settings */
	int_config.int_pin_settg.output_en = BMI160_ENABLE;	// Enabling interrupt pins to act as output pin
	int_config.int_pin_settg.output_mode = BMI160_DISABLE;	// Choosing push-pull mode for interrupt pin
	int_config.int_pin_settg.output_type = BMI160_ENABLE;	// Choosing active High output
	int_config.int_pin_settg.edge_ctrl = BMI160_ENABLE;	// Choosing edge triggered output
	int_config.int_pin_settg.input_en = BMI160_DISABLE;	// Disabling interrupt pin to act as input
	int_config.int_pin_settg.latch_dur = BMI160_LATCH_DUR_NONE;	// non-latched output
	int_config.int_type_cfg.acc_step_detect_int.step_detector_mode = BMI160_STEP_DETECT_NORMAL;
	int_config.int_type_cfg.acc_step_detect_int.step_detector_en = BMI160_ENABLE;

	/* Set the interrupt configuration */
	rslt = bmi160_set_int_config(&int_config, &imu);

	/* Enable step counter */
	rslt = bmi160_set_step_counter(step_enable, &imu);

	/* Step interrupt detection and step counter data read by polling method */
	while (cnt > 0) {
		/* Read the interrupt status for step interrupt */
		rslt = bmi160_get_int_status(BMI160_INT_STATUS_0, &int_status, &imu);

		if ((rslt == BMI160_OK) && (int_status.bit.step == BMI160_ENABLE)) {
			step_detector_count++;
			bmi160_read_step_counter(&step_count, &imu);
			printf("\n Step_detected and no of step = %d - count = %d and status reg(0x1C) value is = %x\n", step_count, cnt, int_status.bit.step);
			int_status.data[0] = 0;
			cnt--;
		} else {
			if (first_fail == 1) {
				printf("Kindly shake the board in order to get step detector interrupt(normal),  "
					"step detector Interrupt not Occurred and status reg(0x1C) value is = %x\t", int_status.bit.step);
				first_fail = 0;
			}
		}
		/* Delay due to polling mode */
		imu.delay_ms(40);
	}

```

ouki-wang, 2018-01-17:
@BoschSensortec Thanks very much. follow your code, by the bmi160_get_power_mode, I confirm
 that has in low power model, then found that a resistor consumes 300mA current.

closed 2018-01-17 by ouki-wang