boschsensortec/BMI160_SensorAPI #26: Q: Linux userland compatibility (i2c)

Thannoy, 2018-01-03:
I need to use a bmi160 in a Linux project. Linux kernel integrated bmi160 driver does not support bmi160 hardware fifo (among other features), hence I was looking to use this implementation instead.

This driver relies on provided read and write functions which must read and write to i2c bus (talking about i2c case).

Linux provides a i2c-tools library for i2c userspace access. It handles i2c read operations, write operations and smbus protocol (write/write and write/read operation mix - ie select a register to write into it or to read it).

The closest binding appears to be the smbus API, except it does not handle burst read larger than 32 bytes (smbus specification) and bmi160 FIFO can be as large as 1024 byes.

Is there a very-known Linux API to use in order to fully use this driver on a Linux userland i2c link?

Regards.

Thannoy, 2018-01-03:
Well, as a note I think I found a solution in the BMI160 MasterSpec documentation, chapter 3.2.3:

It says:
> The two parts of the transmission must be separated by a repeated start condition

but it also says: 

> Once a new data read transmission starts, the start address will be set to the register address specified since the latest I2C write command. [..] In this way repetitive multi-bytes reads from the same starting address are possible.

Then it should become possible to read chunks greater than 32 bytes using two i2c requests:
- one i2c write request (which selects the register)
- one i2c read request (of any size)

The only thing to cope with is the risk of another command sent to the bmi160 between those two commands.

BoschSensortec, 2018-01-10:
Hi @Thannoy,

Can you guide us to the documentation that supports the statement

```
The closest binding appears to be the smbus API, except it does not handle burst read larger than 32 bytes (smbus specification) and bmi160 FIFO can be as large as 1024 bytes.
```

Thannoy, 2018-01-30:
Below are links to linux i2c userland API documentation:
* [/dev/i2c-dev API](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/Documentation/i2c/dev-interface)
* [smbus commands details](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/Documentation/i2c/smbus-protocol)

I can confirm that it is possible to dump FIFO greater than 32 bytes using two raw i2c calls (a write to select FIFO register, then a read to dump it). Also, I2C_RDWR ioctl should make it possible to do this a bit faster and without releasing i2c bus between those two steps.

Hence, I am sorry for the time you took on this issue, which can be marked as closed.

BoschSensortec, 2018-01-31:
Hi @Thannoy,

We are happy help. Typically these conversations serve as reference point for other developers using a similar system.