boschsensortec/BMI160_SensorAPI #29: Reading Zero for Acceleration X Y Z

SikkaTheCoder, 2018-01-20:
Hello,

First of all thank you for the great library. I am using the BMI160 with I2C using the Nordic NRF52840 microcontroller. This controller has a driver for I2C referred to as [nrf_drv_twi](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.3.0%2Ftwi_scanner_example.html&cp=4_0_3_4_5_34).

When I use the get sensor data function, I am reading zeros for acceleration. However, the sensor time register that is being read is populating with data. Thus, I am confused as to why the acceleration values are zero as I am vigorously shaking the sensor.

When I started investigating further, I noticed that when I read register 0x40 directly, I get back 0x28. This is the default configuration for the acceleration config register, however, I had changed the acceleration frequency to 1600Hz, so I would expect a C for the LSB.

This is my write implementation.

```
int8_t nrf_drv_bmi160_twi_write(uint8_t dev_addr, uint8_t reg_addr, uint8_t * p_data, uint16_t len)
{
	ret_code_t		err_code;
	uint32_t		timeout = PARAM_TWI_TIMEOUT;
	
	err_code = nrf_drv_twi_tx(&m_bmi160_twi_instance, dev_addr, &reg_addr, len, false);
	if (err_code != NRF_SUCCESS) return BMI160_E_COM_FAIL;
	
	// Wait for TWI_TX_DONE event from the event handler.
	while(!twi_tx_done && --timeout);
	if (!timeout) return BMI160_E_COM_FAIL;
	twi_tx_done = false;
	
	return err_code == NRF_SUCCESS ?  BMI160_OK : BMI160_E_COM_FAIL;
}
```

The NRF driver function for TWI transmits the register address to the slave and waits for a transmit done. I realized that I have not sent the data yet to write to the register in my implementation. How do I form the data that needs to be sent over using the nrf_drv_twi function? Do I do two write processes, one of the registry to write to, and the second with the data?

I am trying to follow this:

[image]

Any help would be appreciated. 

Thanks,

Rishi

SikkaTheCoder, 2018-01-20:
For anyone also working on something similar. I got it to work for now.

Pretty simple, I had to first create a write buffer of two bytes, where the first byte is the register address, and the second byte is the data to send.

```
int8_t nrf_drv_bmi160_twi_write(uint8_t dev_addr, uint8_t reg_addr, uint8_t * p_data, uint16_t len)
{
	ret_code_t		err_code;
	uint32_t		timeout = PARAM_TWI_TIMEOUT;
	
	uint8_t packet[2] = { 0 };
	packet[0] = reg_addr;
	packet[1] = *p_data;
	
	err_code = nrf_drv_twi_tx(&m_bmi160_twi_instance, dev_addr, packet, 2, false);
	if (err_code != NRF_SUCCESS) return BMI160_E_COM_FAIL;
	
	// Wait for TWI_TX_DONE event from the event handler.
	while(!twi_tx_done && --timeout);
	if (!timeout) return BMI160_E_COM_FAIL;
	twi_tx_done = false;
	
	return err_code == NRF_SUCCESS ?  BMI160_OK : BMI160_E_COM_FAIL;
}
```

closed 2018-01-20 by SikkaTheCoder

BoschSensortec, 2018-03-15:
@SikkaTheCoder,

We overlooked it a while ago,  but the packet should be of size ```len + 1```. Here ```*p_data``` is an array of size ```len```. The rest of the bytes need to be copied too.