boschsensortec/BMI160_SensorAPI #30: BMI160 FIFO keeps getting less and less frames & can't fill buffer

ytham, 2018-01-30:
I'm currently running into two issues on the BMI160:

1. The function bmi160_extract_accel gets less and less frames over time until the BMI160 is reset. It happens seemingly randomly. Will start at 32 and then could go down to 30, then down to 20, then down to 5, then down to 1.

2. I am unable to fill my buffer with the FIFO frames. After waiting any number of milliseconds (tried 105-800 ms), the max useful full frames I can get is 32.

**Configuration:**
FIFO config flags: BMI160_FIFO_GYRO | BMI160_FIFO_ACCEL | BMI160_FIFO_HEADER | BMI160_FIFO_TIME | BMI160_FIFO_TAG_INT2
#define FIFO_BUFFER_SIZE 1024
#define MAX_FIFO_FRAMES 42
- Each FIFO frame is 24 bytes, so we're trying to get 42 full frames.
- Accel ODR is 400Hz
- Gyro is in Suspend mode

BoschSensortec, 2018-01-31:
Hi  @ytham,

We are working on duplicating the issue and will get back to you shortly.

GiampietroPomponio, 2018-02-20:
Hi @BoschSensortec,
I've the same problem. 
Seeing the code, I've seen that the variable lenght ( (struct bmi160_dev)->(struct bmi160_fifo_frame)->lenght  ) is reduced when dev->fifo->length is grater than bytes_to_read ( into the function bmi160_get_fifo_data() ) but  the code never restored it. In your opinion, it could be the cause of this problem?

BoschSensortec, 2018-02-20:
Hi @GiampietroPomponio,

The length should be restored in the user code before calling the function. The reason for this is,

- to know in the user space, how many bytes were actually read,
- to optimize the size of the buffers accordingly in the user space depending on the application.