boschsensortec/BMI160_SensorAPI #32: FOC for gyro causes failure

TobeyWins, 2018-02-05:
Im using BMI160 on a armhf board over spi.
When i want to use FOC with the given example code the bmi160_start_foc returns -11 or BMI160_FOC_FAILURE.
FOC seems to succeed when i turn of FOC for the gyro and just use it for accelerometer.

Is this a common problem or what can i do to find out what is not working properly?

Many thanks in advance.

BoschSensortec, 2018-02-05:
Hi @TobeyWins,

Have you already checked the issue https://github.com/BoschSensortec/BMI160_driver/issues/17?

TobeyWins, 2018-02-05:
Hi @BoschSensortec,
thank you very much for your response!
As it appears I had an issue with my delay function not evaluationg the given period.
Thank you for helping me solve the issue.
Now it works as intended.

Kind regards

TobeyWins, 2018-02-05:
Hello @BoschSensortec,
Im sorry to reopen again.
But it appears as I have randomly errors because my delay time is not enough.

I don't want my sensor to delay so i want to keep the delay ms function as short as possible.
What does the sensor call when it calls my function?

I needed to increase the factor by 500 µs to get all time success at FOC.
But this will also slow down my measurements wont it?

BoschSensortec, 2018-02-05:
Hi @TobeyWins,

the API uses the typical time of 20ms to wait for the self-test results of the Gyroscope to finish. As it appears, some sensors take slightly longer to complete their self-test. 

We will patch the API with a snippet that polls from 20ms to 30ms after which only is the warning thrown that the self-test failed.

In the meantime you may modify the macro [BMI160_GYRO_SELF_TEST_DELAY](https://github.com/BoschSensortec/BMI160_driver/blob/e0a96d443e5cf9c313e1019c1293616aa923c5b6/bmi160_defs.h#L376) to 30ms so that the rest of the delays obey the typical timing.

TobeyWins, 2018-02-05:
Thanks for your quick responses @BoschSensortec,

your answer improved the behaviour of my sensor but did not completely solve it.
Even with higher delays (like 100ms) I get the problem that FOC sometimes just fails and after 2 or 3 retries just completes fine. I still have to increase the delay time to ensure that it will finish in the first try to like 1,5 or 2 times the given delay period.
I'm happy that it runs after 2 or 3 retries but this behaviour does not seem pretty deterministic to me.

I just performed a self-test myself and I get Error Code `-4 BMI160_E_OUT_OF_RANGE`

BoschSensortec, 2018-02-05:
Hi @TobeyWins,

Do you trigger a soft reset after running the self-test?

TobeyWins, 2018-02-05:
@BoschSensortec,
no, but i thaught i had to call self test just some time after bmi160_init.
I now found out that i have to call it right after bmi160_init. Now i either success for both or warning for accelerometer. 

My core problem still exists. 
I have to call the FOC routine twice to get a success for it.
I tested in on 2 different BMI160's to inspect if my sensor is just not working as intended but even with your suggestions in higher delay times i can't get the FOC successful in one-shot.

```
FOC FAILURE
retry...

 FOC DONE SUCCESSFULLY 
 OFFSET VALUES AFTER FOC : 
 OFFSET VALUES ACCEL X : 62 
 OFFSET VALUES ACCEL Y : 34 
 OFFSET VALUES ACCEL Z : 17 
 OFFSET VALUES GYRO  X : -7 
 OFFSET VALUES GYRO  Y : 1 
 OFFSET VALUES GYRO  Z : 5 

success! after: 2 tries
```

BoschSensortec, 2018-02-05:
Hi @TobeyWins,

Are the sensors firmly mounted during the course of the FOC?

TobeyWins, 2018-02-05:
Hi @BoschSensortec,
yes I have my sensors mounted on a demo circuit board.

I have written a little test now in which I call the FOC 100 times in a row.
when i increase user_delay_ms to be 2 times the selected period it works perfectly.
When i work with the 30ms which u suggested and stick on the correct factor to multiply the period value it is not deterministic wether the function succeeds or not.
Somehow after 20 or 30 tries it just succeeds and then keeps succeeding until the 100 tries are completed.
I completely do not understand why this is happening.

Im using this snippet of the provided example code:
```
	/* An example for configuring FOC for accel and gyro data */
	int8_t rslt = 0;

	/* FOC configuration structure */
	struct bmi160_foc_conf foc_conf;
	/* Structure to store the offsets */
	struct bmi160_offsets offsets;

	/* Enable FOC for accel with target values of z = 1g ; x,y as 0g */
	foc_conf.acc_off_en = BMI160_ENABLE;
	foc_conf.foc_acc_x = BMI160_FOC_ACCEL_0G;
	foc_conf.foc_acc_y = BMI160_FOC_ACCEL_0G;
	foc_conf.foc_acc_z = BMI160_FOC_ACCEL_POSITIVE_G;

	/* Enable FOC for gyro */
	foc_conf.gyro_off_en = BMI160_ENABLE;
	foc_conf.foc_gyr_en = BMI160_ENABLE;

	rslt = bmi160_start_foc(&foc_conf, &offsets, &sensor);

	if (rslt == BMI160_OK) {
		printf("\n FOC DONE SUCCESSFULLY ");
		printf("\n OFFSET VALUES AFTER FOC : ");
		printf("\n OFFSET VALUES ACCEL X : %d ", offsets.off_acc_x);
		printf("\n OFFSET VALUES ACCEL Y : %d ", offsets.off_acc_y);
		printf("\n OFFSET VALUES ACCEL Z : %d ", offsets.off_acc_z);
		printf("\n OFFSET VALUES GYRO  X : %d ", offsets.off_gyro_x);
		printf("\n OFFSET VALUES GYRO  Y : %d ", offsets.off_gyro_y);
		printf("\n OFFSET VALUES GYRO  Z : %d ", offsets.off_gyro_z);

		std::cout << std::endl;
		return 0;
	}
	else{
		std::cout << "FOC FAILURE - Error: " << (int32_t)rslt;
		std::cout << std::endl;
		return -1;
	}
```

which leads to this result:
I have printed the given period to try to find out which Macro is used to cause the delay.

In this example i used the above user_delay_ms function.
```

FOC FAILURE - Error: -11
retry...
period: 1
period: 1
period: 1
period: 1
period: 1
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 25
period: 1
period: 1
period: 1

 FOC DONE SUCCESSFULLY 
 OFFSET VALUES AFTER FOC : 
 OFFSET VALUES ACCEL X : -18 
 OFFSET VALUES ACCEL Y : -6 
 OFFSET VALUES ACCEL Z : -10 
 OFFSET VALUES GYRO  X : 8 
 OFFSET VALUES GYRO  Y : -2 
 OFFSET VALUES GYRO  Z : -6 
retry...
success! after: 27 tries
```

BoschSensortec, 2018-02-14:
Hi @TobeyWins,

it seems that inserting a delay of about a 100ms before calling the bmi160_start_foc function causes the FOC to pass everytime. Can you verify the same?

TobeyWins, 2018-02-14:
Hi @BoschSensortec, thanks for your help.
Sadly i experience the complete opposite.
I figured out a way now where i call the foc twice in a row when it has never run. 
The first time crashes but after that it allways succeeds and does not need more than one try. 
Here is my code snippet for this which works quite good even though it is a bit of an ugly solution.

I can live with that but I wonder if somebody is experiencing the same error as me and what I do wrong using the FOC.

```
int32_t BMI160::foc() {
	/* An example for configuring FOC for accel and gyro data */
	int8_t rslt = 0;

	/* FOC configuration structure */
	struct bmi160_foc_conf foc_conf;
	/* Structure to store the offsets */
	struct bmi160_offsets offsets;

	/* Enable FOC for accel with target values of z = 1g ; x,y as 0g */
	foc_conf.acc_off_en = BMI160_ENABLE;
	foc_conf.foc_acc_x = BMI160_FOC_ACCEL_0G;
	foc_conf.foc_acc_y = BMI160_FOC_ACCEL_NEGATIVE_G;
	foc_conf.foc_acc_z = BMI160_FOC_ACCEL_0G;

	/* Enable FOC for gyro */
	foc_conf.gyro_off_en = BMI160_ENABLE;
	foc_conf.foc_gyr_en = BMI160_ENABLE;

	rslt = bmi160_start_foc(&foc_conf, &offsets, &sensor);
	int32_t tries = 1;

	while (tries < 3) {
		if (rslt == BMI160_OK)
		{
                        //some printing of the FOC values
			return 0;
		}
		else if (rslt != BMI160_OK && tries < 3)
		{
                        //retry if crashed
			tries++;
			rslt = bmi160_start_foc(&foc_conf, &offsets, &sensor);
		}
		else
		{
                           //some error cout
		}
	}
	return -1;
}
```