boschsensortec/BMI160_SensorAPI #34: Weak performance when getting sensor data

TobeyWins, 2018-02-14:
Hello @BoschSensortec,

I have traced the driver from you with the Linux ftrace tool to find out, how long it takes to get a acceleration + gyro value.
I'm using BMI160 over SPI and my SPI read/write (ioctl) -routine takes about 500µs to complete.
But the data from the senor is first available after 1.4 - 2.8 ms.
This is too slow for my application.
When I read the sensor directly from the registers over SPI im much faster (below 1ms) but i don't really want to handle all the bit/register setting stuff myself.
What does the driver do internally that slows it down?

Thanks in advance,
Tobias

BoschSensortec, 2018-02-14:
Hi @TobeyWins,

In the source code, you can observe that we have checks for different cases before we read and process the data. Depending on your processor speed, the total duration of the call will vary. 

If this latency is not suitable, then you will have to read and process the data directly using the source as reference.

TobeyWins, 2018-02-15:
Hi @BoschSensortec,

thanks for your answer.
I might need to implement myself but i got one more question.
In your code on many places you delay for 1ms.
In the data sheet it is specified, that the sensor needs 2us delay after writes when in normal power mode.
I just lowered the delay_ms for the case that the driver requests 1ms to 50us and the sensor works fine but much faster and can fulfill my requirement.
Are there any problems i can run in with doing this?

BoschSensortec, 2018-02-15:
@TobeyWins,
It currently set to 1ms as this is the lowest resolution for the given function millisecond delay callback. In the future we might change this to a microsecond delay in order to acquire a better resolution.
As a workaround, when you are in 'data acquisition mode'  you can switch the delay function pointer with a function that only pauses for a smaller duration or none at all - basically an empty callback. When you enter back into the configuration section, then make sure the appropriate function pointer is swapped into place.

TobeyWins, 2018-02-15:
Hi @BoschSensortec,

thanks you again for that answer.
I will go with the swapping solution, as this works fine for me.

closed 2018-02-15 by TobeyWins