boschsensortec/BMI160_SensorAPI #35: What are the units of the accel and gyro data from bmi160_get_sensor_data()?

trixr4kdz, 2018-03-07:
Hi,

I was able to get some data from the IMU sensor but the numbers I get from a stationary position looks something like this:
```
(697, -1127, 17119) 3132        <-- accel: (x, y, z) with sensortime
(2, -18, -52) 3132              <-- gyro: (x, y, z) with sensortime
```
With trying to eyeball the data, I see that the accel x, y values vary between [-1000, 1000] approximately and accel z value averaging to around 17100. The gyro data for each of the x, y, z values seem to vary between [-50, 50].

I was wondering if there's some scale factor I should think about to actually be able to use these readings.

BoschSensortec, 2018-03-08:
Hi @trixr4kdz,

Depending on the range set,
For the accelerometer, in order to get the data in the Gs (Gravity - 1G is 1 times earth's gravity ~ 9.81m/s^2)
```c
/* For range 4G, 
   = 2^16 / (2 * 4) - Multiply by 2 to account for the +4G and -4G
   = 65536 / 8
   = 8192
*/
#define G_TO_LSB (8192.0f)

float accelX = ((float)accel.x) / G_TO_LSB;
```
For the gyroscope, in order to get the data in the dps (Degrees per second)
```c
/* For range 2000dps, 
   = 2^16 / (2 * 2000) - Multiply by 2 to account for the +2000dps and -2000dps
   = 65536 / 4000
   = 16.384
*/
#define DPS_TO_LSB (16.384f)

float gyrX = ((float)gyr.x) / DPS_TO_LSB;
```

The default for the accelerometer is 2G and gyroscope is 2000dps and sensor time is 39.0625us per LSB.
```
Your data should be
A: (0.0425, -0.0688, 1.0449) @ 122343.75us  <-- accel: (x, y, z) in Gs
G: (0.1221, -1.0986, -3.1738) @ 122343.75us  <-- gyro: (x, y, z) in dps
```

trixr4kdz, 2018-03-09:
Thank you! That definitely makes more sense! 😃 

However, the numbers I'm getting are slightly off when I try to convert the accel to m/s^2. It seems that the x accel is mostly between [-0.5, -0.1], y accel ranges between [0.5, 1.5] and z accel is greater than 10.0 when not moving. Is there a way for me to calibrate these numbers?

I'm also trying to integrate a BMM150 into it. What would be the units for the data I get from the BMM150?

BoschSensortec, 2018-03-09:
Hi @trixr4kdz,

Calibration of sensors is quite a complex topic to cover within a comment, we typically provide our customers with a fusion software like the BSX Fusion library. This library includes these calibrators and also fuses it to provide orientation data in 9 axes. You may use the readily available version called BSXLite with a limited feature set [here](https://www.bosch-sensortec.com/en/bst/products/all_products/bsxlite). 
There have been users that use the BSXLite library to receive calibration information at 50Hz and then apply it to the data stream at the desired rate. 

In case you need something simple, then at a static position subtract/add the offsets observed to achieve the desired results.

The units of the BMM150 are in micro Tesla.

lbegani, 2018-03-20:
I need to fill ROS IMU message (http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html) with the output from BMI160 sensor(read from bmi160_get_sensor_data() ).
What kind of conversions do I need to use? Accelerations should be in m/s^2 (not in g's), and rotational velocity should be in rad/sec

BoschSensortec, 2018-03-20:
Hi @lbegani,

 Based on [standard gravity](https://en.wikipedia.org/wiki/Gravity_of_Earth) definitions, acceleration should be multiplied by ```9.80665``` and rotational velocity by ```0.0174532925``` (pi / 180).

lbegani, 2018-03-20:
Is this correct ?
```
float accelX = (((float)accel.x) / G_TO_LSB) * 9.80655; // in m/s^2
float gyrX = (((float)gyr.x) / DPS_TO_LSB) * 0.0174532925; // in rad/sec
```

BoschSensortec, 2018-03-20:
@lbegani,

Yes that should work provided you are using the right constants for the ```G_TO_LSB``` and ```DPS_TO_LSB``` macros.

lbegani, 2018-03-20:
Thanks. Below is the snippet

```
    /* Select the Output data rate, range of accelerometer sensor */
    mSensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
    mSensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
    mSensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

    /* Select the power mode of accelerometer sensor */
    mSensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

    /* Select the Output data rate, range of Gyroscope sensor */
    mSensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
    mSensor.gyro_cfg.range = BMI160_GYRO_RANGE_250_DPS;
    mSensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

    /* Select the power mode of Gyroscope sensor */
    mSensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;
```

```
/* For range 2G, 
   = 2^16 / (2 * 2) - Multiply by 2 to account for the +2G and -2G
   = 65536 / 4
   = 16384
*/
#define G_TO_LSB (16384.0f)

/* For range 250dps, 
   = 2^16 / (2 * 250) - Multiply by 2 to account for the +250dps and -250dps
   = 65536 / 500
   = 131.072
*/
#define DPS_TO_LSB (131.072f)
```

```
    value.accel[0] = (((double)accel.x) / G_TO_LSB) * 9.80655;
    value.accel[1] = (((double)accel.y) / G_TO_LSB) * 9.80655;
    value.accel[2] = (((double)accel.z) / G_TO_LSB) * 9.80655;

    value.angrate[0] = (((double)gyro.x) / DPS_TO_LSB) * 0.0174532925;
    value.angrate[1] = (((double)gyro.y) / DPS_TO_LSB) * 0.0174532925;
    value.angrate[2] = (((double)gyro.z) / DPS_TO_LSB) * 0.0174532925;
```

lbegani, 2018-03-21:
```
	/*! sensor time */
	uint32_t sensortime;
```
What is the unit of time?

BoschSensortec, 2018-03-21:
Hi @lbegani,

1 LSB = 39.0625us.

lbegani, 2018-05-31:
@BoschSensortec Is there a way to convert the `sensortime` to system time (ros::time to be more specific)?

BoschSensortec, 2018-06-01:
Hi @lbegani,

Can you describe your goal regarding synchronizing the clocks? There are features that may be useful to achieving this goal.

lbegani, 2018-06-01:
I need to synchronize data from IMU and Camera and feed the synchronized data to VIO algorithm