boschsensortec/BMI160_SensorAPI #36: bmi160_get_fifo_data() Missing Frames

mjanke67, 2018-03-09:
Hi,

We are setting up continuous read of the FIFO and reading after 100 mSec. To do this we are setting up acquisition using (result checks, etc removed for brevity):

`   
struct bmi160_dev m_sensor;

m_sensor.id = 0;
m_sensor.interface = BMI160_SPI_INTF;
m_sensor.read = bmi160_spiReadRegister;
m_sensor.write = bmi160_spiWriteRegister;
m_sensor.delay_ms = bmi160_spiDelayMs;
 
bmi160_init(&m_sensor);

m_sensor.accel_cfg.odr = BMI160_ACCEL_ODR_800HZ;
m_sensor.accel_cfg.range = BMI160_ACCEL_RANGE_16G;
m_sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

m_sensor.gyro_cfg.power = BMI160_GYRO_SUSPEND_MODE;
m_sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
m_sensor.aux_cfg.aux_sensor_enable = 0; 
    
bmi160_set_sens_conf(&m_sensor);
    
memset(&m_fifo_frame, 0, sizeof(bmi160_fifo_frame)); 
m_fifo_frame.data = m_fifo_buff;
m_fifo_frame.length = 7 * 100; // 100 frames at 7 bytes each.
m_sensor.fifo = &m_fifo_frame;

bmi160_set_fifo_config(BMI160_FIFO_ACCEL | BMI160_FIFO_HEADER,	BMI160_ENABLE, &m_sensor);
`

And then we loop while executing (while using printfs to show the number of frames read and mSec between frame reads - code highlights for brevity):

struct bmi160_sensor_data accel_data[100]; // ~7 bytes per frame
uint8_t accel_frames_req = 100;
uint8_t accel_index;
m_sensor.delay_ms(100);
bmi160_get_fifo_data(&m_sensor);
bmi160_extract_accel(accel_data, &accel_frames_req, &m_sensor);

We are noticing that we can read 81 samples with loop frequency of 102-103 mSec (100 mSec delay, plus time for the bmi160_get_fifo_data() call, printf() calls, e.t.c.). At ODR of 800 Hz, we would expect to see 82 or 83 samples in over 102 or 103 mSec. It appears that we are losing 1-2 samples per loop. We expect that reading the FIFO might take 1-2 mSec and we are wondering if the FIFO fill pauses during FIFO reads? We did not expect this limitation. Perhaps there Is a special technique required to ensure that FIFO fill continues at a rate of 800 Hz even when periodic FIFO reads are occuring (at a period of just over 100 mSec in our case)?

Any help is greatly appreciate.

Mark J

delay_ms(100);

BoschSensortec, 2018-03-09:
@mjanke67,

Could you check the value ```m_fifo_frame.length``` after reading the FIFO data? This member should hold the number of bytes read from the FIFO.

Also, under typical scenarios, one would use the  FIFO watermark interrupt set at around 700 bytes in order to have the sensor data in sync.

mjanke67, 2018-03-09:
The m_sensor.fifo->length always matches the number of frames read. So if we read 81 frames, the length in bytes is 81 * 7 or 567 bytes. No frames are being marked as skipped.

We could use an interrupt to initiate the call to read the FIFO, but would expect that the behaviour would be the same (i.e. only what triggers the read would change).

I think the fundamental question is whether reading from the FIFO prevents the FIFO from being filled (i.e. while the read occurs).

BoschSensortec, 2018-03-09:
Hi @mjanke67,

The filling of the FIFO is not stalled while a read is in progress. However, it has to be noted that the FIFO length register is updated after the FIFO is filled with a new sample. It is possible that the length read before reading out the FIFO was only 567 and the rest of the bytes were filled in during the read.

The FIFO read API first reads the number of bytes in the FIFO and a few more depending on whether the sensor time was requested. These additional bytes are defaulted to 25 bytes as observed here [```BMI160_FIFO_BYTES_OVERREAD```](https://github.com/BoschSensortec/BMI160_driver/blob/44f0df94d6b000a821afd7ed31c432bc677cd723/bmi160_defs.h#L498). You can override this by defining a custom length in your C flags. For example ```-DBMI160_FIFO_BYTES_OVERREAD=40```.

mjanke67, 2018-03-09:
Hi,

I understand your comment and it matches what we are seeing. We can always overread as your suggest, but how do we know how many Frames are actually in the buffer after the read is completed?

Should we clear the buffer (fill it with 0x00) before each read and then look for non-zero byte values to determine how many frames have been read? This should be fine as long as the header within each frame is guaranteed to be non-zero. We cannot rely on the data to be non-zero as it is possible (however unlikely) that x, y and z data are all zero in a frame.

Thank you for your quick responses. You have been very helpful.

Mark J

BoschSensortec, 2018-03-12:
Hi @mjanke67,

The FIFO parser APIs like ```bmi160_extract_accel(accel_data, &accel_frames_req, &m_sensor);``` can be used to extract how much of valid frames are available in the read FIFO. The variable ```accel_frames_req``` after the call of the API contains the number of valid frames.

Regarding filling the FIFO with zeroes, there is possibly no impact to the current parsing code in place due to the fact that parsing stops at the FIFO length that was read before the FIFO readout.

mjanke67, 2018-03-12:
Hi,

The accel_frames_req shows the number frames matching the raw buffer length (567 bytes in our case correlates to 81 frames) - it does not show a value that we would expect at the end of a 103 mSec read process (82 or 83 frames at least). When we look at the raw data, bytes after index 566 are all "0x00. We are worried that the frame 82 data that should be read in a 103 mSec FIFO read period is lost. If we read every 103 mSec, both the raw buffer length (m_sensor.fifo->length) shows 567 and accel_frames_req shows 81 when we expect a raw buffer of 574 and accel_frames_req of 82.

BoschSensortec, 2018-03-12:
Hi @mjanke67,

As mentioned earlier, the FIFO read API will not read more than the bytes it saw in the FIFO length register even though bytes were filled during the read. In order to read the remaining data in the FIFO, you will have to read it twice consecutively. Note that you have parse and save the frames before the next consecutive FIFO read.

mjanke67, 2018-03-12:
Hi,

We are reading every 103 mSec continuously and reading only 81 frames each time. After 100 cycles (time 10300 mSec) we would expect to see 10300 mSec * 0.8 or 8240 frames (ODR = 800), but we are only seeing 8100 frames. We are missing 1 or more frames per cycle and these frames are not accessible on subsequent cycle reads. Because frames come in every 1.25 mSec and we believe that it is taking just over 1.25 mSec to read, we are worried that reading from the FIFO causes it to stop being filled (my original question). 

If sample data added to the FIFO during a read was available during the next read, we would expect to read 81 samples during the first FIFO read (t=103 mSec) and then 82 samples every other read (t=206 mSec, t=309 mSec, etc), but we are not. We always see 81 samples every read (m_sensor.fifo->length = 567 and accel_frames_req = 81).

Thanks,

Mark J

mjanke67, 2018-03-13:
Hi,

We have tried some additional approaches.

We tried 2 successive bmi160_get_fifo_data() calls. The first call was made to read the buffer contents (approx 80 frames) and the second was used to quickly read any frames that were received during the first call. The second returned quickly as there were only 1-2 samples to read. But, we were still reading fewer samples than expected over a time span. 

Next, we tried to "overread" the raw buffer. Because the DBMI160_FIFO_BYTES_OVERREAD value only applies to overreading timing info, it could not be used. As an alternative, we disabled the reset of  dev->fifo->length to bytes_to_read within the bmi160_get_fifo_data() implementation within bmi160.c (we commented out the lines below).

if (dev->fifo->length > bytes_to_read) {
				/* Handling the case where user requests
				more data than available in FIFO */
				dev->fifo->length = bytes_to_read;
}

... and we were able to read 1-2 ADDITIONAL samples than were returned from get_fifo_byte_counter(). For example, if we specified a FIFO dev->fifo->length of 600 when there were only 560 bytes in the FIFO (returned from get_fifo_byte_counter()), we saw valid data up to 574 bytes (2 more samples read from using dev->read() than indicated by get_fifo_byte_counter()). Passing the 574 bytes to bmi160_extract_accel() yielded 82 valid frames. This indicated that we could read FIFO data that entered the FIFO after the call to bmi160_get_fifo_data() started. This was positive. Unfortunately, when we overread the FIFO, the next cycle yielded fewer samples within the FIFO within the call to bmi160_get_fifo_data().  get_fifo_byte_counter() returned 1 sample less than expected. It had been 100 mSec since the last overread, but only 553 bytes were in the FIFO. 

Last - we tried to read when servicing a Watermark interrupt instead of a polled read. 

All approaches yielded similar results. We seem to be unable to read 800 frames a second at ODR 800. We can only read approx 790 frames. We are reading on a period of just over 100 mSec (delay_ms(100)).

Can Bosch provide a simple example that iteratively reads Accell or Gyro data with no missing frames?

Thanks in advance.

Mark J

mjanke67, 2018-03-14:
Hi,

Have we lost support on this issue? We have been spending days trying to get data acq from the BMI160 to work without good results. Any help is appreciated.

Should we get in touch with a Bosch FAE through our reseller?

Mark J

BoschSensortec, 2018-03-15:
Hi @mjanke67,

We are working on duplicating the issue but haven't been able to see the observations reported by you. We would recommend getting in touch with one of our FAEs to help look more closely at your setup. 

In the meantime, could you check 
- enabling the gyroscope even though you don't read it's data over the FIFO?
- the data ready interrupt's period?

mjanke67, 2018-03-15:
Hi,

Thank you for the update. We have made progress in figuring out the issue. 

We enabled BMI160 Sensor Time and identified that we are seeing the correct number of frames for 39 uSec ticks within the Sensor Time values stored at the end of each FIFO at FIFO read time. We will now proceed to check our Linux time and compare it to the Sensor Time. We strongly suspect that the BMI160 Sensor Time is correct and the problem is with the Linux time function implementation on the TI AM335x host that we use on our product.

One option is to use a scope on the SPI lines to identify FIFO read cycle times and see if they align with the Linux clock or the BMI160 clock. A faster approach (we think) is to simply measure host processor ticks instead of clock uSec values. 

I will update this thread when we have identified the problem with our time measurement.

Thanks,

Mark J

mjanke67, 2018-03-15:
Hi,

We have altered our Linux time calcs and we are seeing Linux FIFO acq times that are close to the BMI160 Sensor Time values. Linux time is not very accurate (relatively speaking) and this has been a challenge, but we believe that we have confirmed the BMI160 clock is accurate.

I think that we can close off this issue.

Thank you for your responses to our questions.

Mark J

closed 2018-03-15 by mjanke67