boschsensortec/BMI160_SensorAPI #37: BMI160 initialize

scratliner, 2018-03-14:
I have a Problem, to become realistic parameters off my BMI160. It seems, that bei BMI is reseted by something, because my Debug count of my X Y Z axxis is jump between 000 ans 255. Sometimes i become a value, if i move the BMI.

Here is my Init Code:

Could You please help me?

/* ------------------------------------------------------------------------------------------------
 * BMI160-sensor state machine...
 */
void BMI160_stm( void)
{
	#ifdef __SEND_XYZ_ERR_STS
	uint8_t  lu8_cnt;
	uint8_t  lu8_byte;
	uint8_t  lu8_char[2];
	#endif
	
	static enum e_BMI160_states lse_BMI160_state = BMI160_INIT;
	static enum e_BMI160_init_states lse_BMI160_init_state = BMI160_INIT_ACC;
	static enum e_BMI160_read_states lse_BMI160_read_state = BMI160_READ_ACC_START;
	static enum e_BMI160_power_change_states lse_BMI160_power_change_state = BMI160_PWR_CHG_ACC;

	#ifdef __ACC_SELF_TEST
	uint8_t  lu8_cnt;
	static uint8_t	 su8_tx_data[I2C_MX_RX_BYTES];
	static uint8_t  lsu8_i2c_1st_chk_data[I2C_MX_RX_BYTES];
	static uint8_t  lsu8_i2c_2nd_chk_data[I2C_MX_RX_BYTES];
	#endif

	#ifdef __SEND_XYZ_ERR_STS
	static uint8_t  lsu8_acc_data[I2C_MX_RX_BYTES];
	static uint8_t  lsu8_gyr_data[I2C_MX_RX_BYTES];
	static uint8_t  lsu8_status_data[2];
	#endif

	#ifdef __DEBUG_USART
	uint8_t  lu8_debug_cnt = 0;
	#endif

	#ifdef __DEBUG_USART
	#ifdef __DEBUG_THIS_HERE
	gu8_debug_tx_buffer[lu8_debug_cnt++] = 0xFA;
	#endif
	#endif

	switch( lse_BMI160_state)
	{	
		case BMI160_INIT:
			switch ( lse_BMI160_init_state)
			{
				case BMI160_INIT_ACC:
					ss_wr_packet.data_length = 2;
					su8_i2c_wr_data[0] = 0x7E;	// command register address
					su8_i2c_wr_data[1] = 0x11;	// command value -> accelerometer in normal mode, 0x11 	ACCEL_MODE_NORMAL, 0x12 ACCEL_LOWPOWER, 0x10 ACCEL_SUSPEND
					lse_BMI160_init_state = BMI160_INIT_ACC_CuR;
					gvu8_i2c_delay = 100;
					break; // eof BMI160_INIT_ACC

				case BMI160_INIT_ACC_CuR:
				
					#ifdef __ACC_SELF_TEST
					ss_wr_packet.data_length = 4;
					#else
					ss_wr_packet.data_length = 3;
					#endif
					su8_i2c_wr_data[0] = 0x40;	// register address
					su8_i2c_wr_data[1] = 0x28;	// register value -> 0x28 is default for ACC_CONFIG
					su8_i2c_wr_data[2] = 0x03;	// register value -> 0x03 is default for ACC_RANGE
					#ifdef __ACC_SELF_TEST
					su8_i2c_wr_data[2] = 0x2C;	// register value -> 0x2C must be set for g-range 8, otherweise self test will not function correctly					
[... 393 lines cut ...]

	} // eof switch( lse_BMI160_state)
	
	#ifdef __DEBUG_USART
	#ifndef __DEBUG_THIS_HERE
	//gu8_debug_tx_buffer[lu8_debug_cnt++] = 0xFA;
	gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(su8_i2c_rd_data[0]); // gyr_x_7_0
	//gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(su8_i2c_rd_data[1]); // gyr_x_15_8
	gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(su8_i2c_rd_data[2]); // gyr_y_7_0
	//gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(su8_i2c_rd_data[3]); // gyr_y_15_8
	gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(su8_i2c_rd_data[4]); // gyr_z_7_0
	//gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(su8_i2c_rd_data[5]); // gyr_z_15_8
	
	//gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(lse_BMI160_state);
	//gu8_debug_tx_buffer[lu8_debug_cnt++] = (uint8_t)(lse_BMI160_init_state);
	
	
	usart_write_buffer_job( &usart_debug, gu8_debug_tx_buffer, lu8_debug_cnt);

	#endif
	#endif

	return;

} // eof void BMI160_stm( void)

BoschSensortec, 2018-03-14:
Hi @scratliner,

This is an issue tracker for the sensor API of the bmi160. It seems that you are not using the BMI160 sensor API. Kindly use our contact form for support from one of our FAEs.

scratliner, 2018-03-14:
ok thanks.
now i will use the API from the BMI160. 
but what does it meens in the init:

	sensor.read = user_i2c_read;
	sensor.write = user_i2c_write;

BoschSensortec, 2018-03-14:
Hi @scratliner,

They are function pointers to the read/write callbacks. Similar for delay_ms too.

```c
int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
    ss_wr_packet.address = dev_id; // Store the sensor's i2c address in dev_id of struct bme680_dev
    ss_wr_packet.data_length = 1;
    ss_wr_packet.data = &reg_addr;

    if(i2c_master_write_packet_wait(&i2c_BMI160_master_instance, &ss_wr_packet) != STATUS_OK)
        return -1;

    ss_wr_packet.data_length = length;
    ss_wr_packet.data = reg_data;

    if(i2c_master_read_packet_wait(&i2c_BMI160_master_instance, &ss_wr_packet) != STATUS_OK)
        return -1;

    return 0;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
    uint16_t i = 0;
    uint8_t data[length + 1];

    data[0] = reg_addr;
    for(i = 0; i < length; i++)
        data[i+1] = reg_data[i];
   
    ss_wr_packet.address = dev_id; // Store the sensor's i2c address in dev_id of struct bme680_dev
    ss_wr_packet.data_length = length + 1;
    ss_wr_packet.data = data;

    if(i2c_master_write_packet_wait(&i2c_BMI160_master_instance, &ss_wr_packet) != STATUS_OK)
        return -1;
  
    return 0;
}
```

scratliner, 2018-03-14:
ok thanks a lot. 
this works.
but now i have a problem in my main(), because i get a return value from the bmi160_get_sensor_data(BMI160_GYRO_SEL, NULL, &gyro, &sensor)
of 0xFE; I Think this is the return failure "rslt = BMI160_E_NULL_PTR;"

void i2c_BMI160_init_new ( void)
{	

	
	sensor.id = BMI160_I2C_ADDR;
	sensor.interface = BMI160_I2C_INTF;
	sensor.read = user_i2c_read;
	sensor.write = user_i2c_write;
	//sensor.delay_ms = user_delay_ms;
	
	
	rslt = bmi160_init(&sensor);
	/* After the above function call, accel and gyro parameters in the device structure
	are set with default values, found in the datasheet of the sensor */

	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;

	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&sensor);

	return;
}

void BMI160_read ( void)
{
	//int8_t rslt = BMI160_OK;
	struct bmi160_sensor_data accel;
	struct bmi160_sensor_data gyro;

	/* To read only Accel data */
	//rslt = bmi160_get_sensor_data(BMI160_ACCEL_SEL, &accel, NULL, &sensor);

	/* To read only Gyro data */
	rslt = bmi160_get_sensor_data(BMI160_GYRO_SEL, NULL, &gyro, &sensor);

	/* To read both Accel and Gyro data */
	//bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL), &accel, &gyro, &sensor);

	/* To read Accel data along with time */
	//rslt = bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_TIME_SEL) , &accel, NULL, &sensor);

	/* To read Gyro data along with time */
	//rslt = bmi160_get_sensor_data((BMI160_GYRO_SEL | BMI160_TIME_SEL), NULL, &gyro, &sensor);

	/* To read both Accel and Gyro data along with time*/
	//bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL | BMI160_TIME_SEL), &accel, &gyro, &sensor);
return;
}

BoschSensortec, 2018-03-14:
@scratliner,

You need to assign the delay_ms function pointer too.

```c
void user_delay_ms(uint32_t period)
{
    // Call another function to wait until 'period' amount of milliseconds have passed
}
```

scratliner, 2018-03-14:
my main() is doing a function for my milliseconds, because my tc counter gives me an interrupt, if 1 ms is passed.
i want "only" to give me the x,y,z coordination about the I2C to my µC atsamd20e16 and then over the uart to my debug/monitoring.
sorry if i dont understand this sensor :D :-/

BoschSensortec, 2018-03-15:
@scratliner,

Since you already have a ms counter. Wait until period amount of ticks have passed in the ```user_delay_ms``` function. This is crucial for sensor configuration.

scratliner, 2018-04-16:
Hey,
after our discussion from the last time, iam become only 0 datas from the BMI160, if i want to read the gyro datas.
I have read, that i could read out the error states from the sensor, but in my code the IO-Pin "TESTPULS" is not coming and this is not high.
If i want to do a selft test accel, then i become a error "BMI_E_COM_FAIL" as return.

Could you please help me, why i only become "0" in my read function?

void self_test_accel()
{
	uint8_t reg_addr = 0x40;
	uint8_t data = 0x2C;
	uint16_t len = 1;
	
	rslt = bmi160_set_regs(reg_addr, &data, len, &sensor); 
	rslt = bmi160_perform_self_test(BMI160_GYRO_ONLY, &sensor);
	
	if (rslt == BMI160_E_NULL_PTR)
	{
		TESTPULS;
	}
	else if (rslt == BMI160_E_COM_FAIL)
	{
		TESTPULS;TESTPULS; // rslt = bmi160_perform_self_test(BMI160_ACCEL_ONLY, &sensor); -> bmi160_set_regs -> its not ok 
	}
	else if (rslt == BMI160_E_INVALID_INPUT)
	{
		TESTPULS;TESTPULS;TESTPULS;
	}
	else if (rslt == BMI160_W_GYRO_SELF_TEST_FAIL)
	{
		TESTPULS;TESTPULS;TESTPULS;TESTPULS;
	}
}
/* ------------------------------------------------------------------------------------------------
 * BMI160-sensor I2C new initialization...
 */
int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{	
	ss_wr_packet.address = dev_id; // Store the sensor's i2c address in dev_id of struct bme680_dev
	ss_wr_packet.data_length = 1;
	ss_wr_packet.data = &reg_addr;
	
	if(i2c_master_write_packet_wait(&i2c_BMI160_master_instance, &ss_wr_packet) != STATUS_OK)
	return -1;
	
	ss_wr_packet.data_length = length;
	ss_wr_packet.data = reg_data;

	if(i2c_master_read_packet_wait(&i2c_BMI160_master_instance, &ss_rd_packet) != STATUS_OK)
	return -1;

	return 0;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	uint16_t i = 0;
	uint8_t data[length + 1];

	data[0] = reg_addr;
	for(i = 0; i < length; i++)
	data[i+1] = reg_data[i];
		
	ss_wr_packet.address = dev_id; // Store the sensor's i2c address in dev_id of struct bme680_dev
	ss_wr_packet.data_length = length + 1;
	ss_wr_packet.data = data;

	if(i2c_master_write_packet_wait(&i2c_BMI160_master_instance, &ss_wr_packet) != STATUS_OK)
	return -1;
		
	return 0;
}
	
void user_delay_ms(uint32_t period)
{
	delay_ms(100);
		
#ifdef __soso
	static uint16_t lsu16_cnt = 0u;

	/* count ms for system timer 
		*/
	if ( lsu16_cnt < 30)
	{
		lsu16_cnt++;
	}
	else
	{
		//gvb_timer_tic = true;
		lsu16_cnt = 0u;
	}
#endif // eof #ifdef __soso	
}
	
/* ------------------------------------------------------------------------------------------------
 * BMI160-sensor initialization...
 */	
void i2c_BMI160_init_new ( void)
{	
	sensor.id = BMI160_I2C_ADDR;
	sensor.interface = BMI160_I2C_INTF;
	sensor.read = user_i2c_read;
	sensor.write = user_i2c_write;
	sensor.delay_ms = user_delay_ms;
	//sensor.delay_ms = delay_ms(30);
	
	#ifndef __BMI160_READ_ERROR_FLAG
	uint8_t data = 0;
	
	/* Read the error reg */
	rslt = bmi160_get_regs(BMI160_ERROR_REG_ADDR, &data, 1, &sensor);

	if (rslt == BMI160_OK) {
		data = data >> 1;
		data = data & BMI160_ERR_REG_MASK;
		if (data == 1)
		{
			rslt = BMI160_E_ACCEL_ODR_BW_INVALID;	
			TESTPULS; -> this is not comming
		}
		else if (data == 2)
		{
			rslt = BMI160_E_GYRO_ODR_BW_INVALID;
			TESTPULS;TESTPULS; -> this is not comming
		}
		else if (data == 3)
		{
			rslt = BMI160_E_LWP_PRE_FLTR_INT_INVALID;
			TESTPULS;TESTPULS;TESTPULS; -> this is not comming
		}
		else if (data == 7)
		{
			rslt = BMI160_E_LWP_PRE_FLTR_INVALID;
			TESTPULS;TESTPULS;TESTPULS;TESTPULS; -> this is not comming
		}
	} // eof if (rslt == BMI160_OK)
	#endif // eof #ifdef __BMI160_READ_ERROR_FLAG

	rslt = bmi160_init(&sensor);
	/* After the above function call, accel and gyro parameters in the device structure
	are set with default values, found in the datasheet of the sensor */

} // eof void i2c_BMI160_init_new ( void)

void BMI160_conf_accel_and_gyro ( void)
{
	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_8G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;
	
	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
	
	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;
	
	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&sensor);
	
	bmi160_soft_reset( &sensor);
	
	gvb_i2c_done = true;
	
	return;

} // eof void BMI160_conf_accel_and_gyro ( void)

BoschSensortec, 2018-04-16:
Hi @scratliner,

It seems the read function has a bug,

```c

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{	
	ss_wr_packet.address = dev_id; // Store the sensor's i2c address in dev_id of struct bme680_dev
	ss_wr_packet.data_length = 1;
	ss_wr_packet.data = &reg_addr;
	
	if(i2c_master_write_packet_wait(&i2c_BMI160_master_instance, &ss_wr_packet) != STATUS_OK)
	return -1;
	
	//ss_wr_packet.data_length = length;
	//ss_wr_packet.data = reg_data;
	ss_rd_packet.address = dev_id;
	ss_rd_packet.data_length = length;
	ss_rd_packet.data = reg_data;  

	if(i2c_master_read_packet_wait(&i2c_BMI160_master_instance, &ss_rd_packet) != STATUS_OK)
	return -1;

	return 0;
}
```