boschsensortec/BMI160_SensorAPI #38: Interfacing bgm11s with bhi160

yashasvi2696, 2018-03-16:
Hello,

I am trying to interface bgm11s with BHI160. For read and write functions, i refereed the following code : https://github.com/BoschSensortec/BMI160_driver/issues/12. 

However, the debugger gets hanged on a switch case(in em_i2c.c) file, and i am not able to initialize the driver. Can you please tell me what the problem may be? Are there any changes to be made in the read and write function for BHI160?

BoschSensortec, 2018-03-16:
Hi @yashasvi2696,

Can you specify which switch case it is that the debugger hangs? We are not sure if we can help resolve the debugging issue with the em_lib or the toolchain you are using.

There is no specific read/write function required other than that in the example that you have already referred.

yashasvi2696, 2018-03-16:
In em_i2c.c file, inside the I2C_Transfer(I2C_TypeDef *i2c) function there is a switch case where the debugger gets hanged. 

[em_i2c.txt](https://github.com/BoschSensortec/BMI160_driver/files/1818777/em_i2c.txt)