boschsensortec/BMI160_SensorAPI #40: BMI160 i2c initialise

Crumblypirate, 2018-03-19:
Hi,

I've been playing around with the BMI160 and a Nordic nrf52832 board and seemed to have hit a wall fairly early on with initialising it for i2C using the nrf twi driver. I currently keep just getting -3 as an output, from what I can gather means device not found.

Here's my initialisation code:

```
void acc_init(void)
{	
       static const nrf_drv_twi_t m_twi = NRF_DRV_TWI_INSTANCE(TWI_INSTANCE_ID);
	struct bmi160_dev sensor;
	ret_code_t err_code;
        int8_t result;
	
	uint8_t sample_data;
	
	bool detected_Acc = false;
	
	const nrf_drv_twi_config_t accPins = { 
		.scl = ARDUINO_SCL_PIN,        //27
		.sda = ARDUINO_SDA_PIN,     //26
		.frequency = NRF_TWI_FREQ_100K,
		.interrupt_priority = APP_IRQ_PRIORITY_HIGH,
		.clear_bus_init = false	
	};
	
	err_code = nrf_drv_twi_init(&m_twi, &accPins, NULL, NULL);
	
	if (NRF_SUCCESS == err_code)
	{
		nrf_drv_twi_enable(&m_twi);
		NRF_LOG_INFO("success");	
	}
	APP_ERROR_CHECK(err_code);	
	
	sensor.id = BMI160_I2C_ADDR;
	sensor.interface = BMI160_I2C_INTF;
	sensor.read = Acc_i2c_Read;
	sensor.write = Acc_i2c_Write;
	sensor.delay_ms = Acc_delay_ms;
	
	result = BMI160_OK;
	result = bmi160_init(&sensor);
	
        //This outputs -3
	NRF_LOG_DEBUG("acc value: %d\r\n", result);
	
	//set output, data range and range
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_100HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;
	
	//set power mode of accelerometer
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
	
	result = bmi160_set_sens_conf(&sensor);	
}
```

 Any help would be greatly appreciated, cheers!

BoschSensortec, 2018-03-20:
Hi @Crumblypirate,

Could you also include the snippets for the i2c_read, i2c_write and delay_ms functions?

Crumblypirate, 2018-03-20:
Hi,
Apologies, I've edited my original question and added my read, write and delay. cheers :)

BoschSensortec, 2018-03-20:
Hi @Crumblypirate,

It should rather be,
```c
int8_t Acc_i2c_Write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
	int8_t rslt = 0;
	uint8_t data[len + 1];
	data[0] = reg_addr;
	for (uint16_t i = 0; i < len; i++) {
		data[i + 1] = reg_data[i];
	}
	
	rslt = nrf_drv_twi_tx(&m_twi, dev_id, data, len + 1, false);
	return rslt;
}

int8_t Acc_i2c_Read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
	int8_t rslt = 0;

	rslt = nrf_drv_twi_tx(&m_twi, dev_id, &reg_addr, 1, false);
	if (rslt == 0)
	{
		rslt = nrf_drv_twi_rx(&m_twi, dev_id, reg_data, len);
	}
	
	return rslt;
}

void Acc_delay_ms(uint32_t period)
{
	nrf_delay_ms(period);
}
``` 

Note that the I2C address is defined in the initialization of the values for the instance of ```bmi160_dev```
>sensor.id = BMI160_I2C_ADDR;

osea, 2018-03-31:
@BoschSensortec I tried the code you provided. However, I kept getting rslt of 2 from the Acc_i2c_Read. I modified in the Acc_i2c_Read with a delay between tx and rx. the rslt is 0 now. However, I kept getting zero from the IMU.

closed 2018-04-16 by Crumblypirate