boschsensortec/BMI160_SensorAPI #41: Current consumption in low power mode

dimitridou, 2018-03-22:
Hello,

I am using a BMI160 wired to a ST Bluenrg2 device through 4wire SPI. I am putting the gyroscope in suspend mode, and the accelerometer in low power mode (undesampling on, ODR = 12.5Hz, 4AVG cycle), and according to the datasheet I am expecting a current consumption of about 10uA. But when I measure it, I have something around 90uA. 
I checked all the configurations registers, and they all have the expected value. 
I am thinking that it could be the SPI ports, and I was wondering if the BMI 160 had some pull-up/down resistors on those pins, and if they could be set/unset through the library, as I couldn't see any information about it on the .h file nor on the datasheet.

Thanks,
Dimitri

BoschSensortec, 2018-03-22:
Hi @dimitridou,

Are you using the sensor APIs in this repository? If so, can you describe your configuration sequence using the APIs?

dimitridou, 2018-03-23:
Hi @BoschSensortec ,

Yes I am, here is my initialisation code:

``
ErrorStatus initialize_BMI() {
	uint8_t data;
	/* You may assign a chip select identifier to be handled later */
	sensor.id = 0;
	sensor.interface = BMI160_SPI_INTF;
	sensor.read = BMI160_SPI_bus_read;
	sensor.write = BMI160_SPI_bus_write;
	sensor.delay_ms = wait_blocking;

	int8_t rslt = BMI160_OK;
	rslt = bmi160_init(&sensor);
	if (rslt != 0) {
		printf("erreur bmi160Init");
	}
	rslt = bmi160_get_regs(0x40,&data,1,&sensor);
	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_12_5HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_LOWPOWER_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_SUSPEND_MODE;

	/* Set the sensor configuration */
	rslt += bmi160_set_sens_conf(&sensor);

	rslt = bmi160_get_power_mode(&pmu_status, &sensor);

	if ((pmu_status.accel_pmu_status == BMI160_ACCEL_PMU_LOW_POWER) &&
			(pmu_status.gyro_pmu_status == BMI160_GYRO_PMU_SUSPEND)) {
		printf("\n Accel in LP and gyro in suspend mode\r\n");
	}
``