boschsensortec/BMI160_SensorAPI #42: Watermark interrupt - getting FIFO buffer values

kont40, 2018-03-29:
Hello,
I am using BMI160 in watermark configuration. I am facing a strange issue:
FIFO is set to get filled by the accelerometer only. The watermark interrupt seems to happen correctly. However, when reading the buffer in the watermark interrupt, most of the returned content is just zeros. The behavior seems to be determined to some extent by the watermark level set in bmi160_set_fifo_wm . For instance, if it is set to 56 (= FIFO filled up to 224 bytes) all bytes are filled with values that seem meaningful. However, If the level is set to 150 for example (i.e. FIFO fills up to 600 bytes), there are values at the beginning, but most of the buffer is filled with zeros, as shown below. In that dev_bmi160.fifo->length always shows correct value after the reading, and rslt = bmi160_get_fifo_data(&dev_bmi160);  returns BMI160_OK.
=======
CE BD FE 0A 62 F3 BB BD 0C 0B 7A F3 DE BD D0 0A 72 F3 
C5 BD 03 0B 5E F3 C4 BD 20 0B 95 F3 FC BD D1 0A A8 F3 
D4 BD F8 0A 4E F3 DB BD EC 0A 9E F3 F1 BD CE 0A 96 F3 
A1 BD 01 0B 61 F3 F3 BD AB 0A 7C F3 D1 BD F5 0A 5B F3 
E3 BD F7 0A 53 F3 E8 BD E8 0A 99 F3 CD BD FD 0A 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00
==============

The BMI initialization is like this:
Operation mode is first set by:

```
int8_t set_bmi160_operating_mode(struct bmi160_dev *dev)
{
		/* Select the Output data rate, range of accelerometer sensor */
		dev->accel_cfg.odr 		= BMI160_ACCEL_ODR_25HZ;
		dev->accel_cfg.range 	= BMI160_ACCEL_RANGE_2G;
		dev->accel_cfg.bw 		= BMI160_ACCEL_BW_NORMAL_AVG4;

		/* Select the power mode of accelerometer sensor */
		dev->accel_cfg.power 	= BMI160_ACCEL_NORMAL_MODE;

		/* Select the Output data rate, range of Gyroscope sensor */
		dev->gyro_cfg.odr 		= BMI160_GYRO_ODR_25HZ;
		dev->gyro_cfg.range 	= BMI160_GYRO_RANGE_250_DPS; //BMI160_GYRO_RANGE_2000_DPS;
		dev->gyro_cfg.bw 			= BMI160_GYRO_BW_NORMAL_MODE;

		/* Select the power mode of Gyroscope sensor */
		dev->gyro_cfg.power 	= BMI160_ACCEL_SUSPEND_MODE; //BMI160_GYRO_NORMAL_MODE; 

		/* Set the sensor configuration */
		bmi160_set_sens_conf(dev);	
		// To enable the interrupt need to put the power in normal state first
		//configure_any_motion_interrupt();
		//init_bmi160_interrupt_pin();		
	return 0;	
}
```

Next, watermark interrupt is defined by:
```
int8_t set_bmi160_fifo_watermark_interrupt(struct bmi160_dev *dev)
{
	int8_t rslt = BMI160_OK;

	/* Disable other FIFO settings */
	rslt = bmi160_set_fifo_config(BMI160_FIFO_CONFIG_1_MASK , BMI160_DISABLE, dev);

	rslt=bmi160_set_fifo_wm(BMI160_FIFO_BUF, dev);

	fifo_frame.data		= fifo_buff;
	fifo_frame.length = BMI160_FIFO_BYTES; // //get only acceleration - 3 axes x 2 bytes = 6 bytes per frame; 
	dev->fifo		= &fifo_frame;
	
	/* Configure the sensor's FIFO settings */
	rslt=bmi160_set_fifo_config(BMI160_FIFO_ACCEL, BMI160_ENABLE, dev); //get only acceleration - 3 axes x 2 bytes = 6 bytes per frame; put into buffer 100 frames 

	/* Select the Interrupt channel/pin */
	int_config.int_channel = BMI160_INT_CHANNEL_2;// Interrupt channel/pin 2

	/* Select the Interrupt type */
	int_config.int_type = BMI160_ACC_GYRO_FIFO_WATERMARK_INT;
	/* Select the interrupt channel/pin settings */
	int_config.int_pin_settg.output_en 			= BMI160_ENABLE;					// Enabling interrupt pins to act as output pin
	int_config.int_pin_settg.output_mode 		= BMI160_DISABLE;					// Choosing push-pull mode for interrupt pin
	int_config.int_pin_settg.output_type 		= BMI160_DISABLE;					// Choosing active low output
	int_config.int_pin_settg.edge_ctrl 			= BMI160_ENABLE;					// Choosing edge triggered output
	int_config.int_pin_settg.input_en 			= BMI160_DISABLE;					// Disabling interrupt pin to act as input
	int_config.int_pin_settg.latch_dur 			= BMI160_LATCH_DUR_NONE;	// non-latched output
	int_config.fifo_WTM_int_en							= BMI160_ENABLE; // enable the interrupt
	
	/* Set the interrupt */
	bmi160_set_int_config(&int_config, dev); /* sensor is an instance of the structure bmi160_dev  */	

	return 0;	
}
```

The buffer to read data from is defined as:
```
#define BMI160_FIFO_BUF 150 
#define BMI160_FIFO_BYTES BMI160_FIFO_BUF*4

uint8_t fifo_buff[BMI160_FIFO_BYTES];	
```

and data is read in the watermark interrupt handler as follows:
```
void process_fifo_watermark_interrupt_handler(nrf_drv_gpiote_pin_t pin, nrf_gpiote_polarity_t action)
{
	uint16_t index = 0;
	int8_t rslt = BMI160_OK;

	memset(fifo_buff, 0, sizeof(fifo_buff));
	rslt = bmi160_get_fifo_data(&dev_bmi160); // Note that next watermark interrupt would not happen unless we read the data from the current one!
	

      /* Print the raw FIFO data */
      for (index = 0; index < dev_bmi160.fifo->length; index++) 
			{
				app_uart_put(fifo_buff[index]);
      }
}
```

Any advice for resolving the problem is welcome.
Thanks

BoschSensortec, 2018-03-29:
Hi @kont40,

This could happen if your I2C or SPI read API doesn't support a burst read of more than a certain number of bytes. Can you confirm that the linked function can support up to 600 bytes or more?

kont40, 2018-03-31:
Hello @BoschSensortec It seems you are right; I found a thread on the topic here https://devzone.nordicsemi.com/f/nordic-q-a/19891/transfer-more-than-255-bytes-with-spi-master-on-nrf52  . So I will investigate the matter and report. Thanks