boschsensortec/BMI160_SensorAPI #43: BMI160 internal pull-up

oagu5785, 2018-03-29:
Hello,
I'm struggling to figure out whether there are internal pull-ups available for the secondary interface of BMI160 when this secondary interface is configured as I2C and the primary interface is configured as 4-wire SPI.

BMI160 datasheet states the following:

" 3.3.1.1 Configuration options of the secondary interface as magnetometer interface
The secondary interface configured  as  a  magnetometer  interface is  a I2C master  interface supporting Fast-Mode I²C communication. In  this  mode the ASCx pad  is  output  only,  since  both  master  and  slave  don’t  support  clock stretching  and  only  master  can drive  the  clock.  ASCx  is configured  as  push-pull  output  when magnetometer interface is enabled. The ASDx pad is used for data transfer between master and slave devices and is bidirectional. Since  I²C  master/slave  are using  open-drain  configuration  for  bidirectional  lines,  there  is  an internal, configurable pull-up resistor connected to the ASDx pad. "

Also, in figure 4.3.1of datasheet (Primary SPI 4-wire and secondary magnetometer interface (I2C)" there is a wiring diagram that supports the above text abstract, ie, pull-up resistor in I2C in the secondary interface as not shown.

So every seems to support the fact that pull-up should be available internally in BMI160 for the I2C. However I can seem to find the way to enable these pull-ups. Could you shed some light on this ?

Thanks
Oriol

BoschSensortec, 2018-04-03:
Hi @oagu5785,

You need to connect external pull-ups to the auxiliary I2C interface.

oagu5785, 2018-04-03:
Hello,
I also saw there is this register: BMI160_INTL_PULLUP_CONF_ADDR that suggests that some kind of configurable pull-ups might exist. So no way to have internal pull-ups despite datasheet statement...

Regards,
Oriol

BoschSensortec, 2018-04-04:
Hi @oagu5785,

You will need a special application note for that which can be received only under an NDA. Kindly use our [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form) to request for the same.

closed 2018-04-12 by oagu5785