boschsensortec/BMI160_SensorAPI #44: BMI160 + NRF52832 reading zero of the data

osea, 2018-03-31:
Hello,

Thank you for the great library. I am using the BMI160 with I2C using the Nordic NRF52832 microcontroller. This controller has a driver for I2C referred to as nrf_drv_twi. The SDK version is 15 with  S132.

Here is the setup of the BMI160.
                sensor.id = BMI160_ADDR; //BMI160_ADDR is 0x68U
		sensor.interface = BMI160_I2C_INTF;
		sensor.read = Acc_i2c_Read;
		sensor.write = Acc_i2c_Write;
		sensor.delay_ms = Acc_delay_ms;
	
		int8_t rslt = BMI160_OK;
		NRF_LOG_INFO("\r\nrslt is :%d",rslt);
		rslt = bmi160_init(&sensor);
		NRF_LOG_INFO("\r\nrslt is :%d",rslt);

The write function is
int8_t Acc_i2c_Write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
	int8_t rslt = 0;
	uint8_t data[len + 1];
	data[0] = reg_addr;
	for (uint16_t i = 0; i < len; i++) {
		data[i + 1] = reg_data[i];
	}
	
	rslt = nrf_drv_twi_tx(&m_twi, dev_id, data, len + 1, false);
	return rslt;
}

The read function is:
int8_t Acc_i2c_Read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
	int8_t rslt = 0;

	rslt = nrf_drv_twi_tx(&m_twi, dev_id, &reg_addr, 1, false);
	nrf_delay_ms(5);
	if (rslt == 0)
	{
		rslt = nrf_drv_twi_rx(&m_twi, dev_id, reg_data, len);
	}
	
	return rslt;
}

And I put the read_data function in a while loop.
static void read_data()
{
	
    int8_t rslt = BMI160_OK;
		struct bmi160_sensor_data accel;
		struct bmi160_sensor_data gyro;

		/* To read only Accel data */
		rslt = bmi160_get_sensor_data(BMI160_GYRO_SEL, NULL, &gyro, &sensor);
	
	NRF_LOG_INFO("rslt:%d, data: %d", rslt,gyro.x);

}

The rslt is all 0 that indicates the code is working. However, I kept reading 0 from the BMI160. 

Any help would be appreciated.

Thanks,

Brandon,

BoschSensortec, 2018-04-03:
Hi @osea,

You will need to configure the sensor; select power modes, data rates, range, etc.

osea, 2018-04-07:
@BoschSensortec I configured the IMU with following functions.

int8_t rslt = BMI160_OK;

	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE; 

	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&sensor);
	nrf_delay_ms(100);
	NRF_LOG_INFO("setup result is : %d",rslt);

The rslt is "0" indicating that the configuration is done. However, the data is still zero.

BoschSensortec, 2018-04-09:
Hi @osea,

Can you run the following snippet after the delay to track if there are internal errors? 

```c
int8_t rslt;
uint8_t data = 0;

/* Read the error reg */
rslt = bmi160_get_regs(BMI160_ERROR_REG_ADDR, &data, 1, dev);

if (rslt == BMI160_OK) {
	data = data >> 1;
	data = data & BMI160_ERR_REG_MASK;
	if (data == 1)
		rslt = BMI160_E_ACCEL_ODR_BW_INVALID;
	else if (data == 2)
		rslt = BMI160_E_GYRO_ODR_BW_INVALID;
	else if (data == 3)
		rslt = BMI160_E_LWP_PRE_FLTR_INT_INVALID;
	else if (data == 7)
	rslt = BMI160_E_LWP_PRE_FLTR_INVALID;
}

NRF_LOG_INFO("setup result is : %d",rslt);
```

osea, 2018-04-09:
Finally, I got the data from the BMI160 by enable the chip with CS.  However, When I integrate the code with the BLE-uart, I kept getting zero with the right configuration...

chrisluhui, 2018-06-16:
Hi,@osea! Have you figured out the solution?  The similar problem came to me.  I can read the data from the BMI160 correctly separately, but can read out just zero from the BMI160 with BLE template.  I need both your help, @osea and @BoschSensortec ! Thx!

lihongzhan, 2018-09-18:
i was too