boschsensortec/BMI160_SensorAPI #45: Orientation Recognization and Offset Compensation(FOC)

nimeshkotak, 2018-03-31:
Hi there,

In our design, we are using BMI160.  The product has display in Landscape mode and the PCB which has the sensor makes and angle of 27 degree with the plane of viewer's eyes. 
Due to that  when we fit the boards into  enclosure we are not able to see the 360 degree video properly. During the up and down movements the video moves in a curved manner. For left and right movements the video also moves linearly.
So what orientation should we set for x,y, and z co-ordinates of accelerator? And also how to perform fast offset compensation and Inline calibration for the device.

BoschSensortec, 2018-04-03:
Hi @nimeshkotak,

This space is currently only reserved for issues related to the sensor APIs. Please use our [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form) to report the issue and one of our engineers will get in touch with you.

nimeshkotak, 2018-04-03:
Hi @BoschSensortec ,

We have already tried to contact Bosch but haven't heard back from them. Can you please suggest some direct contact with whom we can work and find solution.  
By the way can you comment anything on the implementation we have done? We have sensor at an angle of 27 degree with display and the eys.