boschsensortec/BMI160_SensorAPI #46: some error on the use of BMI160?

lhxy1991, 2018-04-03:
when I use BMI160,i find some error on the outputdata,

[image]
lwhen car is stopping,the gyro data is like this,how can I solve this problem?

BoschSensortec, 2018-04-03:
Hi @lhxy1991,

Can you give us more details about how you are calling the sensor API, your platform, and the implementation of your read/write/delay functions.

Thank you.

lhxy1991, 2018-04-03:
YES,  i can give you the drive,the bmi160_prob is the start.
I use it in linux,and i use it to obtain angle information in car.
`#include <linux/kernel.h>
#include <linux/module.h>
#include <linux/i2c.h>
#include <linux/uaccess.h>
#include <linux/poll.h>
#include <linux/slab.h>
#include <asm/div64.h>
#include <linux/delay.h>
#include <linux/spi/spi.h>
#include <linux/miscdevice.h>
#include <linux/input.h>
#include <linux/bmi160.h>
//#include <mach/sigdet.h>

extern short sigdet_speed_get(void);

/* bmi160 chip id */
#define DEVICE_ID_BMI160                0xD1
#define DEVICE_ID_VALID_STEERING_WHEEL  0xFF
#define DEVICE_ID_VALID_SPEED_ONLY         0

#define BMI160_USER_PMU_TRIGGER_ADDR		0x6c
#define BMI160_CMD_COMMANDS_ADDR			0x7e
#define BMI160_USER_ACCEL_CONFIG_ADDR		0x40
#define BMI160_USER_ACCEL_RANGE_ADDR		0x41
#define BMI160_USER_GYRO_CONFIG_ADDR		0x42
#define BMI160_USER_GYRO_RANGE_ADDR			0x43
#define BMI160_USER_DATA_8_ADDR				0x0c	/* gyro data */
#define BMI160_USER_DATA_8_ACCEL_ADDR		0x12	/* accel data */
#define BMI160_USER_TEMPERATURE_0_ADDR		0x20	/* temperature data */
#define BMI160_USER_FIFO_LENGTH_0_ADDR		0x22
#define BMI160_USER_FIFO_LENGTH_1_ADDR		0x23
#define BMI160_USER_FIFO_CONFIG_0_ADDR      0x46
#define BMI160_USER_FIFO_CONFIG_1_ADDR      0x47

/**************************************************/
/**\name	BIT SLICE GET AND SET FUNCTIONS  */
/*************************************************/
#define BMI160_GET_BITSLICE(regvar, bitname)\
		((regvar & bitname##__MSK) >> bitname##__POS)

#define BMI160_SET_BITSLICE(regvar, bitname, val)\
		((regvar & ~bitname##__MSK) | \
		((val<<bitname##__POS)&bitname##__MSK))

/* Acc_Conf Description - Reg Addr --> (0x40), Bit --> 4...6 */
#define BMI160_USER_ACCEL_CONFIG_ACCEL_BW__POS               (4)
#define BMI160_USER_ACCEL_CONFIG_ACCEL_BW__LEN               (3)
#define BMI160_USER_ACCEL_CONFIG_ACCEL_BW__MSK               (0x70)
#define BMI160_USER_ACCEL_CONFIG_ACCEL_BW__REG	(BMI160_USER_ACCEL_CONFIG_ADDR)

/* Gyro_Conf Description - Reg Addr --> (0x42), Bit --> 4...5 */
#define BMI160_USER_GYRO_CONFIG_BW__POS               (4)
#define BMI160_USER_GYRO_CONFIG_BW__LEN               (2)
#define BMI160_USER_GYRO_CONFIG_BW__MSK               (0x30)
#define BMI160_USER_GYRO_CONFIG_BW__REG               \
(BMI160_USER_GYRO_CONFIG_ADDR)

/**************************************************************/
/**\name	GYRO CONFIGURATION LENGTH, POSITION AND MASK*/
/**************************************************************/
/* Gyro_Conf Description - Reg Addr --> (0x42), Bit --> 0...3 */
#define BMI160_USER_GYRO_CONFIG_OUTPUT_DATA_RATE__POS               (0)
#define BMI160_USER_GYRO_CONFIG_OUTPUT_DATA_RATE__LEN               (4)
#define BMI160_USER_GYRO_CONFIG_OUTPUT_DATA_RATE__MSK               (0x0F)
#define BMI160_USER_GYRO_CONFIG_OUTPUT_DATA_RATE__REG               \
(BMI160_USER_GYRO_CONFIG_ADDR)

[... 1629 lines cut ...]
static const struct spi_device_id bmi160_id[] = {
	{"bmi160"},
	{ }
};
MODULE_DEVICE_TABLE(spi, bmi160_id);

static struct spi_driver bmi160_gyro_driver = {
	.id_table = bmi160_id,
	.driver = {
		.name = BMI160_GYRO_DEV_NAME,
		.owner = THIS_MODULE,
	},
	.probe = bmi160_probe,
	.remove = bmi160_remove,
//	.suspend = bmi160_suspend,
//	.resume = bmi160_resume,
};

module_spi_driver(bmi160_gyro_driver);

MODULE_AUTHOR("wenjunhe@pateo.com.cn>");
MODULE_DESCRIPTION("BST-BMI160 Tri-axis gyroscope driver");
MODULE_LICENSE("GPL");

`

lhxy1991, 2018-04-03:
thanks for your help.
i use to get angle information for car in linux,and i get data from FIFO,
and I give you the driver what i used.
The start function is function:bmi160_prob, then it will visit function :bmi160_initialize_sensor
and  bmi160_work_func. they include the configion and output.
I check the ripple of the supply power, its no problem.