boschsensortec/BMI160_SensorAPI #47: Values not updating

SimoneNascivera, 2018-04-05:
Hi,
I'm trying to implement the BMI160 for an ARM system but when I initialize the sensor and I start a loop like:
`` while(1){
                result = bmi.read_sensor_data(sensore, 0);
                cout << "x: " << result.x << "    y: " << result.y << "    z: " << result.z << endl;
                usleep(1000000);
        }
``
It gives me always the same values even if I manually change the spatial position of the sensor.
Am I missing some coding?
Thanks

SimoneNascivera, 2018-04-05:
Yes I'm using it applied to a code that enables to use integrated i2c bus with ARM. This is the library:
`
#include "bmi160_arm.h"

int file;
int adapter_nr = 2;
char filename[40];

BMI160::BMI160(){
        sprintf(filename,"/dev/i2c-1");
        file = open(filename, O_RDWR);//, O_NONBLOCK,O_NDELAY);
        if (file < 0)
        {
                printf("Failed to open the bus.");
                /* ERROR HANDLING; you can check error */
                exit(1);
        }

        /* The I2C address : 0x10/0x11/0x12/0x13 --> change in bmm150_defs.h */

        if (ioctl(file, I2C_SLAVE, BMI160_I2C_ADDR) < 0)
        {
                printf("Failed to acquire bus access and/or talk to slave.");
                /* ERROR HANDLING; you can check errno to see what went wrong */
                exit(1);
        }
        else{
                printf("BMI160 found at 0x%02X\n",  BMI160_I2C_ADDR);
        }
}

int8_t i2c_read(uint8_t dev_addr, uint8_t reg_addr, uint8_t *data, uint16_t len){
        uint8_t buffer_write[2];
        memset(buffer_write,'\0',2);
	int n_writ;
        n_writ = 0;
        // Request data
        buffer_write[0] = reg_addr;

        n_writ = write(file,buffer_write,1);
        if (n_writ != 1)
        {
                /* ERROR HANDLING: i2c transaction failed */
        	printf("BMI150 Reading Error (cannot request data): Failed to write");
       	}
	int n_read;
    	// Read data
    	n_read = read(file,data,len);
        if (n_read != len)
        {
        /* ERROR HANDLING: i2c transaction failed */
        	printf("BMI160 Reading Error (not enough data readed) :Failed to read");
    	}
}

int8_t i2c_write(uint8_t dev_addr, uint8_t reg_addr, uint8_t *data, uint16_t len){
        uint8_t buffer_write[len+1]; // cast needed???!?
        memset(buffer_write,'\0',len+1);
        int n_writ;
        buffer_write[0] = reg_addr;
	for(int i = 0; i<len; i++){
	        buffer_write[i+1] = data[i];
	}
        n_writ = write(file,buffer_write, len+1);
        if( n_writ < len+1)
        {
                /* ERROR HANDLING: i2c transaction failed */
                printf("BMI160 Writing Error :Failed to write");
        }
}

void delay_ms(uint32_t period){
	usleep(period*1000);
}

bmi160_dev BMI160::initialize(){
	struct bmi160_dev sensor;

	sensor.id = BMI160_I2C_ADDR;
	sensor.interface = BMI160_I2C_INTF;
	sensor.read = i2c_read;
	sensor.write = i2c_write;
	sensor.delay_ms = delay_ms;

	int8_t rslt = BMI160_OK;
	rslt = bmi160_init(&sensor);
	/* After the above function call, accel and gyro parameters in the device structure 
	are set with default values, found in the datasheet of the sensor */

	return sensor;
}

void BMI160::normal_mode_config(bmi160_dev sensor){
	int8_t rslt = BMI160_OK;

	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_3200HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE; 

	/* Set the sensor configuration */ 
	rslt = bmi160_set_sens_conf(&sensor);
}

bmi160_sensor_data BMI160::read_sensor_data(bmi160_dev sensor, int mode=0){
	int8_t rslt = BMI160_OK;
	struct bmi160_sensor_data accel;
	struct bmi160_sensor_data gyro;
	struct bmi160_sensor_data result;
	switch(mode)
	{
		case 0:
			/* To read only Accel data */
			rslt = bmi160_get_sensor_data(BMI160_ACCEL_SEL, &accel, NULL, &sensor);
			result = accel;
			break;
		case 1:
			/* To read only Gyro data */
			rslt = bmi160_get_sensor_data(BMI160_GYRO_SEL, NULL, &gyro, &sensor);
			result = gyro;
			break;
		/*case 2:
			/* To read both Accel and Gyro data 
			bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL), &accel, &gyro, &sensor);
			break;
		case 3:
			/* To read Accel data along with time 
			rslt = bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_TIME_SEL) , &accel, NULL, &sensor);
			break;
		case 4:
			/* To read Gyro data along with time 
			rslt = bmi160_get_sensor_data((BMI160_GYRO_SEL | BMI160_TIME_SEL), NULL, &gyro, &sensor);
			break;
		case 5:
			/* To read both Accel and Gyro data along with time
			bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL | BMI160_TIME_SEL), &accel, &gyro, &sensor);
			break;
		*/
	}
	return result;
}
`

and this is the code:
`#include "bmi160_arm.h"
#include <iostream>

using namespace std;

BMI160 bmi = BMI160();

int main(){
	bmi160_dev sensore;
	sensore = bmi.initialize();
	bmi.normal_mode_config(sensore);
	bmi160_sensor_data result;
	while(1){
		result = bmi.read_sensor_data(sensore, 0);
		cout << "x: " << result.x << "    y: " << result.y << "    z: " << result.z << endl;
		usleep(1000000);
	}
	cout << "done" << endl;
	return 0;
}

`

BoschSensortec, 2018-04-06:
Hi @SimoneNascivera,

- It would be recommended to add a return at the end of your `i2c_read` and `i2c_write` functions. 0 for success, non-zero otherwise.
- What are the values of `rslt` after each of the API calls? Preferably this should be checked against `BMI160_OK` like below
```c
if (rslt != BMI160_OK)
{
    /* ERROR HANDLING: BMI160 API failure */
    printf("BMI160 Error code : %d", rslt);
}
```

SimoneNascivera, 2018-04-06:
Thank you so much. That was the issue and now it all works just fine.

closed 2018-04-06 by SimoneNascivera