boschsensortec/BMI160_SensorAPI #48: bmi160+esp wrover i2c

hawres, 2018-04-09:
Hi,
I have some problems probably with sensor configuration. I'm using esp wrover, [this is documentation](http://esp-idf.readthedocs.io/en/latest/index.html). 

These are my write/read functions:

    int8_t i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
    {
        if (length == 0) {
            return (int8_t)ESP_ERR_INVALID_SIZE;
        }
        i2c_cmd_handle_t cmd = i2c_cmd_link_create();
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, ( SENSOR_ADDR << 1 ) | WRITE_BIT, ACK_CHECK_EN);
        i2c_master_write_byte(cmd, reg_addr, ACK_CHECK_DIS);
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, ( SENSOR_ADDR << 1 ) | READ_BIT, ACK_CHECK_EN);
        if (length > 1) {
            i2c_master_read(cmd, reg_data, length - 1, ACK_VAL);
        }
        i2c_master_read_byte(cmd, reg_data+length-1,NACK_VAL);
        i2c_master_stop(cmd);
        esp_err_t ret = i2c_master_cmd_begin(I2C_PORT, cmd, 1000/portTICK_PERIOD_MS);
        i2c_cmd_link_delete(cmd);
        return (int8_t)ret;
    }

    nt8_t i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
    {	
        i2c_cmd_handle_t cmd = i2c_cmd_link_create();
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, ( SENSOR_ADDR << 1 ) | WRITE_BIT, ACK_CHECK_EN);
        i2c_master_write_byte(cmd, reg_addr, ACK_CHECK_EN);
        i2c_master_write(cmd, reg_data, length, ACK_CHECK_EN);
        i2c_master_stop(cmd);
        esp_err_t ret = i2c_master_cmd_begin(I2C_PORT, cmd, 1000/portTICK_PERIOD_MS);
        i2c_cmd_link_delete(cmd);
        return (int8_t)ret;
    }

where macros:

    WRITE_BIT            0x0           
    READ_BIT             0x1          
    ACK_CHECK_EN          0x1        
    ACK_CHECK_DIS        0x0          
    ACK_VAL              0x0            
    NACK_VAL             0x1      

my sensor init function:

    void initSensor()
    {
        sensor.id = BMI160_I2C_ADDR;
        sensor.interface = BMI160_I2C_INTF;
        sensor.read = i2c_read;
        sensor.write = i2c_write;
        sensor.delay_ms = delay;
        int8_t rslt = BMI160_OK;
        rslt = bmi160_init(&sensor);
        printf("init %d\n", rslt); // rslt is 0, so it seems work
    }     

accelerometer configuration: 

    void confAccel()
    {
        /* Select the Output data rate, range of accelerometer sensor */
        sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
        sensor.accel_cfg.range = BMI160_ACCEL_RANGE_8G;
        sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;
        /* Select the power mode of accelerometer sensor */
        sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
        int8_t rslt = BMI160_OK;
        /* Set the sensor configuration */
        rslt = bmi160_set_sens_conf(&sensor);
        printf("conf %d\n", rslt); // rslt is 0, so it seems work
    }

so I made 

    void selfTestAccel()
    {
        int8_t rslt = BMI160_OK;
        uint8_t reg_addr = 0x40;
        uint8_t data = 0x2C;
        uint16_t len = 1;
        rslt = bmi160_set_regs(reg_addr, &data, len, &sensor); //  rslt is 0, so it seems work
        rslt = bmi160_perform_self_test(BMI160_ACCEL_ONLY, &sensor); // rslt is 0, so it seems work
    }

but when i tried to read 

    void readAccel()
    {
        int8_t rslt = BMI160_OK;
        rslt = bmi160_get_sensor_data(BMI160_ACCEL_SEL, &accel, NULL, &sensor);
    }

i got 0 on all axes.  Could You please help me?

BoschSensortec, 2018-04-09:
Hi @hawres ,

Can you run the following snippet after the delay to track if there are internal errors? 

```c
int8_t rslt;
uint8_t data = 0;

/* Read the error reg */
rslt = bmi160_get_regs(BMI160_ERROR_REG_ADDR, &data, 1, &sensor);

if (rslt == BMI160_OK) {
	data = data >> 1;
	data = data & BMI160_ERR_REG_MASK;
	if (data == 1)
		rslt = BMI160_E_ACCEL_ODR_BW_INVALID;
	else if (data == 2)
		rslt = BMI160_E_GYRO_ODR_BW_INVALID;
	else if (data == 3)
		rslt = BMI160_E_LWP_PRE_FLTR_INT_INVALID;
	else if (data == 7)
	rslt = BMI160_E_LWP_PRE_FLTR_INVALID;
}

printf("Error register's rslt: %d\n", rslt);
```

hawres, 2018-04-09:
Hi, i have next problem. I configured any motion interrupt - its works very well. Now i'm trying with high-g.
```c    
    void confInterrupt()
    {
            /* Select the Interrupt channel/pin */
        int_config.int_channel = BMI160_INT_CHANNEL_1;// Interrupt channel/pin 1

        /* Select the Interrupt type */
        int_config.int_type = BMI160_ACC_HIGH_G_INT;
        /* Select the interrupt channel/pin settings */
        int_config.int_pin_settg.output_en = BMI160_ENABLE;// Enabling interrupt pins to act as output pin
        int_config.int_pin_settg.output_mode = BMI160_DISABLE;// Choosing push-pull mode for interrupt pin
        int_config.int_pin_settg.output_type = BMI160_DISABLE;// Choosing active low output
        int_config.int_pin_settg.edge_ctrl = BMI160_ENABLE;// Choosing edge triggered output
        int_config.int_pin_settg.input_en = BMI160_DISABLE;// Disabling interrupt pin to act as input
        int_config.int_pin_settg.latch_dur = BMI160_LATCH_DUR_1_28_SEC;// 

        //int_config.int_type_cfg.acc_any_motion_int.anymotion_en = BMI160_ENABLE;
        int_config.int_type_cfg.acc_high_g_int.high_g_x = BMI160_ENABLE;
        int_config.int_type_cfg.acc_high_g_int.high_g_y = BMI160_ENABLE;
        int_config.int_type_cfg.acc_high_g_int.high_g_z = BMI160_ENABLE;
        int_config.int_type_cfg.acc_high_g_int.high_hy = 0; 
        int_config.int_type_cfg.acc_high_g_int.high_dur = 10;
        int_config.int_type_cfg.acc_high_g_int.high_thres = 20;
        
        int8_t rslt = BMI160_E_NULL_PTR;
        /* Set the Any-motion interrupt */
        rslt = bmi160_set_int_config(&int_config, &sensor); /* sensor is an instance of the structure bmi160_dev  */
        printf("interrupt %d\n", rslt);
    }
```
but without success- interrupt appears every time, even when sensor is lying down. Could You send me please any working configuration? Or say more about about high_hy, high_dur and high_thres?
Thanks

hawres, 2018-04-11:
Hi, i have next question. My code:
```c
    initSensor();    //both according to documentation
    confAccelAndGyro();
```
[specific moves detection]
Later after a timer I want to disable gyroscope to save energy before I sent the MCU to sleep. I tried 
```c    
    bmi160_soft_reset(&sensor);
    disableGyro(); //in fact, this is the same config, but with  sensor.gyro_cfg.power = BMI160_GYRO_SUSPEND_MODE;
 
    confInterrupt();
```
and it works well, but is there any better way to do it? 
later i'm going to deep sleep
```c
    esp_sleep_enable_ext0_wakeup(GPIO_NUM_39, INPUT_LEVEL_LOW); 
    printf("Entering deep sleep\n");
    esp_deep_sleep_start();
```
and interruption from accelerometer will wake up esp.
Thanks

BoschSensortec, 2018-04-11:
Hi @hawres,

What would you like to improve here?

hawres, 2018-04-11:
Is any command to disable gyro without re-configuring all sensor?

BoschSensortec, 2018-04-11:
Hi @hawres,

You doesn't need to trigger a soft-reset to disable the gyroscope. The `bmi160_set_power_mode` API should suffice.

hawres, 2018-04-12:
Could You explain more, please? And show example how to disable gyro only using this command? I see that is 
```c
int8_t set_gyro_pwr(struct bmi160_dev *dev)
```
but i can't use it correct. Should i modify 
```c
sensor.gyro_cfg.power = BMI160_GYRO_PMU_SUSPEND;
```
and later call set_power_mode?
Thanks