boschsensortec/BMI160_SensorAPI #50: Watermark interrupt - pin is not always used as interrupt

T721, 2018-04-11:
Hi @BoschSensortec ,

i'm using the watermark interrupt of your system which shall be fired when more than 1 byte (which should mean a complete frame) is in the fifo. I want to retrieve one frame with the additional info.
This works pretty good right now but i have a problem with the sensor itself.
Sometimes the Interrupt pin is just not set and the watermark interrupts are not fired, even if there is content in the buffer
(I read from the registern even tho the interrupt pin was low)

here is a screenshot of how it should look like and how it does in about 80% of the time:
[image]

Sometimes it simply does not register and it comes to a result like this
[image]

in my source code this looks like the following where im reading gyro data in headermode + the 4 bytes of the sensor timeframe.
[image]

**Do you have an idea why sometimes the interrupt pin is not set correctly?**

Im using the standard configuration from your sensor examples.

```
	sensor.id = 0;
	sensor.interface = BMI160_SPI_INTF;
	sensor.read = user_spi_read;
	sensor.write = user_spi_write;
	sensor.delay_ms = user_delay_ms;

	int8_t rslt = BMI160_OK;
	rslt = bmi160_init(&sensor);

	/* Select the Output data rate, range of accelerometer sensor */
	sensor.accel_cfg.odr = BMI160_ACCEL_ODR_100HZ;
	sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
	sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

	/* Select the power mode of accelerometer sensor */
	sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;

	/* Select the Output data rate, range of Gyroscope sensor */
	sensor.gyro_cfg.odr = BMI160_GYRO_ODR_100HZ;
	sensor.gyro_cfg.range = BMI160_GYRO_RANGE_2000_DPS;
	sensor.gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

	/* Select the power mode of Gyroscope sensor */
	sensor.gyro_cfg.power = BMI160_GYRO_NORMAL_MODE;

	/* Set the sensor configuration */
	rslt = bmi160_set_sens_conf(&sensor);
```

watermark interrupt:
```
    struct bmi160_int_settg int1_config;

    // Select the Interrupt channel/pin
    int1_config.int_channel = BMI160_INT_CHANNEL_1;// Interrupt channel/pin 1

    // Select the Interrupt type
    int1_config.int_type = BMI160_ACC_GYRO_FIFO_WATERMARK_INT;

    // Select the interrupt channel/pin settings
    int1_config.int_pin_settg.output_en = BMI160_ENABLE;// Enabling interrupt pins to act as output pin
    int1_config.int_pin_settg.output_mode = BMI160_DISABLE;// Choosing push-pull mode for interrupt pin
    int1_config.int_pin_settg.output_type = BMI160_ENABLE;// Choosing active high output
    int1_config.int_pin_settg.edge_ctrl = BMI160_ENABLE;// Choosing edge triggered output
    int1_config.int_pin_settg.input_en = BMI160_DISABLE;// Disabling interrupt pin to act as input
    int1_config.int_pin_settg.latch_dur = BMI160_LATCH_DUR_NONE;// non-latched output

    int8_t res = bmi160_set_int_config(&int1_config, &sensor); // sensor is an instance of the structure bmi160_dev

    if(res != 0) return res;

    //set wartermark interrupt thres to 1
    uint8_t data[1] = {1};
    res = bmi160_set_regs(0x46, data, 1, &sensor);
    return res;
```

Fifo conf, actually not used because i read via get_regs but i set it up with the drivers function so all regs are written correctly:

```
    uint8_t fifo_buff[33];

    // Modify the FIFO buffer instance and link to the device instance
    struct bmi160_fifo_frame fifo_frame;
    fifo_frame.data = fifo_buff;
    fifo_frame.length = 7;
    bmi.sensor.fifo = &fifo_frame;

    bmi.sensor.delay_ms = user_delay_ms;
    // Configure the sensor's FIFO settings
   std::cout << "configure bmi status: "  << (int32_t) bmi160_set_fifo_config(BMI160_FIFO_G_ENABLE | BMI160_FIFO_HEAD_ENABLE | BMI160_FIFO_TIME_ENABLE, BMI160_ENABLE, &bmi.sensor) << std::endl;

```

T721, 2018-04-11:
Hi @BoschSensortec,
the delay is about 0,3ms from intterupt occurance to clear out of the buffer.

Sadly i cannot use the `BMI160_FIFO_BYTES_OVERREAD` becaus emy sensor does not support reading over 31 bytes at once.

I will try to get a scope with the i2c communication and the IR pin.
In my opinon sometimes the interrupt is just not set because it never gets fired even if there is content to read which seems pretty trange to me.
When the interrupt is running correctly, then the interrupt is allways fired and read.
There are these two situations with the same piece of code: perfect working or not working at all.
Aditionally i can't reproduce the behaviour in debugging mode.
I will provide a screenshot of the communication.

T721, 2018-04-13:
Hi @BoschSensortec ,

im still working on the issue and i can't solve it.
The communication of the sensor is working fine but sadly i cannot provide a screenshot because i can't capture the i2c pins.
But the readouts from fifo are good and the data is comming if i read.
Only the interrupt is not working in some runs of my application.
It seems like a timing problem to me. 

Do i need to uninstall the interrupt on the BMI when i close my Application?

T721, 2018-04-13:
Hi @BoschSensortec ,

i found an some kind of an error.
in the function `enable_fifo_wtm_int` the BMI160_SET_BITS macro does sometimes not setup the wtm interrupt correctly. I found this by reading out the data transfered to the sensor.
When initializing the interrupts, the following bits are transfered:

bad initialize:
tx: 1   addr: 51 bits: [0]00000000 

goot initialize:
tx: 1   addr: 51 bits: [0]01000000 

how can it happen that these bits are not correctly set up by the macro?

BoschSensortec, 2018-04-16:
Hi @T721,

There are 2 parts to the interrupt. A soft map and and a hardware map. It was working because a soft map which in essence enables the feature and its status is available via the status registers was done correctly. For it to trigger a hardware interrupt pin, you will need to map to a hardware pin, in this context int1 or int2. To map it to int1, you will need to enable it manually as per your comment.

T721, 2018-04-16:
Hi @BoschSensortec 
thanks for your comment.
It is a really strange behaviour as it worked as hardware interrupt before, at least in my debugger...
Something really got strange there. I fixed it now by correctly setting it up and my issue is solved.
I don't know if this is worth a hint in your docs.

I will close this now.
Greetings and thanks again.

closed 2018-05-03 by T721