boschsensortec/BMI160_SensorAPI #51: BMM150 I2C coms with BMI160

oagu5785, 2018-04-12:
Hello,
I am working with a custom board with nRF52832 interfaced with BMI160 through 4-Wire SPI as its primary interface. BMI160 interfaces through I2C with BMM150. 1K external pull-ups are used for the I2C.

I am using the latest Bosch APIs for both BMM150 and BMI160. What I see is that BMM150 does not always respond as expected when asked through BMI160. As an example of that, once the code executes bmm150_init(), ie, power_control bit is set to 1 and the dev_id is read, the dev_id returned value is zero unless a couple of changes are done.

These are the two simultaneous 'tweaks' to be able to read the 0x32 dev_id code:
- set bmi160.aux_cfg.aux_rd_burst_len = 0 . By default BMI160 does 6bytes burst reads to BMM150.
- increase the BMM150_START_UP_TIME delay in bmm150_init() from 3ms to 120ms.

I do not understand why such an increase in the delay is required in my hardware. 
Is there a condition to be met so that 6burst bytes can be performed properly ?

I've also realized that if several reads to the dev_id are done in a loop inside function bmm150_init(), additional delays are necessary to be able to read the dev_id  as shown in the code snippet. 

int8_t bmm150_init(struct bmm150_dev *dev)
{
	int8_t rslt;
	uint8_t chip_id = 0;

	/* Check for null pointer in the device structure*/
	rslt = null_ptr_check(dev);
	/* Proceed if null check is fine */
	if (rslt ==  BMM150_OK) {
		/* Power up the sensor from suspend to sleep mode */
		rslt = set_power_control_bit(BMM150_POWER_CNTRL_ENABLE, dev);
		/* Start-up time delay of 3ms*/
		dev->delay_ms(120);
		if (rslt ==  BMM150_OK) {
			/* Chip ID of the sensor is read */
			while(1) {
			      rslt = bmm150_get_regs(BMM150_CHIP_ID_ADDR, &chip_id, 1, dev);
			      dev->delay_ms(120); 
			}
			/* Proceed if everything is fine until now */
			if (rslt ==  BMM150_OK) {
				/* Check for chip id validity */
				if (chip_id == BMM150_CHIP_ID) {
					dev->chip_id = chip_id;
					/* Function to update trim values */
					rslt = read_trim_registers(dev);
				} else {
					rslt = BMM150_E_DEV_NOT_FOUND;
				}
			}
		}
	}
	return rslt;
}

If a use shorter delays than 120ms, at some point I start getting 0x00 instead of 0x32. When that point is reached, any other reads I do to the dev_id or to any other registers seem to be always 0x00. If the delay is 120ms or bigger it seems to return the expected (default) values as far as I have checked until now.

I've not been able to find and explanation to all this yet. Any help is welcome.

BoschSensortec, 2018-04-13:
@oagu5785,

Just for clarification, when you said `dev_id` you are referring to `chip_id`? The `dev_id` parameter is not modified or used within the API, it is a user variable to identify which sensor is connected. 

Which SDK are you using to program the NRF52832?

oagu5785, 2018-04-13:
Right, I meant chip_id (0x32). I am using nRF5 SDK v14.1.0

oagu5785, 2018-04-16:
Hello BoschSensortec 
Is there any debug information we could extract that helps us to get some insight about what's going on?

Thanks

oagu5785, 2018-05-03:
This issue can be closed. What happened is that there were soldering issues in the BMM150 ICs. Once the boards/BMM150 chips were reflowed again all issues disappeared.

closed 2018-05-03 by oagu5785