boschsensortec/BMI160_SensorAPI #52: bmi160_get_regs delay

yoonlee95, 2018-04-13:
Is the 1 ms delay necessary after a read operation?
section 3.2.4  doesn't specify anything regarding delay after a read.

T721, 2018-05-24:
I had the same question in #34 
You can reduce the delay when reading but need to have the 1ms when writing.
Bosch meant they implemented it like this because they had no smaller resolution.
I'm working with 1µs which is fine when you only read.

mEdTech88, 2018-07-02:
i have a question on the delay between transmit and recieve date.
I want to read the FIFO every timr the wartermark interrupt hits. My Problem is i just get Data with a HAL_Delay(600) between Transmitting the register 0x24 and the dummy byte from master to slave. 
But if I do this I loos onecomplete FIFO payload.DO I?

mEdTech88, 2018-07-03:
@T721 
hi. I want to read FIFO with SPI and i am wondering why I have to wait befor receiving data. I have read u dealed with the delay too. could you help me to get an idear whats going on? Thank you

T721, 2018-07-03:
You do not nead to wait when you just read the data. (From my experience)
You need to wait the driver's given time when you configure the sensor. In most other cases the code just puts the minimum wait time into it (which is 1).
You could delete the delay when reading sensor frames but i wanted to keep the driver as original as possible and just created a second wait function with a micro second resolution. (this is what my other thread is about)
When i now read data the sonsor just triggers my wait function wit 1us delay which is ok for me.
When i configure the registers, i switch back to the ms_wait function.

Here is an example:
  
```
Foo( ) {

    //Use ms delay function to setup the registers in the bmi
    sensor.delay_ms = Bmi160::MsResolutionDelay;

    struct bmi160_int_settg int1_config;

    some more initialisations done here.......

    //use Us delay function for further use
    sensor.delay_ms = Bmi160::UsResolutionDelay;
}
```

In the first lane i use the function pointer to ms wait for configurations. at the end i switch back to th us function to keep delays short.

arurke, 2019-01-05:
@T721 I tried using no delay when doing read and I cannot see any adverse effects. Have other experiences? I.e. why did you add a 1us delay on reads instead of no delays at all?