boschsensortec/BMI160_SensorAPI #56: BMI160 accelerometer & gyro sometimes do not get the same number of FIFO frames

ytham, 2018-04-21:
Hi, we are seeing an intermittent issue (~1-2% of the time) where the BMI160 gyro will consistently get less FIFO frames than the accel (we poll every ~100ms and the accel will get 10~11 frames, whereas the gyro will get 5~6 frames). 

`void IMU_ReadFIFO(struct bmi160_sensor_data *gyroFrames,
                  uint8_t *                  gyroFrameCount,
                  struct bmi160_sensor_data *accelFrames,
                  uint8_t *                  accelFrameCount) {
  mSensor.fifo->length = sizeof(mFifoBuffer);
  int8_t rslt = bmi160_get_fifo_data(&mSensor);
  APP_ERROR_CHECK_BOOL(rslt == BMI160_OK);

  if (gyroFrames != NULL && *gyroFrameCount != 0) {
    rslt = bmi160_extract_gyro(gyroFrames, gyroFrameCount, &mSensor);
    APP_ERROR_CHECK_BOOL(rslt == BMI160_OK);
  }

  if (accelFrames != NULL && *accelFrameCount != 0) {
    rslt = bmi160_extract_accel(accelFrames, accelFrameCount, &mSensor);
    APP_ERROR_CHECK_BOOL(rslt == BMI160_OK);
  }
}`

Accel & gyro are in normal mode @ 100Hz.

BoschSensortec, 2018-04-24:
Hi @ytham,

What are the values of **gyroFrameCount** and **accelFrameCount** before the respective extract calls?

ytham, 2018-04-24:
The values are always a constant calculated value (10 in this case):

  struct bmi160_sensor_data gyroFrames[MAX_IMU_FRAME_COUNT_PER_INTERVAL];  // 300 bytes / ~7bytes per frame ~ 42 data frames
  uint8_t gyroFrameCount = sizeof(gyroFrames) / sizeof(struct bmi160_sensor_data);
  struct bmi160_sensor_data accelFrames[MAX_IMU_FRAME_COUNT_PER_INTERVAL];  // 300 bytes / ~7bytes per frame ~ 42 data frames
  uint8_t accelFrameCount = sizeof(accelFrames) / sizeof(struct bmi160_sensor_data);

  IMU_ReadFIFO(gyroFrames, &gyroFrameCount, accelFrames, &accelFrameCount);

ytham, 2018-04-27:
@BoschSensortec hi, any thoughts on what could be causing the gyro and accel to get different numbers of frames?