boschsensortec/BMI160_SensorAPI #57: Unable to get BMM150 mag data thru BMI160

ytham, 2018-04-24:
I'm attempting to set up the BMI160 w/ the BMM150 according to the directions on the BMI160 github. The problem that I am having is that I have followed the directions exactly, but it doesn't look like the BMI160 is able to get the magnetometer data from the BMM150. I am just getting a bunch of empty data frames. The BMM150 works when I use its own read_mag_data command to read magnetometer data, the issue is that I cannot access it through the BMI160. 

I double checked the settings and everything seems to be as written. 

I believe the issue is the bmi160_read_aux_data_auto_mode does not seem to be getting any data from the bmm150. Is there anything I can do to test that these guys are hooked up correctly?

BoschSensortec, 2018-04-25:
Hi @ytham,

Do you already have pull-up resistors on the secondary I2C bus? In the meantime we will prepare a short guide for use with an auxiliary sensor that you can use as reference.

ytham, 2018-04-25:
@BoschSensortec yes, we currently have 2.2K pull-ups (to 1.8V) on the BMI160's ASDx and ASCx pins.

ytham, 2018-04-25:
Are there any registers we can read on the BMI160 to see if the setup is correct and the mag data is being sent to the BMI160 correctly?

ytham, 2018-04-26:
@BoschSensortec following up here.

BoschSensortec, 2018-05-02:
Hi @ytham,

Can you compare your implementation to that in the wiki's example [here](https://github.com/BoschSensortec/BMI160_driver/wiki/How-to-use-an-auxiliary-sensor-or-magnetometer-with-the-BMI160.)?