boschsensortec/BMI160_SensorAPI #58: temperature compensation

AdronTech, 2018-05-11:
Is there a temperature compensation on the BMI160?

There is a part in the datasheet which is a little bit unclear about it.
(https://www.mouser.com/ds/2/783/BST-BMI160-DS000-07-786474.pdf on page 12)

If it is so, is there a possibility to turn it off?

T721, 2018-05-24:
Hi @AdronTech,

i did some test with heating up the sensor to about 100 degrees Celsius and it had a decent change in its timinings.
I got a change of about 50µs but i cannot exactly tell how much it is per degree as i don't know the exact sensor temp. It was more like a test if the sensor is affected by temperatur changes.

AdronTech, 2018-05-25:
That's interesting. Did you see some changes in the gyro/accelerometer values? (drift, offset) 
We think, that the values are percompensated and  therefore not so useful for our application.
It would be nice if there is an option to turn the compensation off. (if there is one)

Thank you for your efforts and dedication.

T721, 2018-05-25:
I upload my picture of measurements. This displays "sensortime - 10000µs" , what means the offset from the ideal timestamp at an ODR of 10ms. I started to heat up the sensor with a heating fan up to about 100-150 degrees and then cooled down. You can clearly see in the picture how the sensor timestamps are affected by the temperature. i cannot say if the measured values itself are affected. I wouldn't know either how to measure them exactly as i don't have defined measurement environments.
I did not find any information about temperature compensation so i would think it is not compensated.

y - axis = offset from ideal 10ms
x -axis = measurement number 

[image]