boschsensortec/BMI160_SensorAPI #61: Sensor setting problem.

alextran1502, 2018-05-23:
Hello, 

When I tried to change the range of accelerometer from 2G, 4G, 8G, 16G. I don't see any changes in the raw  output value getting from `get_accel_data` 1G is always at ~17900. Is there anything I need to do to get the right sensor value?

Thanks,
Alex

alextran1502, 2018-05-25:
Hi @BoschSensortec 

Thanks for the explanation.
I just wan to clarify that in the function `int8_t bmi160_get_regs(uint8_t reg_addr, uint8_t *data, uint16_t len, const struct bmi160_dev *dev)`, the `len` will determine how many register address is read after the `reg_addr`, correct?

BoschSensortec, 2018-06-01:
Hi @alextran1502,

The `len` parameter determines how many registers are read including `reg_addr`.

closed 2018-06-06 by alextran1502