boschsensortec/BMI160_SensorAPI #66: BMI160 FIFO read missing samples

mEdTech88, 2018-07-02:
Hello every one. 
I am new doning embedded projects and I need some help doing my Project reading FIFO-Data of Accelerometer and Gyro  from BMI160. I am using SPI in Mastermode with an STM32F100 MCU ( 8MHz clock). 
I am working with CubeMX and HAL-Driver to implement my code.
When I do HAL_Transmitt() --> HAL_Delay(600) -->HAL_receive() I get Fifo Data (from register 0x24) like it is described in the Data sheet. Even the enabled Watermark interrupt and the Timestamp are working well. Do I need this DELAY() between transmit and receive? Observing the Timestamps of each FIFO I just read every second FIFO. I do need to read every FIFO without loosing dataframes. Without the Delay(600) just the timestamp is written to the FIFO. How could I read Data without loosing Frames?
Could please some one help me with this issue. 

Here my settings of BMI and SPI2 .
''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
```
> void MX_SPI2_Init(void)
> {
> 
>   hspi2.Instance = SPI2;
>   hspi2.Init.Mode = SPI_MODE_MASTER;
>   hspi2.Init.Direction = SPI_DIRECTION_2LINES;
>   hspi2.Init.DataSize = SPI_DATASIZE_8BIT;
>   hspi2.Init.CLKPolarity = SPI_POLARITY_HIGH;
>   hspi2.Init.CLKPhase = SPI_PHASE_2EDGE;
>   hspi2.Init.NSS = SPI_NSS_SOFT;
>   hspi2.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_32;
>   hspi2.Init.FirstBit = SPI_FIRSTBIT_MSB;
>   hspi2.Init.TIMode = SPI_TIMODE_DISABLE;
>   hspi2.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
>   hspi2.Init.CRCPolynomial = 10;
>   HAL_SPI_Init(&hspi2);
> 
> }
```

```
> uint8_t BMI160_Fifo_read()
> 	{
> 		Dummy_reg[0]=0x24 |0x80;
> 		SpiEn;
> 		/
> 		HAL_SPI_Transmit(&hspi2,(uint8_t*)Dummy_reg,1,sizeof(Dummy_reg));	
> 		HAL_Delay(600);
> 		HAL_SPI_Receive(&hspi2,(uint8_t*)FIFO,1024,sizeof(FIFO)); 
> 		
> 		SpiDis;		
> 	}
```

Thank U for yout Help. I really have to go on with this project to end up my Master.

mEdTech88, 2018-07-03:
@BoschSensortec I am wondering why I have to wait that long to read valid data from FIFO. When I read FIFO in my ISR (watermark Interrupt) the programm is blocked for that time. What do I miss?

BoschSensortec, 2018-07-03:
Hi @mEdTech88,

This issue tracker is reserved for queries and issues related to the BMI160 driver. We recommend you to use the driver. 
Specifications regarding the read and write communication is available in our [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000-07.pdf) on chapter 3.2.2.

mEdTech88, 2018-07-03:
Thank u @BoschSensortec 

I had problems implement the correct read and write functions with HAL so I tried it myself. Could you please tell me where to find an example to orientate?
best regards

BoschSensortec, 2018-07-03:
@mEdTech88,

In general, based on the SPI driver header [here](https://github.com/stm32duino/Arduino_Core_STM32/blob/c392140415b3cf29100062ecb083adfa0f59f8b1/system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_spi.h),

```c
void read_regs(uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
    uint32_t timeout = 100; /* Set your timeout here */
    reg_addr |= 0x80;
    SpiEn;

    HAL_SPI_Transmit(&hspi2, &reg_addr, 1, timeout );	
    HAL_SPI_Receive(&hspi2, reg_data, length, timeout ); 

    SpiDis;
}
```

```c
void write_regs(uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
    uint32_t timeout = 100; /* Set your timeout here */
    reg_addr &= 0x7f;
    SpiEn;

    HAL_SPI_Transmit(&hspi2, &reg_addr, 1, timeout );	
    HAL_SPI_Transmit(&hspi2, reg_data, length, timeout ); 

    SpiDis;
}
```

```c
/* To read the FIFO */
read_regs(0x24, FIFO, 1024);
```

mEdTech88, 2018-07-03:
@BoschSensortec 
reading the chip ID worked and writing to the registers worked as well. but if I want to read FIFO I just get the Timestamp. I work with an ODR of 100 HZ for ACC and GYRO and an STM32F100R8 MCU. The SPI rate of the sensor should be high enough with up to 10MHz.
I wonder the delay. Where do I have to wait to read all FIFO data? The watermar interrupt is aktive and set to 0x96 (600 Byte). I have to wait another 600 ms to get data. This could not be correct, isn't it?

mEdTech88, 2018-07-03:
[image]
this is what my Fifo output looks like. could u give me a hint to solve the problem. Thank u

mEdTech88, 2018-07-03:
@BoschSensortec 
@BoschSensortec 
thank you for the calculation. This i pretty much what i have calculated. 
I have configured the FIFO register with set watermark 0x96 and BMI160_FIFO_CONFIG_1_ADDR to 0xDE to enable ACC, GYRO , enable Header Mode and TIMER. I dont understand why I have to wait even if I have the watermark interrupt. When the interrupt triggers there are data to read out via SPI .... with a Delay to wait my programm is blocked....
What I need to do is read 4 FIFO blocks to calculate acceleration/movements in "real time mode"
But at the moment my BMI160_init(&sensor) crashes and stucks in the Delay_loop

```
	sensor.id=0x1D;
	sensor.interface = BMI160_SPI_INTF;
	sensor.read = read_regs;
	sensor.write = write_regs;
	sensor.delay_ms = delay_ms;
	
	rslt=BMI160_OK;
	rslt= bmi160_init(&sensor);
```

```
void delay_ms(uint32_t period)
{
	HAL_Delay(period);
	
}

```
Is this implemnentation right for init BMI160. I dont know why it crashes now.

BoschSensortec, 2018-07-03:
Hi @mEdTech88,

If you are using an interrupt as a trigger, you don't need to wait while reading out the FIFO. 

If you are using the driver, change the linked functions like below,
```c
int8_t read_regs(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
    /* dev_id can be used to help select the chip select line */
    uint32_t timeout = 100; /* Set your timeout here */
    reg_addr |= 0x80;
    SpiEn;

    HAL_SPI_Transmit(&hspi2, &reg_addr, 1, timeout );	
    HAL_SPI_Receive(&hspi2, reg_data, length, timeout ); 

    SpiDis;
    return 0;
}

int8_t write_regs(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
    /* dev_id can be used to help select the chip select line */
    uint32_t timeout = 100; /* Set your timeout here */
    reg_addr &= 0x7f;
    SpiEn;

    HAL_SPI_Transmit(&hspi2, &reg_addr, 1, timeout );	
    HAL_SPI_Transmit(&hspi2, reg_data, length, timeout ); 

    SpiDis;
    return 0;
}
```