boschsensortec/BMI160_SensorAPI #67: Issue with reading the Gyro and Accel with SPI4 Interface

ranajyotic, 2018-07-23:
Hi
I am trying to read the BMI160 and I get values on Gyro.x which will be between 0 - 255. I am using your SDK 3.7.5. Here are the configuration and read, write callbacks. The board is a nRF pca10040_xxAA. Here is the Init and Configuration code

```
void bmi160_initialize(struct bmi160_dev * p_sensor, bmi160_timeout_handler_t, bmi160_timeout_handler)
{
	int8_t rslt = BMI160_OK;
	uint32_t err_code;

	NRF_LOG_INFO("BMI160 Initialize starting\r\n");

	nrf_drv_spi_config_t spi_config = NRF_DRV_SPI_DEFAULT_CONFIG;
	spi_config.ss_pin	= APP_SS_PIN;
	spi_config.miso_pin = APP_MISO_PIN;
	spi_config.mosi_pin = APP_MOSI_PIN;
	spi_config.sck_pin  = APP_SCK_PIN;
	spi_config.mode     = NRF_SPIS_MODE_3; ///< SCK active low, sample on trailing edge of clock. 

	NRF_LOG_INFO("Initializing SPI interface\r\n");
	err_code = nrf_drv_spi_init(&m_spi, &spi_config, NULL);
	NRF_LOG_INFO("Initializing SPI interface. Error code=%d\r\n", err_code);
	APP_ERROR_CHECK(err_code);

	NRF_LOG_INFO("Initializing BMI160 timer\r\n");
	bmi160_timer_init(bmi160_timeout_handler);
	bmi160_delay(10);

	//initalize bmi160 	
	p_sensor->id = 0;
	p_sensor->interface = BMI160_SPI_INTF;
	p_sensor->read = bmi160_spi_read_callback;
	p_sensor->write = bmi160_spi_write_callback;
	p_sensor->delay_ms = bmi160_delay;

	NRF_LOG_INFO("Initializing BMI160\r\n");
	rslt = bmi160_init(p_sensor);
	if (rslt != BMI160_OK)
	{
		NRF_LOG_ERROR("BMI160 Initilization failed. Error code=%d\r\n",rslt);
		NRF_LOG_FLUSH();
		APP_ERROR_CHECK(NRF_ERROR_INTERNAL);
	}
	NRF_LOG_INFO("BMI160 Initialization complete\r\n");
	NRF_LOG_FLUSH();
}

int8_t bmi160_conf(struct bmi160_dev * p_sensor)
{
    int8_t rslt = BMI160_OK;
    NRF_LOG_INFO("Configuring BMI160 sensors\r\n");
    /* Select the Output data rate, range of accelerometer sensor */
    p_sensor->accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
    p_sensor->accel_cfg.range = BMI160_ACCEL_RANGE_2G;
    p_sensor->accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;

    /* Select the power mode of accelerometer sensor */
    p_sensor->accel_cfg.power = BMI160_ACCEL_SUSPEND_MODE;

    /* Select the Output data rate, range of Gyroscope sensor */
    p_sensor->gyro_cfg.odr = BMI160_GYRO_ODR_800HZ;
    p_sensor->gyro_cfg.range = BMI160_GYRO_RANGE_500_DPS;
    p_sensor->gyro_cfg.bw = BMI160_GYRO_BW_NORMAL_MODE;

    /* Select the power mode of Gyroscope sensor */
    p_sensor->gyro_cfg.power = BMI160_GYRO_SUSPEND_MODE; 

    /* Set the sensor configuration */
    rslt = bmi160_set_sens_conf(p_sensor);
    if (rslt != BMI160_OK)
    {
    	NRF_LOG_INFO("Failed to configure BMI160 sensor\r\n");
    	APP_ERROR_CHECK(NRF_ERROR_INTERNAL);
    }
    NRF_LOG_INFO("Completed configuring BMI160 sensorss\r\n");
    return rslt;
}

```
Here is the start function.

```
void start_bmi160(uint16_t conn_handle, struct bmi160_dev * p_sensor)
{
    /* Select the power mode */
	uint16_t err_code;
    int8_t rslt;
    NRF_LOG_INFO("Starting BMI160\r\n");
    m_conn_handle = conn_handle;
    
    p_sensor->accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
    p_sensor->gyro_cfg.power = BMI160_GYRO_NORMAL_MODE; 

    /*  Set the Power mode  */
    rslt = bmi160_set_power_mode(p_sensor);
    if (rslt != BMI160_OK)
    {
    	NRF_LOG_INFO("Error starting BMI160\r\nError code=%d",rslt);
		NRF_LOG_FLUSH();
		APP_ERROR_CHECK(NRF_ERROR_INTERNAL);
    }
    NRF_LOG_INFO("Starting BMI160 timer\r\n");
	err_code = app_timer_start(m_bmi160_timer_id, BMI160_SAMPLING_INTERVAL, NULL);
	APP_ERROR_CHECK(err_code);
    NRF_LOG_INFO("Started BMI160\r\n");
    m_bmi160_started = true;
}
```

Here is the read callback.

```
int8_t bmi160_spi_read_callback (uint8_t dev_addr, uint8_t reg_addr, uint8_t *data, uint16_t len)
{
    uint32_t 	err_code;

    NRF_LOG_INFO("BMI160 read callback. Device Address=0x%02x, Register Address=0x%02x, Length of data=%d\r\n",dev_addr, reg_addr, len);

	err_code = nrf_drv_spi_transfer(&m_spi, &reg_addr, 1, data, len);
	NRF_LOG_INFO("BMI160 read callback exit\r\n");
	NRF_LOG_HEXDUMP_INFO(data,len);
	return (int8_t)err_code;
}
```

BoschSensortec, 2018-07-24:
Hi @ranajyotic,

What do you observe in the `NRF_LOG_HEXDUMP_INFO(data,len);` in the `bmi160_spi_read_callback`?

ranajyotic, 2018-07-24:
This is the call I make to read from the sensor

`rslt = bmi160_get_sensor_data((BMI160_ACCEL_SEL | BMI160_GYRO_SEL | BMI160_TIME_SEL), p_accel, p_gyro, p_sensor);`

Some of the resulting hex dump is given below and this pattern repeats itself. So essentially I get data only on the Gyro.x 
```
FF 00 00 00 00 00 00 00 00 00 00 00 00 60 01
FF 00 00 00 00 00 00 00 00 00 00 00 00 39 06
00 00 00 00 00 00 00 00 00 00 00 00 00 15 0B
```
The last 32 bits is the sensor ticks

BoschSensortec, 2018-07-25:
Hi @ranajyotic,

Seems like the sensor isn't configured correctly. Can you share your `bmi160_spi_write_callback` too?

ranajyotic, 2018-07-25:
Here it is. It's a bit urgent so if you can respond today if you find something.

```
int8_t bmi160_spi_write_callback (uint8_t dev_addr, uint8_t reg_addr, uint8_t *data, uint16_t len)
{
    uint32_t 	err_code;
    uint8_t		tx_buf[len + 1];
    NRF_LOG_INFO("BMI160 write callback. Writing to register=0x%2X\r\n", reg_addr);

    tx_buf[0] = reg_addr;
    for (int i = 0; i < len; i++)
    {
    	tx_buf[i + 1] = data[i];
    }
	err_code = nrf_drv_spi_transfer(&m_spi, tx_buf, 1, NULL, 0);
	NRF_LOG_INFO("BMI160 write callback exit\r\n");
	return (int8_t)err_code;
}
```

ranajyotic, 2018-07-25:
Thanks a lot for pointing me to the right direction with the `bmi160_spi_write_callback` . The issue was in the code below in `bmi160_spi_write_callback`
`err_code = nrf_drv_spi_transfer(&m_spi, tx_buf, 1, NULL, 0);`
It needed to be changed to 
`err_code = nrf_drv_spi_transfer(&m_spi, tx_buf, len + 1, NULL, 0);`

closed 2018-07-25 by ranajyotic