boschsensortec/BMI160_SensorAPI #68: Confusion with sensor time data

sayan000, 2018-07-28:
I want to get some insight into the sensor time data that is send with accelerometer and gyrometer data in the FIFO frame. I get it that it is the number of ticks elapsed when the data was synchronized. Or is it something else? Pardon if it seems naive being a newbie here.

But is there a way to configure it? How to get the ticks in ms?

BoschSensortec, 2018-07-30:
Hi @sayan000,

You should find answers to your question in a previously reported [question](https://github.com/BoschSensortec/BMI160_driver/issues/22  ).

BoschSensortec, 2018-07-30:
@sayan000,

In addition, it cannot be configured.

BoschSensortec, 2018-08-07:
Hi @sayan000,

it is in ticks. Each tick represents 39.0625us.

sayan000, 2018-08-07:
thanks a lot..
So, whatever be the frequency of data-frame getting updated, how is it ensured that a particular bit in the timer register will toggle, or is it the other way  round and the bit togging is just the representation and lower the frequency, higher bits toggle because more number of ticks will occur between the time period and toggle that higher bit? Is my understanding correct?

BoschSensortec, 2018-08-08:
Hi @sayan000,

Let's assume you are reading out data at 100Hz, this implies a new sample every 10ms or in the case of the timer, every 256 ticks. You will observe that the 9th bit toggles with every sample. The lower 8 bits represent the age of the sample since it was measured.

sayan000, 2018-08-09:
Now I could get it completely..Thanks for the explanation!

closed 2018-08-09 by sayan000