boschsensortec/BMI160_SensorAPI #69: What is delay_ms in bmi160_dev struct for?

sayan000, 2018-08-11:
I can pass a user defined function pointer to the bmi160_delay_fptr_t member of bmi160_dev.
But I am not able to map it's usage or to what purpose it is used for.

It will help immensely if I can get an explanation.

BoschSensortec, 2018-08-13:
Hi @sayan000,

Hope this example [gist](https://gist.github.com/BoschSensortec/cdaa6aa244335388d914de7f59443860 ) helps.

sayan000, 2018-08-14:
Thanks..
I also wanted to know why delay_ms is passed around. Like whether the callbacks are called after this much delay or something like that?

BoschSensortec, 2018-08-14:
Hi @sayan000,

The callback is a way for the API to wait for a period amount of milliseconds. Since the execution time differs across platforms, this callback is required to make sure the API accesses the sensor with the correct delays between calls.