boschsensortec/BMI160_SensorAPI #72: compile error : undefined reference to "bmi160 driver functions"

namwoolee, 2018-11-01:
Hi All,
I have ported to my project bmi160 drivers.(bmi160.c , bmi160.h, bmi160_defs.h)
Also, it was add application source to userspace.
But, it is generated compile error as below in the main routine.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
LTIB> make
cat: /home/leps/ltib/Work/dIP: No such file or directory
rm -rf drdr
gcc -o drdr main.o server.o version.o rtc.o configini.o bmi160_i2c.o -lrt -lpthread -lm
bmi160_i2c.o: In function `bmi160_i2c':
bmi160_i2c.c:(.text+0x354): undefined reference to `bmi160_init'
collect2: ld returned 1 exit status
Makefile:43: recipe for target 'drdr' failed
make: *** [drdr] Error 1
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

<application source to userspace as follow>
======================================================================
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <fcntl.h>
#include <ctype.h>
#include <termios.h>
#include <errno.h>
#include <pthread.h>
#include <time.h>

#include <arpa/inet.h>
#include <netinet/in.h>
#include <sys/ioctl.h>
#include <sys/stat.h>
#include <sys/types.h>
#include <sys/resource.h>
#include <sys/socket.h>
#include <sys/mman.h>
#include <sys/time.h>
#include <poll.h>

#include <linux/kernel.h>
#include <sys/syscall.h>

#include <linux/i2c-dev.h>

#include "bmi160_i2c.h"
#include "../bmi160_driver/bmi160.h"

//#include "i2c-dev.h"

//#define TWI_INSTANCE_ID     0

struct bmi160_dev sensor;
//static const drv_twi_t m_twi = DRV_TWI_INSTANCE(TWI_INSTANCE_ID);

/* Indicates if operation on TWI has ended. */
//static volatile bool m_xfer_done = false;

void delay_ms(long ms)
{
    time_t lInitTime;
    time_t lCurrentTime;
    time(&lInitTime);

    do {
        time(&lCurrentTime);
    } while(((lCurrentTime - lInitTime) * 1000) < ms);
}

[... 570 lines cut ...]
bmi160_i2c.o: bmi160_i2c.c
	$(CC) $(CFLAGS) -c bmi160_i2c.c  

configini.o: ../_Aura_Common/libconfigini/src/configini.c
	$(CC) $(CFLAGS) -c ../_Aura_Common/libconfigini/src/configini.c

$(OUTPUT): main.o server.o version.o rtc.o configini.o bmi160_i2c.o
	rm -rf $(OUTPUT)
	$(CC) -o $(OUTPUT) main.o server.o version.o rtc.o configini.o bmi160_i2c.o $(LIBS)
	rm -rf *.o *.bak .*.swp

clean :
	rm -rf *.o $(OUTPUT) *.bak .*.swp

up : $(OUTPUT)
	wput -B --reupload -nc $(OUTPUT) ftp://root:@$(DIP)
	======================================================================
	
	
	
Is the system call problem? or not, driver API link problem?
Could you let me know your idea?

Thanks & BR,
nwlee

BoschSensortec, 2018-11-02:
Hi @namwoolee,

It is a linking issue. You haven't compiled the bmi160.c file. It should be
```
gcc -o drdr main.o server.o version.o rtc.o configini.o bmi160_i2c.o bmi160.o -lrt -lpthread -lm
```

namwoolee, 2018-11-07:
Hi Bosch,

Thank you for your support.
I have solved this issue with your comment.
Thank you so much.
But, I have a question.
I have ported bmi160 driver in kernel. and then, I have made kernel object module.
but, your comment is seem to be add in application code. Is it right?
So, Is it not used system call?

Thanks & BR,
nwlee.
Thanks,