boschsensortec/BMI160_SensorAPI #73: Trouble writing user_spi_read and user_spi_write function

ainsleynand, 2018-11-02:
I am using an ATtiny441 to interface with the BMI160, however I am currently having issues with setting up the user_spi_read and user_spi_write functions.

```
sensor.read = user_spi_read;
sensor.write = user_spi_write;
```

As per the ATtiny data sheet I have a transmit function: 

```
unsigned char SPI_MasterTransmit(char cData) {
	/* Start transmission */
	SPDR = cData;
	/* Wait for transmission complete */
	while(!(SPSR & (1<<SPIF)))
	;
	
	return(SPDR);
}
```

I have tried to write the spi_read_transfer function, but I don't know where the slaveDeviceId and reg_addr should be used or how

```
int8_t spi_read_transfer(uint8_t slaveDeviceId, uint8_t reg_addr, uint8_t *txData, uint16_t bytesNumberTx) {
	
	for (int i = 0; i < bytesNumberTx; i++) {
		SPI_MasterTransmit(txData[i]);
	}
	
	return BMI160_OK;
}
```

Cheers

BoschSensortec, 2018-11-02:
Hi @ainsleynand,

It should look like,

```c
int8_t spi_read_transfer(uint8_t slaveDeviceId, uint8_t reg_addr, uint8_t *txData, uint16_t bytesNumberTx) {
        /* Put the chip select low */

        SPI_MasterTransmit(reg_addr);	

	for (int i = 0; i < bytesNumberTx; i++) {
		SPI_MasterTransmit(txData[i]);
	}
	
        /* Put the chip select high */

	return BMI160_OK;
}
```

ainsleynand, 2018-11-03:
Cheers, I have implemented that section of code. However, I noticed an issue with the bmi160_init function when running the device as SPI 4-wire.

```
int8_t bmi160_init(struct bmi160_dev *dev) {
	/* I am just showing the code I am having an issue with */
	if (rslt == BMI160_OK) {
		/* Assign chip id as zero */
		dev->chip_id = 0;

		while ((try--) && (dev->chip_id != BMI160_CHIP_ID)) {
			/* Read chip_id */
			rslt = bmi160_get_regs(BMI160_CHIP_ID_ADDR, &dev->chip_id, 1, dev);
		}
        }
```
The value of rslt after the section above is BMI160_OK. But, dev->chip_id = 0x00, which means the following section below returns BMI160_E_DEV_NOT_FOUND.

```
if ((rslt == BMI160_OK) && (dev->chip_id == BMI160_CHIP_ID)) {
	dev->any_sig_sel = BMI160_BOTH_ANY_SIG_MOTION_DISABLED;
	/* Soft reset */
	rslt = bmi160_soft_reset(dev);
} else {
	serialWrite("bmi160 isn't okay\n");
	rslt = BMI160_E_DEV_NOT_FOUND;
}
```

How can I rectify this issue to allow the BMI160 to run correctly as SPI 4 wire.

Cheers

BoschSensortec, 2018-11-05:
Hi @ainsleynand,

Can you describe your connection diagram between the ATtiny and the BMI160 assuming they are both running at less than 3.6V.

ainsleynand, 2018-11-08:
Both ATtiny and BMI160 are running on 3.3V, and connected as per 4.2.3. SPI 4-wire as primary interface to the ATtiny's MISO, MOSI, SCK and SS