boschsensortec/BMI160_SensorAPI #74: How to set-up I2C communication

namwoolee, 2018-11-05:
Hi bosch,

Thank you for your support always.
I'd like to know How to set-up I2C communication read/write.
my source is as follow.

=================================================
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <fcntl.h>
#include <ctype.h>
#include <termios.h>
#include <errno.h>
#include <pthread.h>
#include <time.h>
#include <arpa/inet.h>
#include <netinet/in.h>
#include <sys/ioctl.h>
#include <sys/stat.h>
#include <sys/types.h>
#include <sys/resource.h>
#include <sys/socket.h>
#include <sys/mman.h>
#include <sys/time.h>
#include <poll.h>

#include <linux/kernel.h>
#include <sys/syscall.h>

#include <linux/i2c-dev.h>

#include "bmi160.h"

#include "bmi160_i2c.h"

#include "i2c-dev.h"

//#define TWI_INSTANCE_ID     0

struct bmi160_dev sensor;
//static const drv_twi_t m_twi = DRV_TWI_INSTANCE(TWI_INSTANCE_ID);

/* Indicates if operation on TWI has ended. */
//static volatile bool m_xfer_done = false;

void delay_ms(long ms)
{
    time_t lInitTime;
    time_t lCurrentTime;
    time(&lInitTime);

    do {
        time(&lCurrentTime);
    } while(((lCurrentTime - lInitTime) * 1000) < ms);
}

void bmi_160_ioctl(void)
{
	int file;
	int adapter_nr = 3;
	char filename[20];

//	int addr = 0x40; //I2c address
	int addr = 0x68; //I2c address
//	__u8 reg = 0x10; //device register to access
	__u8 reg = 0x00; //device register to access
	__s32 res;
	char buf[10];
[... 546 lines cut ...]

and, result for output is as below.
============================================
reg_addr:96, data:20, len:1
reg_addr:0, data:20, len:1
rslt : 0, BMI160_I2C_ADDR : 104, data : 0, dev : 268564568
BMI160_CHIP_ID : 209, BMI160_CHIP_ID_ADDR : 0, dev->chip_id : 0, dev : 268564568
BMI160_CHIP_ID : 209, BMI160_CHIP_ID_ADDR : 0, dev->chip_id : 0, dev : 268564568
BMI160_CHIP_ID : 209, BMI160_CHIP_ID_ADDR : 0, dev->chip_id : 0, dev : 268564568
rslt : -5, BMI160_BOTH_ACCEL_AND_GYRO : 3, sensor:26624

 ACCEL SELF TEST RESULT FAIL
[get_accel_gyro_data]gyro->x : 0
[get_accel_gyro_data]gyro->y : 0
[get_accel_gyro_data]gyro->z : 0
[get_accel_gyro_data]accel->x : 0
[get_accel_gyro_data]accel->y : 0
[get_accel_gyro_data]accel->z : 0
./auto.sh: line 9:  2485 Segmentation fault      ./drdr
============================================

I think that i2c comm is not working.
Could you let me know your idea?
Thanks & BR,
nwlee.

BoschSensortec, 2018-11-05:
@namwoolee,

Can you describe your hardware and software platform in brief?

BoschSensortec, 2018-11-05:
Hi @namwoolee,

Could you please provide a leaner snippet. The code provided in the comment doesn't seem to produce the output you are showing.

namwoolee, 2018-11-06:
What is leaner snippet?
I want to know “user_i2c_read” and “user_i2c_write” sample code.
My source is not working that’s point.

Thanks,
nwlee

namwoolee, 2018-11-07:
Hi Bosch,

Thank you for your support always.

First of all, I want to output raw data for accelometer.
but, outuput of the raw data is 0 as follows.

=================================================================================================================
reg_addr:0x0, data:16, len:1
reg_addr:0x60, data:20, len:1
dev->id:0x68, reg_addr:0xe8, data:2127826007, len:1
[bmi160_init] rslt : 0, BMI160_I2C_ADDR : 0x68, data : 0, dev : 1001f7c8
dev->id:0x68, reg_addr:0x80, data:268564424, len:1
[bmi160_init] rslt : 0, BMI160_CHIP_ID : 0xd1, BMI160_CHIP_ID_ADDR : 0x0, dev->chip_id : 0x0, dev : 1001f7c8
dev->id:0x68, reg_addr:0x80, data:268564424, len:1
[bmi160_init] rslt : 0, BMI160_CHIP_ID : 0xd1, BMI160_CHIP_ID_ADDR : 0x0, dev->chip_id : 0x0, dev : 1001f7c8
dev->id:0x68, reg_addr:0x80, data:268564424, len:1
[bmi160_init] rslt : 0, BMI160_CHIP_ID : 0xd1, BMI160_CHIP_ID_ADDR : 0x0, dev->chip_id : 0x0, dev : 1001f7c8
dev->id:0x68, reg_addr:0xc0, data:2127826004, len:2
dev->id:0x68, reg_addr:0xc2, data:2127826004, len:2
dev->id:0x68, reg_addr:0xc0, data:2127825982, len:1
dev->id:0x68, reg_addr:0x92, data:2127825968, len:6
[get_accel_data] accel->x : 0
[get_accel_data] accel->y : 0
[get_accel_data] accel->z : 0
dev->id:0x68, reg_addr:0xc0, data:2127825948, len:2
dev->id:0x68, reg_addr:0x40, data:2127825948, len:1
dev->id:0x68, reg_addr:0x41, data:2127825949, len:1
dev->id:0x68, reg_addr:0xc2, data:2127825948, len:2
dev->id:0x68, reg_addr:0x42, data:2127825948, len:1
dev->id:0x68, reg_addr:0x43, data:2127825949, len:1

 ACCEL SELF TEST RESULT FAIL
=================================================================================================================
 
My source is as below.

=================================================================================================================

struct bmi160_dev sensor;

int8_t user_i2c_Read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	int8_t rslt = 0;
	return rslt;

}

int8_t user_i2c_Write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	int8_t rslt = 0;
	return rslt;

}

void user_delay_ms(uint32_t period)
{ 
  delay_ms(100) ;
}

void delay_ms(long ms)
{
    time_t lInitTime;
    time_t lCurrentTime;
    time(&lInitTime);

    do {
        time(&lCurrentTime);
    } while(((lCurrentTime - lInitTime) * 1000) < ms);
}
[... 335 lines cut ...]
//		if ((rslt == BMI160_OK) && (dev->interface == BMI160_SPI_INTF)) {
		if ((rslt == BMI160_OK) && (dev->interface == BMI160_I2C_INTF)) {
			/* Dummy read of 0x7F register to enable SPI Interface
			if SPI is used */
//			rslt = bmi160_get_regs(BMI160_SPI_COMM_TEST_ADDR, &data, 1, dev);
			rslt = bmi160_get_regs(BMI160_I2C_ADDR, &data, 1, dev);
			
			printf("[bmi160_soft_reset] rslt : %d, BMI160_I2C_ADDR : 0x%x, data : %d, dev : %x\n", rslt, BMI160_I2C_ADDR, data, dev);
			
		}
		if (rslt == BMI160_OK) {
			/* Update the default parameters */
			default_param_settg(dev);
		}
	}

	return rslt;
}
=================================================================================================================

What's wrong?
How can I correct modify this code?

Thanks & BR,
nwlee.

namwoolee, 2018-11-08:
My snippet is as follows.

		/* Select the Output data rate, range of accelerometer sensor */
		sensor.accel_cfg.odr = BMI160_ACCEL_ODR_1600HZ;
		sensor.accel_cfg.range = BMI160_ACCEL_RANGE_2G;
		sensor.accel_cfg.bw = BMI160_ACCEL_BW_NORMAL_AVG4;
		/* Select the power mode of accelerometer sensor */
		sensor.accel_cfg.power = BMI160_ACCEL_NORMAL_MODE;
		/* Set the sensor configuration */
		rslt = bmi160_set_sens_conf(&sensor);
		/* To read only Accel data */
		rslt = bmi160_get_sensor_data(BMI160_ACCEL_SEL, &accel, NULL, &sensor);

Thanks,
nwlee